dcowden/cadquery #67: transformed() Fuction Uses Vectors Instead of Tuples

jmwright, 2015-01-05:
Is there any reason that a Vector is used here instead of accepting a Vector or a three tuple like CQ does in other places? Seems like we're heading more towards using standard data structures instead of (or as an alternative to) Vectors, so I wanted to check. This will force me to do an extra import in my script.

https://github.com/dcowden/cadquery/blob/a735b74a4fe78f5d689b2f3a07c7f90ff5b823f0/cadquery/workplane.py#L92

dcowden, 2015-01-05:
No... we should definitely be moving to 3-tuples....

jmwright, 2015-01-06:
So, I just looked around a little more and noticed that this is a discrepancy between cadquery/workplane.py/Workplane and cadquery/CQ.py/Workplane. The CQ version of Workplane handles Vectors and tuples. 

https://github.com/dcowden/cadquery/blob/65381225d0b77084d20dece2821f336e446b447c/cadquery/CQ.py#L865

We've talked before about getting rid of the workplane.py file since it seems like an artifact of the port from PParts.com. I may experiment with removing that and then seeing if all the tests still pass.

dcowden, 2015-01-06:
yeah i think that's exactly whats needed.  i moved things around when i
ported it over, and i was trying to move from Vector to tuple.  obviously
there should be only one workplane.

jmwright, 2015-01-06:
There was no matching pyc file of workplane.py on my CQ installation, so I figured that was a good start. All of the tests ran, so I think we're good.

I've done a pull request and the new Travis build (yay!) is in progress.

jmwright, 2015-01-08:
Forgot to close this. Addressed in commit ec3a1de09e8a38e437d9b582812486eae159d668

closed 2015-01-08 by jmwright