dcowden/cadquery #69: Ex027_Remote_Enclosure.py Won't Execute in Windows

jmwright, 2015-01-14:
This example works fine in FreeCAD 0.15 on Ubuntu 14.04. The last line complains about `ChFi3d_Builder:only 2 faces`.
```python
Opened C:\Program Files\FreeCAD 0.14\Mod\CadQuery\Examples\Ex027_Remote_Enclosure.py
Running the Python command '<p>FreeCAD Standard File</p>' failed:
Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.14\Mod\CadQuery\Gui\Command.py", line 135, in Activated
    CadQueryExecuteScript().Activated()
  File "C:\Program Files\FreeCAD 0.14\Mod\CadQuery\Gui\Command.py", line 178, in Activated
    imp.load_source('temp.module', tempFile.name)
  File "c:\users\jeremy\appdata\local\temp\tmp6tshb5", line 82, in <module>
    .fillet(.020)
  File "C:\Program Files\FreeCAD 0.14\Mod\CadQuery\Libs\cadquery\CQ.py", line 793, in fillet
    s = solid.fillet(radius,edgeList)
  File "C:\Program Files\FreeCAD 0.14\Mod\CadQuery\Libs\cadquery\freecad_impl\shapes.py", line 813, in fillet
    return Shape.cast(self.wrapped.makeFillet(radius, nativeEdges))

ChFi3d_Builder:only 2 faces
```
I'm thinking FreeCAD bug, but I'm not sure yet. Curious if @emdash has any thoughts on what might cause this.

emdash, 2015-01-15:
It could be a floating point rounding error. Is this a 32-bit or 64-bit python interpreter?

--Brandon

jmwright, 2015-01-15:
Windows 8 64-bit with FreeCAD 0.14 revision 3700 (Git). I checked the Python interpreter that's embedded with FreeCAD directly, and it's 64-bit.

Here's the [link](http://stackoverflow.com/questions/1405913/how-do-i-determine-if-my-python-shell-is-executing-in-32bit-or-64bit-mode-on-os) I used to help me verify that the embedded Python version is 64-bit. Only significant change was that I had to escape the interior double quotes.
```python
.\python.exe -c 'import sys;print(\"%x\" % sys.maxsize, sys.maxsize > 2**32)'
```

jmwright, 2015-03-03:
This looks like a bug with the version of OCC (or how FreeCAD is using OCC's fillet operation) that's packaged with FreeCAD 0.14 on Windows.

closed 2015-03-03 by jmwright