dcowden/cadquery #70: How to make a sphere?

tobltobs, 2015-02-03:
cadquery looks promising, but how to make a sphere?

jmwright, 2015-02-03:
I haven't tried this yet, but there is a [makeSphere function](http://parametricparts.com/docs/primitiveref.html?highlight=sphere#cadfile.cadutils.cad.Solid.makeSphere). CQ also has a revolve function that can be used to make a sphere.

jmwright, 2015-02-04:
@tobltobs I just tried out CQ's makeSphere function, and it is much clunkier to use than most things in CQ. Another problem is that it doesn't seem like CQ is using FreeCAD's makeSphere implementation correctly. The implementation may have changed since the CQ function was written. The following work-around code creates a sphere, although I'm sure that my code is far from optimal.
```python
import cadquery as cq
from FreeCAD import Base
import Part

pnt1 = Base.Vector(0, 0, 0)
pnt2 = Base.Vector(0, 0, 1)
sphere = cq.Solid.makeSphere(2, pnt1, pnt2, 0, 360)
Part.show(sphere.wrapped)
```
The FreeCAD documentation shows the proper makeSphere call [here](http://www.freecadweb.org/wiki/index.php?title=Topological_data_scripting) in the "Creating a Sphere" section.

Another way would be to use a revolution operation like so...
```python
import cadquery as cq
from Helpers import show

wire = cq.Workplane("XY").threePointArc((1.0, 1.0), (0.0, 2.0)).close()
sphere = wire.revolve()

show(sphere)
```
The second option has the advantage of keeping your solid as a higher level CQ object, which makes the code cleaner. However, I'm guessing that the FreeCAD makeSphere code probably does some extra things to make sure that you get a proper sphere with one continual surface. In theory the second method should give you that too, but the resulting solid would have to be tested thoroughly to be sure.

Does any of that help?

dcowden, 2015-02-04:
yeah definitely the need to create Base.Vectors , or to explicitly
reference .Solid is not cq-esque.
 I dont think that's intentional, i think its just not bee pushed up to the
cq object layer right.  I think the code should be as easy as :

w = cq.Workplane("XY").sphere(2.0);

of course there should be other forms that include other optional
arguments... for example the angle of sweep.

tobltobs, 2015-02-04:
Thank you, that helped.

closed 2015-02-04 by tobltobs