dcowden/cadquery #71: How to Union two elements at a specific position?

tobltobs, 2015-02-04:
I try to place dots on a ground plate (think braille sign).
```python
import cadquery as cq
import Part

def get_dot():
        return cq.Workplane("XY").threePointArc((0.4, 0.6), (0.0, 0.2)).close().revolve()

r = cq.Workplane("XY").box(10, 10, 0.5)

#r = r.faces(">Z").workplane()
#r = r.center(2,2)
#r.pushPoints([(2,2), (1,1)])

r = r.union(get_dot())

Part.show(r.toFreecad())
```
The outcommented lines shows different approaches I tried (pushPoint, center) . But the result was always a dot in the center of the box. Any idea what I am doing wrong or suggestions for a better approach to solve this problem.

dcowden, 2015-02-04:
Alright so i have not run your example, but here's what i think is
happening:

your code for pushpoints was fine.
when you run get_dot(), you are always creating a new workplane. that
workplane is always centered at (0,0), so you always get a dot thats at
that lociation.

Then, you iterate over a bunch of points, but the dot is not added where
you want it, its always at the center.

What you need to do is to make sure that when get_dot() is run, that it
gets the center of the current point, not the default center of the newly
created work plane.

There are two approaches you can consider:

(1) modify get_dot to accept the coordinates for each point-- iterate over
your points and union the bumps yourself. This woudl work, but would not be
very 'elegant'.

(2) write a cadquery plugin.  Here, you hook your function into CQ, so that
you can do:
r.pushPoints([(2,2),(1,1,)].makeBumps();

accomplishing what you want-- which is to create a bump at every point on
the CQ stack-- in one operation. To help do that, have a look at this
example, which illustrates doing nearly exactly what you want-- it creates
cubes at each point rather than a hemisphere.

http://parametricparts.com/docs/extending.html

tobltobs, 2015-02-04:
Ok, I will try your suggestions. 

 But why does it not work if I move the center on the plate and than union.
```python
def get_dot():
        return cq.Workplane("XY").threePointArc((0.4, 0.6), (0.0, 0.2)).close().revolve()

r = cq.Workplane("XY").box(10, 10, 0.5)
r.center(2,2)                                 # or is it r = r.center(2,2)?
r = r.union(get_dot())

Part.show(r.toFreecad())
```

dcowden, 2015-02-04:
it doesnt work because doing   r.center(2,2) moves the center of THAT
workplane to (2,2), but NOT the center of the workplane you are creating in
the get_dot function.

No matter what your code outside of get_dot does, get_dot will always
produce a workplane that is centered at (0,0). That's because you're
creating a new one in that function with the Workplane("XY") constructor.

jmwright, 2015-02-07:
@tobltobs I wanted to check in to see how you're doing with this. Any progress?

closed 2015-02-10 by jmwright