dcowden/cadquery #72: Implement new Sphere Function with CQ Best Practices

jmwright, 2015-02-10:
The current makeSphere function isn't very CQ-ish (requires vectors instead of standard tuples), and no longer matches FreeCAD's implementation of makeSphere. Dave made a suggestion on a re-implementation [here](https://github.com/dcowden/cadquery/issues/70). I would like to make sure that the new sphere function works the same as something like cboreHole, where a sphere would be placed at the position of each point on the stack. For that matter, I think it would be good to have all high level solids act that way, if they don't already.

jmwright, 2015-03-30:
I'd like to see if @tobltobs is still interested in this feature.

tobltobs, 2015-03-30:
For the moment I am using freecad directly, so I am not interested anymore.

jmwright, 2015-03-30:
OK, thanks for letting me know.

jmwright, 2015-04-07:
This is addressed in PR #82

jmwright, 2015-04-08:
Inspired by a comment from @tobltobs I went ahead and fixed this feature in CQ, but it's not in master yet. However, I did push that branch into the latest commit of the [CQ FreeCAD module](https://github.com/jmwright/cadquery-freecad-module) and have been using it. I started a text to 3D braille translator, which I hope to do an example and demo video for soon. Here's the code I have so far.
```python
import cadquery as cq
from Helpers import show

# Resources:
# http://www.brailleauthority.org/sizespacingofbraille/sizespacingofbraille.pdf
# http://www.dotlessbraille.org/AboutBraille.htm

# Enter your text here to be translated into braille
text = "abcde"

# Specs for our dots
dotDist = 2.54  # The distance between the dots, on center
cellDist = 5.08 # The distance between cells, second column of dots to first column
dotRad = 0.635  # The radius of the dots

# Store these in a dictionary with the letters as keys to make for clean code
letters = {'a': [(0.0, 0.0)], 'b': [(0.0, 0.0), (0.0, -dotDist)],
           'c': [(0.0, 0.0), (dotDist, 0.0)],
           'd': [(0.0, 0.0), (dotDist, 0.0), (dotDist, -dotDist)],
           'e': [(0.0, 0.0), (dotDist, -dotDist)]}

# These help us set our braille cells in the right relation to each other
dots = []
offset = 0

# Step through each letter of the text to convert to 3D braille dots
for char in text:
    # Step through each tuple that makes up that character
    for dotTuple in letters[char]:
        # We need to put this set of dots to the right of previous cells
        dots = dots + [(dotTuple[0] + offset, dotTuple[1])]

    # Our offset for the next cell if from the right-most dot of the previous
    # cell, to the left-most dot of the next cell
    offset = dots[-1][0] + cellDist

# Create a plate to put the dots on and push our points for the dots onto it's
# topmost Z face
plate = cq.Workplane('XY').box(40, 10, 1.0).faces('>Z').workplane() \
                          .center(-15, 1) \
                          .pushPoints(dots)

# Create our dots
plate = plate.sphere(dotRad, angle1=0, angle2=90, angle3=360,
                     centered=(False, False, False))

show(plate, (20, 153, 211, 0.0))
```
That code produces this...
[image]

This is far from complete, of course. I'm not very familiar with braille signage specifications, and it looks like the standards in the US are not well defined. There are all sorts of directions you could take this though, including automatically outputting an STL so that the signage could be sent straight to a 3D printer.

tobltobs, 2015-04-08:
Take a look at http://braille.solutions 
While already functional it is not really ready for production. There are some problems with Spanish braille translation and I have to rethink the decision to use shapeways.com .
Another problem is that the filleting of the letters does not work or is too slow.

The biggest problems for this kind of online 3D thing creator is speed and filleting/chamfering.
I started this project with OpenSCAD and that is a nice tool (especially the minkowski transform), but it is so slow. Next stop was CQ [1]. Which has also a clean syntax and looks promising but misses a lot of functionality. From Cadquery I went to Freecad, which is ok speed wise, but chamfer- and filleting isn't really working. I even tried this military CAD, but I gave up this one ...

Without filleting the corners of the plate would be to sharp. This would be doable with CQ/freecad of course, but the letters on the plate of my sign creator should also chamfered or filleted.
Also the dots should not go straight into the plate. They should look a bit like a sinus curve. Idea is to prevent corners to improve hygiene through better clean ability. 
Sadly I did not find any open source pro grammatically CAD which is able to do those jobs. 

I currently work on a pipeline where those filleting problems are done by mesh converting and vcg reconstruction which results in filleted corners. But it doesn't look promising. Slowly I come to the conclusion, that a online braille plate generator might look easy, but for the moment it is to early as the tools are not ready yet.
 
[1] Using CQ as a abbreviation for Cadquery is a bad idea imho as CQ stands already for Communique, a very enterprisey CMS, formerly from Day now owned by Adobe.

jmwright, 2015-04-08:
Very cool service.  Sorry that the tech is letting you down.

> I have to rethink the decision to use shapeways.com

MakeXYZ has an API that I believe @dcowden has used for ParametricParts.com. That could be one alternative.

> Next stop was CQ [1]. Which has also a clean syntax and looks promising but misses a lot of functionality.

Would you be willing to list out the features that you feel CadQuery is missing? I can look at converting them into feature requests.

> Also the dots should not go straight into the plate. They should look a bit like a sinus curve. 

I think that this may be doable with CadQuery, but it sounds like the least of the problems you're having.

> Using CQ as a abbreviation for Cadquery is a bad idea imho as CQ stands already for Communique, a very enterprisey CMS, formerly from Day now owned by Adobe. 

Thanks for the heads-up.

jmwright, 2015-04-09:
@tobltobs I was thinking that CadQuery had a chamfer function, but I guess I was mistaken. I would expect to be able to model it on the fillet function though, and don't think it would be too hard to add.

I roughed out the following code as one way to remove several sharp edges in one shot. It substitutes fillet for chamfer.
```python
import cadquery as cq
from Helpers import show

r = cq.Workplane('XY').box(100, 29, 1.8)

# Cutouts to do a rough simulation of the lettering on the braille signage
r = r.faces('>Z').workplane().center(-20, 0).circle(10).cutThruAll()
r = r.faces('>Z').workplane().center(20, 0).rect(10, 10).cutThruAll()

r = r.faces('>Z').edges().fillet(1.0)

# Use the following to render your model with grey RGB and no transparency
show(r, (204, 204, 204, 0.0))
```
That code yields the following.
[image]

jmwright, 2015-04-13:
This has been merged into master in commit cde33d82041168ebd1ad271e2de34f16da51189c

closed 2015-04-13 by jmwright