dcowden/cadquery #75: Line Variation that Specifies an Angle

jmwright, 2015-03-02:
AutoCAD allows you to specify lines from an origin in polar coordinate format. You can say something like start here and draw a line 100mm long at 45 degrees. I don't believe that CQ can do that, and it would be helpful for me if it did. Also, having a continuation option where the next line/arc will start from the end of the last one (the way AutoCAD can), would be handy. It would aid in creating quick 2D sketches.

[AutoCAD Line Documentation](http://knowledge.autodesk.com/support/autocad-lt/getting-started/caas/CloudHelp/cloudhelp/2015/ENU/AutoCAD-LT/files/GUID-58C1D97C-A9B3-4C3C-AC1B-95BE3DF2EDB9-htm.html)

dcowden, 2015-03-02:
CQ kind of already has a continuation feature: you can do
.lineTo(2,3).lineTo(4,5).arcTo(..) etc.

I agree polar would be nice. But honestly i think what would be even nicer
is to import DXF. the sad truth is that for doing 2-d work, its a lot
easier to draw armed with a GUI. Though admittedly, they would be very
non-parametric ( thus the reason i didnt want to support this earlier )

jmwright, 2015-03-02:
> CQ kind of already has a continuation feature:

That's true, I wasn't really thinking about chaining. I've used polyline for this purpose too, which does all the continuations for you based on a 2D array of points. But, that doesn't do the arc, as your example does.

>  But honestly i think what would be even nicer is to import DXF.

That would be handy. Maybe this is a case where both would be best though? The DXF import would be the priority, followed by polar?

dcowden, 2015-03-03:
Yeah I  think dxf would be much more used.

jmwright, 2015-03-03:
Added a separate feature request for DXF import: https://github.com/dcowden/cadquery/issues/76

jmwright, 2015-09-17:
We don't need this right now, so I've noted it in the "Future Ideas" section of the developer documentation (wiki).

closed 2015-09-17 by jmwright

jmwright, 2018-08-26:
Added in #292