dcowden/cadquery #76: Feature Request: DXF Import

jmwright, 2015-03-03:
Some 2D work will probably always be easier to do outside of CQ. There should be a way to import the DXF formated 2D into CQ. The drawback is that the imported 2D won't be parametric.

dcowden, 2015-03-03:
I think a reasonable compromise is to import a DXF into a CQ chained
operation.  that way, at least its easy to go back and add variables where
there were previously constants.

jmwright, 2015-03-03:
Would each line in the DXF be imported as a lineTo call, and each arc as an arcTo?

dcowden, 2015-03-03:
thats the general thought, but there are a lot of complex gotchas:

do you detect closed/open curves? is each closed/open curve a sketch?
what about disconnected segments? obviously each connected chain would need
to be separate. but where would you start? is it arbitrary?
what plane do you put those imported sketches on?

jmwright, 2015-03-03:
I have trouble keeping terminology straight, so I'm pasting some content from the sweep operation discussion here first.

> A _curve_ is a geometric construct. curves include splines, circles, lines, etc. 
An edge is a portion of a curve that represents part of an object definition. 
A wire is a collection of more than one edge, in a particular order 
A face is one or more wires, that form a closed loop 

I don't really know, but I'm going to throw these thoughts out there.

> do you detect closed/open curves? 

If possible to do accurately, yes.

> is each closed/open curve a sketch?

I think we'd have to look at it that way.

> obviously each connected chain would need to be separate. but where would you start? is it arbitrary?

I would assume that where we start would be determined by the internal structure of the DXF file (what you get to first), but maybe closed curves would be searched for first.

> what plane do you put those imported sketches on?

I would say that we should default to 'XY'. Seems like that would be the most intuitive for the user.

jmwright, 2015-09-17:
Cleaning up a little bit. This is not a feature that anyone else but me has asked for, and my need for it has passed. I'm going to note it in the "Future Ideas" section of the developer documentation, but don't want it cluttering up the issue tracker.

closed 2015-09-17 by jmwright

greyltc, 2018-10-07:
This feature would be incredibly useful for me!