dcowden/cadquery #77: Coveralls.io

jmwright, 2015-03-03:
Coveralls would add a badge right next to our Travis build status badge that lets us know the percentage of our code that's covered by tests.

https://coveralls.io/

Thoughts?

jmwright, 2015-03-03:
I experimented with this a little bit, and their service won't sync our repos for some reason. Other users have mentioned the same thing, but there's been no answer. I thought this would be a quick add, but am not going to worry about it now.

closed 2015-03-03 by jmwright

dcowden, 2015-03-03:
ok but thanks for doing the investigation!

jmwright, 2015-03-03:
It ended up taking a lot longer than I thought it should for Coverage.io to sync up with GitHub. I'm just impatient I guess. Anyway, it works on the branch and gives a score, so you'll see a PR-merge cycle here in a minute.

jmwright, 2015-03-03:
The build is still being tested, but if you look at README.md you'll see a badge for Coverall. Once the Travis build is done the score should show up there. 

@dcowden - If you log into Coverage.io with your Github account, you should have full access to this and can set thresholds, etc.

dcowden, 2015-03-03:
sweet! that's awesome! i love it!

jmwright, 2015-03-03:
Looks like it would be pretty easy to add a version badge too.

http://badge.fury.io/for/py/cadquery

Two questions come to mind though.

1. Do we want this?
2. If so, do we use the version of CQ in PyPi, the newest version from the repo, or both?

dcowden, 2015-03-03:
i'd say go for it, if you are willing.  it makes the most sense to me to
use the version that was tested-- which i guess is the repo version.

jmwright, 2015-03-04:
It only took me about 30 seconds to add it. I like it. It's based on the latest release from the GitHub repository.

https://github.com/dcowden/cadquery

dcowden, 2015-03-04:
Yes it looks sharp!