dcowden/cadquery #79: Releasing 0.1.8

jmwright, 2015-03-04:
When I was adding the version badge to README.md, it got me thinking about whether v0.1.8 could be released. No major features were added, but quite a bit of good stuff has been done in this version. I'm thinking it's enough to release it, but want to see if @dcowden agrees. We don't have any release guidelines anywhere that I know of, so this is a judgement call.

https://github.com/dcowden/cadquery/blob/master/changes.md#v018-unreleased

dcowden, 2015-03-04:
Yes let's release. Even though the are small it's progress.

jmwright, 2015-03-04:
The version numbers in the source have been updated to v0.1.8, the version has been tagged, and the badge is showing the correct version.

@dcowden can you push this version out to PyPi when you get a chance?

dcowden, 2015-03-04:
yep i'll do that tonight.

jmwright, 2015-03-04:
Great, thanks.

dcowden, 2015-03-04:
version 0.18 should be in pypy now!
thanks for the work to publish!

jmwright, 2015-03-05:
Awesome, thanks.

I've scheduled a tweet for tomorrow morning, and already shared on Google+

jmwright, 2015-03-05:
BTW, I did notice that the readme text wasn't updated on PyPi. I did change the "Recently Added Features" section a bit to reflect the ongoing work.

jmwright, 2015-03-05:
We haven't had any official releases to the CQ module for FreeCAD yet, but I'm considering do it now and locking the version number to the released version of CQ. Thoughts?

dcowden, 2015-03-05:
sounds good. you've got a very usable product. i do agree its pretty
important that you identfy _somehow_ which CQ version is in the freecad
module version. Locking version numbers is a great way to do that.

jmwright, 2015-03-05:
Done. I tagged v0.1.8 of the FreeCAD module, and have posted to G+ about it's release. I've scheduled another tweet for tomorrow morning about it as well.

closed 2015-03-05 by jmwright

dcowden, 2015-03-06:
awesome thanks for all you are doing!