dcowden/cadquery #80: Feature Request: Bezier Curve Support

jmwright, 2015-03-30:
FreeCAD supports the creation of [Bezier curves](http://www.freecadweb.org/wiki/index.php?title=Draft_BezCurve/en). I believe that I'm going to need support for these in a project relatively soon, so I'm going to post my research on adding this support to CQ here.

This might be extended to [B-Spline](http://www.freecadweb.org/wiki/index.php?title=Draft_BSpline/en) support as well depending on what I find in my research.

jmwright, 2015-04-03:
I did a quick experiment based on the doc [here](http://www.freecadweb.org/wiki/index.php?title=Topological_data_scripting/en#Creating_a_Bezier_curve) this morning.

I wrote the following code as a test based on something that I've done previously.

```python
import cadquery as cq
from Helpers import show
import Part
from FreeCAD import Vector as v

def makeBCurveEdge(Points):
    geomCurve = Part.BezierCurve()
    geomCurve.setPoles(Points)
    edge = Part.Edge(geomCurve)
    return(edge)

# The points for the CQ polyline
spike_points = (
    (7.32802871061466, 11.007605598221),
    (7.31058536987926, 11.0165328678702),
    (7.29209592126546, 11.0270071060423),
    (7.27238894084329, 11.039277065777),
    (7.25129565196959, 11.0536201349458),
    (7.22864456252685, 11.0703473914655),
    (7.20425764118421, 11.0898088975634),
    (7.17794732182628, 11.1123994657636),
    (7.14951394663526, 11.1385650784009),
    (7.11874341758329, 11.1688101281429),
    (7.08540490883067, 11.2037056523581),
    (7.04924853649798, 11.243898752355),
    (7.0100029052478, 11.2901234170726),
    (6.96737246182478, 11.3432130094645),
    (6.92103458858197, 11.4041147236265),
    (6.87063636726188, 11.473906383781),
    (6.81579093586704, 11.5538160356454),
    (6.7560733495435, 11.6452448807842),
    (6.69101583966025, 11.749794231057),
    (6.6201023428587, 11.8692973209945),
    (6.5427621424216, 12.0058570212954),
    (6.45836242590977, 12.1618907607559),
    (6.3661995128761, 12.3401843060196),
    (6.26548844074251, 12.5439564949569),
    (6.1553505102395, 12.7769376068103),
    (6.03479827661213, 13.0434648316541),
    (5.90271731838607, 13.3485993456133),
    (5.75784390653807, 13.6982709102955),
    (5.59873741121177, 14.099457845082),
    (5.42374588805677, 14.5604128895892),
    (5.23096273330545, 15.0909492085831),
    (5.01817151248803, 15.7028060883286),
    (4.78277493953884, 16.4101214927169),
    (4.52170233477736, 17.230049779283),
    (4.23128744096091, 18.1835794158505),
    (3.90710476888003, 19.2966305755316),
    (3.54374691517922, 20.6015511662541),
    (3.13451623969209, 22.1391909699118),
    (2.67098960228534, 23.9618326248229),
    (2.14239030043114, 26.1374234847559),
    (1.53465850644412, 28.7558385724394),
    (0.829027589566102, 31.9384397404038),
    (0, 35.8524021267837)
)

# First set of points for FreeCAD "BCurve"
spline_points = [
    v(7.32802871061466, 11.007605598221),
    v(7.31058536987926, 11.0165328678702),
    v(7.29209592126546, 11.0270071060423),
    v(7.27238894084329, 11.039277065777),
    v(7.25129565196959, 11.0536201349458),
    v(7.22864456252685, 11.0703473914655),
    v(7.20425764118421, 11.0898088975634),
    v(7.17794732182628, 11.1123994657636),
    v(7.14951394663526, 11.1385650784009),
    v(7.11874341758329, 11.1688101281429),
    v(7.08540490883067, 11.2037056523581),
    v(7.04924853649798, 11.243898752355),
    v(7.0100029052478, 11.2901234170726),
    v(6.96737246182478, 11.3432130094645),
    v(6.92103458858197, 11.4041147236265),
    v(6.87063636726188, 11.473906383781),
    v(6.81579093586704, 11.5538160356454),
    v(6.7560733495435, 11.6452448807842),
    v(6.69101583966025, 11.749794231057),
    v(6.6201023428587, 11.8692973209945),
    v(6.5427621424216, 12.0058570212954),
    v(6.45836242590977, 12.1618907607559),
    v(6.3661995128761, 12.3401843060196),
    v(6.26548844074251, 12.5439564949569),
    v(6.1553505102395, 12.7769376068103),
    v(6.03479827661213, 13.0434648316541)
]

# Second set of points for FreeCAD "BCurve"
spline_line_2 = [
    v(6.03479827661213, 13.0434648316541),
    v(5.90271731838607, 13.3485993456133),
    v(5.75784390653807, 13.6982709102955),
    v(5.59873741121177, 14.099457845082),
    v(5.42374588805677, 14.5604128895892),
    v(5.23096273330545, 15.0909492085831),
    v(5.01817151248803, 15.7028060883286),
    v(4.78277493953884, 16.4101214927169),
    v(4.52170233477736, 17.230049779283),
    v(4.23128744096091, 18.1835794158505),
    v(3.90710476888003, 19.2966305755316),
    v(3.54374691517922, 20.6015511662541),
    v(3.13451623969209, 22.1391909699118),
    v(2.67098960228534, 23.9618326248229),
    v(2.14239030043114, 26.1374234847559),
    v(1.53465850644412, 28.7558385724394),
    v(0.829027589566102, 31.9384397404038),
    v(0, 35.8524021267837)
]

# The original polyline with many segments
spike_line = cq.Workplane('XY').polyline(spike_points)

# The two segment BCurveline
edge = makeBCurveEdge(spline_points)
edge2 = makeBCurveEdge(spline_line_2)

Part.show(edge)
Part.show(edge2)

show(spike_line)
```
Apparently, `geomCurve.setPoles(Points)` will only take 26 points max, so I had to break the curve down into two segments.

There are actually two separate curves in the following screenshot, but it's hard to tell because they're so close together. Even when I zoom in, the separation is only micrometers at the widest part.

[image]

This is actually what I was looking for. Something that I could feed a bunch of points into to get a smooth curve that's pretty true to the original polyline. Normally when you think of Bezier curves you think of control points, not creating a line through a bunch of points, so I'm thinking about changing the name and scope of this feature request.

dcowden, 2015-04-03:
dude that's awesome. i dont even want to know how long it took to find out
the magic limit was 26 points :)

jmwright, 2015-04-03:
It wasn't too bad, but there was definitely trial and error involved. I'm not sure where that 26 point limit comes from. I'm going to check with the FreeCAD devs to see if they can tell me. It would be ideal if it took an unlimited number of points, but I'm guessing that's really hard to implement on something like this.

@dcowden Any thoughts on the best way to integrate this operation? I'm thinking to maybe mirror [polyline](https://github.com/dcowden/cadquery/blob/73688899aa7d766847e9b258de49e30248e878e6/cadquery/CQ.py#L1602), and maybe just call it bcurve or something. Seems like I should be returning a wire like polyline does instead of an edge too.

dcowden, 2015-04-03:
yes i think polyline is a good starting point. in this case you have two
point sets.  it might be nice to accept the two sets of points, and then
slice them into 26- point blocks for the user, so they dont have to worry
about it.   I agree that returning a wire vs an edge would be best-- that's
what you need to do in order for the wire to then be part of other shapes.

jmwright, 2015-04-03:
The polyline code has the following comment in the docstring.

> Future Enhacement:
>        This should probably yield a list of edges, not a wire, so that
>         it is possible to combine a polyline with other edges and arcs

Should I return a wire, or list of edges with bcurve?

dcowden, 2015-04-03:
hmm. yeah i forgot about that. How often do you think people will add a
bcurve onto other polylines and stuff?

for example, would we want this to work?

lineTo(2.0,3.0).vLine(5.0).bCurve([pointlist1],[pointlist2]).lineTo(3.0,3,0)

jmwright, 2015-04-03:
That's actually exactly how I'm planning to use it.

dcowden, 2015-04-03:
oh ok. in that case, if memory serves, you would want to push the end point
of the bcurve onto the stack at the end of the operation.  Then i think
you'd put a wire onto the list of active wires.

Though my comment suggests using a list of edges ( which may, in fact, be
better),  it would be a lot of work right now, because all of the other
code assumes wires.

jmwright, 2015-04-04:
Based on feedback from the FreeCAD forums [here](http://forum.freecadweb.org/viewtopic.php?f=10&t=10398), I started using Part.BSplineCurve() with it's interpolate function. The curve I had before was passing near instead of trying to pass through the points. I was kind of cheating the Bezier's operation, and the dev said that it was never intended to be used for 26 degrees as I was doing. The new implementation is cleaner, and takes any number of points. This won't change the high level implementation in CQ, but will greatly reduce the complexity of the function's internals. I've changed the name of this issue to more accurately reflect what this feature is going to be.
```python
import cadquery as cq
from Helpers import show
import Part
from FreeCAD import Vector as v

# The points for the CQ polyline
spike_points = (
    (7.32802871061466, 11.007605598221),
    (7.31058536987926, 11.0165328678702),
    (7.29209592126546, 11.0270071060423),
    (7.27238894084329, 11.039277065777),
    (7.25129565196959, 11.0536201349458),
    (7.22864456252685, 11.0703473914655),
    (7.20425764118421, 11.0898088975634),
    (7.17794732182628, 11.1123994657636),
    (7.14951394663526, 11.1385650784009),
    (7.11874341758329, 11.1688101281429),
    (7.08540490883067, 11.2037056523581),
    (7.04924853649798, 11.243898752355),
    (7.0100029052478, 11.2901234170726),
    (6.96737246182478, 11.3432130094645),
    (6.92103458858197, 11.4041147236265),
    (6.87063636726188, 11.473906383781),
    (6.81579093586704, 11.5538160356454),
    (6.7560733495435, 11.6452448807842),
    (6.69101583966025, 11.749794231057),
    (6.6201023428587, 11.8692973209945),
    (6.5427621424216, 12.0058570212954),
    (6.45836242590977, 12.1618907607559),
    (6.3661995128761, 12.3401843060196),
    (6.26548844074251, 12.5439564949569),
    (6.1553505102395, 12.7769376068103),
    (6.03479827661213, 13.0434648316541),
    (5.90271731838607, 13.3485993456133),
    (5.75784390653807, 13.6982709102955),
    (5.59873741121177, 14.099457845082),
    (5.42374588805677, 14.5604128895892),
    (5.23096273330545, 15.0909492085831),
    (5.01817151248803, 15.7028060883286),
    (4.78277493953884, 16.4101214927169),
    (4.52170233477736, 17.230049779283),
    (4.23128744096091, 18.1835794158505),
    (3.90710476888003, 19.2966305755316),
    (3.54374691517922, 20.6015511662541),
    (3.13451623969209, 22.1391909699118),
    (2.67098960228534, 23.9618326248229),
    (2.14239030043114, 26.1374234847559),
    (1.53465850644412, 28.7558385724394),
    (0.829027589566102, 31.9384397404038),
    (0, 35.8524021267837)
)

# First set of points for FreeCAD "BCurve""
spline_points = [
    v(7.32802871061466, 11.007605598221),
    v(7.31058536987926, 11.0165328678702),
    v(7.29209592126546, 11.0270071060423),
    v(7.27238894084329, 11.039277065777),
    v(7.25129565196959, 11.0536201349458),
    v(7.22864456252685, 11.0703473914655),
    v(7.20425764118421, 11.0898088975634),
    v(7.17794732182628, 11.1123994657636),
    v(7.14951394663526, 11.1385650784009),
    v(7.11874341758329, 11.1688101281429),
    v(7.08540490883067, 11.2037056523581),
    v(7.04924853649798, 11.243898752355),
    v(7.0100029052478, 11.2901234170726),
    v(6.96737246182478, 11.3432130094645),
    v(6.92103458858197, 11.4041147236265),
    v(6.87063636726188, 11.473906383781),
    v(6.81579093586704, 11.5538160356454),
    v(6.7560733495435, 11.6452448807842),
    v(6.69101583966025, 11.749794231057),
    v(6.6201023428587, 11.8692973209945),
    v(6.5427621424216, 12.0058570212954),
    v(6.45836242590977, 12.1618907607559),
    v(6.3661995128761, 12.3401843060196),
    v(6.26548844074251, 12.5439564949569),
    v(6.1553505102395, 12.7769376068103),
    v(6.03479827661213, 13.0434648316541),
    v(5.90271731838607, 13.3485993456133),
    v(5.75784390653807, 13.6982709102955),
    v(5.59873741121177, 14.099457845082),
    v(5.42374588805677, 14.5604128895892),
    v(5.23096273330545, 15.0909492085831),
    v(5.01817151248803, 15.7028060883286),
    v(4.78277493953884, 16.4101214927169),
    v(4.52170233477736, 17.230049779283),
    v(4.23128744096091, 18.1835794158505),
    v(3.90710476888003, 19.2966305755316),
    v(3.54374691517922, 20.6015511662541),
    v(3.13451623969209, 22.1391909699118),
    v(2.67098960228534, 23.9618326248229),
    v(2.14239030043114, 26.1374234847559),
    v(1.53465850644412, 28.7558385724394),
    v(0.829027589566102, 31.9384397404038),
    v(0, 35.8524021267837)
]

# The original polyline with many segments
spike_line = cq.Workplane('XY').polyline(spike_points)

curve = Part.BSplineCurve()
curve.interpolate(spline_points)

Part.show(curve.toShape())
show(spike_line)
```

jmwright, 2015-04-27:
Not sure how I missed it before, but CadQuery already has a CQ.spline function that does an interpolated B-Spline. It just doesn't act like the FreeCAD code above. If I can figure out why, I should have what I need.

jmwright, 2015-04-27:
Looks like I just wasn't using it right. Mach 30 has a project coming up that will put CadQuery's line, polyline, and spline functions through their paces. Unless that turns up something, this issue is resolved.

closed 2015-04-27 by jmwright