dcowden/cadquery #81: Lego Example Questions

jmwright, 2015-04-06:
@dcowden I've started looking through the Lego brick example so that I can do a walk-through video, and have a few questions. 

1. The `thickness` variable isn't used. Was that intentional, or an oversight? It seems like this needs to be a boolean `thick` that switches the `height` between 3.2 and 9.6 mm. Here's what I would suggest switching it to:

    ```python
    thick = False     # Whether this is a thick or thin brick
    ...
    height = 9.6 if thick else 3.2  # Use our boolean to switch thickness
    ```
2. What does the `t` variable stand for in the following lines?

    ```python
    t = (pitch - (2 * clearance) - bumpDiam) / 2.0
    postDiam = pitch - t  # works out to 6.5
    ```
3. Are there any concerns about Lego getting upset with us for modelling their product with measurements included?

dcowden, 2015-04-06:
Are you looking at the version on pparts.com  or in the cadquery freecad
module version?

No concerns with lego. Other companies make exact duplicates at this point,
and the patents expired long ago

jmwright, 2015-04-06:
> Are you looking at the version on pparts.com  or in the cadquery freecad
module version?

The CadQuery FreeCAD module version.

dcowden, 2015-04-06:
ok thanks. i thought i cleaned that one up, but let me have a look tonight.

closed 2015-05-27 by jmwright

galou, 2015-10-04:
Corrected in https://github.com/jmwright/cadquery-freecad-module.

jmwright, 2015-10-05:
Thanks @galou !