dcowden/cadquery #83: polyline.mirrorX and mirrorY Don't Work as Expected

jmwright, 2015-04-07:
Had what looks like maybe a bug in CQ or the FreeCAD library posted to the CQ module's issue tracker.

https://github.com/jmwright/cadquery-freecad-module/issues/51

This needs to be investigated to verify that it is a bug and not normal behavior that's non-intuitive. If it's a bug, we'll have to figure out how to address it.

jmwright, 2015-04-13:
The bulk of the work on this is happening over on the CadQuery module's version of this issue.

jmwright, 2015-04-23:
The bug in the use of mirrorX that sparked the issue on the module has been resolved in commit 6a2bc2c98c1816e8f7fc1d01992ac43a5e7a63a2

There is still work to do since chaining mirrorX and mirrorY don't work. This has been a known issue for a long time.

jmwright, 2015-10-23:
I thought that maybe the recent change to the polyline function would have made the resultant wires work better with FreeCAD when chaining mirrorX().mirrorY(), but it did not help. This bug persists. The only resolution for this is for me to set up a development version of FreeCAD and step through the chain of calls until I find the problem.

RustyVermeer, 2017-12-15:
I tried chaining a mirrorX and mirrorY operation after applying the [face validation fix](https://github.com/jmwright/cadquery-freecad-module/issues/111). 

This is a screenshot of a slightly modified Ex011 script, which works as I would expect it to. 
[image]

Compared to the result without the validation fix:

[image]

Perhaps this is too simple a test case to say that this issue is fixed, but this seems promising, at least.

jmwright, 2017-12-15:
@RustyVermeer Does the following example work with your modification? I can't get it to work even with your merge to master integrated into the workbench.
```python
from cadquery import Workplane

r = 20
d1 = 5
d2 = 8
s = 7
t = 1.5
H = 10

points = [
    (r/2-1.5*t, r/2-t),
    (s/2, r/2-t),
    (s/2, r/2),
    (r/2, r/2),
    (r/2, s/2),
    (r/2-t, s/2),
    (r/2-t, r/2-1.5*t),
    (t/2, 0)
    ]

result = Workplane("front").moveTo(0, t/2).polyline(points).mirrorX().mirrorY().extrude(1.0)

show_object(result)
```

RustyVermeer, 2017-12-17:
Hmm... no luck on my end either.
Tried messing around with a few things but haven't reached a conclusion yet either.

jmwright, 2017-12-17:
In the testing I did when this issue first cropped up, I could see that a duplicate edge is added somehow. I guess validating the face doesn't fix the extra edge.

RustyVermeer, 2017-12-17:
Found this:
```
from cadquery import Workplane

r = 20
d1 = 5
d2 = 8
s = 7
t = 1.5
H = 10

points = [
    (0, t/2), #Instead of having this point in a 'moveTo', place it here
    (r/2-1.5*t, r/2-t),
    (s/2, r/2-t),
    (s/2, r/2),
    (r/2, r/2),
    (r/2, s/2),
    (r/2-t, s/2),
    (r/2-t, r/2-1.5*t),
    (t/2, 0),
    ]

result = Workplane("front").polyline(points).mirrorX().mirrorY().extrude(1.0)

show_object(result)
```

[image]

You can then swap the mirrorX and Y around, that order doesn't matter. 
But, a mirrorX  alone WILL work while a mirrorY alone will not.

Not sure if I'm discovering something that is already known about that.

My current thoughts are that this bug relates in some way to the fact that the mirror functions return _wires_ and not edges. 

An open wire cannot be closed with the cadquery .close() function (as far as I can tell), but FreeCAD is able to extrude open wires, as it can result in a surface feature. So, if there's something that prevents FreeCAD from properly closing wires, it just happily extrudes them into surface features, which is an unintended behaviour.

That is a bit of a guess, though. I haven't looked too thoroughly at the mirror capability yet.

dcowden, 2017-12-17:
I think that's a good guess. 

What do you guys think about the idea of reworking the code so that everything returns edges, and then closing hem and processing them into wires at the time an extrude happens?

Originally, I didn't do it because it was hard. But now that the dxf code has node matching and ordering code, it would probably be much more reliable to just throw all of the edges I to a list and build closed wires at the end

jmwright, 2017-12-17:
> What do you guys think about the idea of reworking the code so that everything returns edges, and then closing hem and processing them into wires at the time an extrude happens?

I'm ok with that as long as we can do it in a way that doesn't break existing scripts.

tolot27, 2019-06-01:
Just no note, the following simple example still does not work:

```python
r = cq.Workplane("XY" ).box(120, 120, 10)
pt_cross = [ (0, 49), (28, 49), (28, 28), (49, 28), (49, 0) ]
r = r.faces(">Z").workplane().polyline(pt_cross).mirrorY().mirrorX().extrude(2)

show_object(r, options={"rgba":(204, 204, 204, 0.0)})
```

If I aappend `(0, 0)` to pt_cross, one quarter of the cross is drawn, but nothing is mirrored. I had to switch to CadQuery to get it working.

jmwright, 2019-06-02:
> I had to switch to CadQuery to get it working.

When you say "CadQuery", do you mean CadQuery 1.0 (the FreeCAD-based version)?

tolot27, 2019-06-02:
Sorry, I meant CQ 2.0.