dcowden/cadquery #85: Can we export STL files with arbitrary resolution?

jgrizou, 2015-04-23:
In OpenScad, we can change the [%fn variable](http://en.wikibooks.org/wiki/OpenSCAD_User_Manual/The_OpenSCAD_Language#.24fa.2C_.24fs_and_.24fn) to generate more or less detailled STL files. Is there a way to do this in cadquery? 

This feature is necessary to export parts usable once 3D printed.

jmwright, 2015-04-23:
On first glance at the underlying FreeCAD exportStl function, it doesn't appear so. I'll look into it more though.

dcowden, 2015-04-23:
Thanks for chiming in, Jeremy.  I'm pretty sure you are right-- the Freecad
function doesnt provide tolerance.

It  might not be _too_ difficult to achieve, however.  in
https://github.com/dcowden/cadquery/blob/master/cadquery/freecad_impl/exporters.py
,

we have code that tessellates an object to a specified tolerance.  That
function is used to produce a json mesh or AMF.  It performs fast enough to
render models with satisfactory performance on parametricparts.com, so i
bet it could produce STL pretty fast as well.

The code to create the mesh is already there, and it woudl be trival to
replace the freecad function with our version instead.

The only missing part is taking a tesselation object and creating STL. this
is probably not very hard. There's a template in AMFWriter, which is a very
similar format. The only part that would be a little tricky is computing
the direction of the normal vector in the output.

jmwright, 2015-04-23:
@jgrizou I had my first design ever 3D printed just two weeks ago, so I'm still learning a lot. Can you explain why you would want anything other than the highest quality setting for an STL when it's being printed? Also, several software packages give you a choice of ASCII or binary when you export your STL. Is that option going to be needed eventually as well?

dcowden, 2015-04-23:
@jmwright i can answer that one.

the short answer is time.  The highest quality setting can produce a huge,
huge file-- which takes forever to slice and process.  You are not always
printing on the same printer, so you generally dont want to waste your time
when the accuracy doesnt matter.  So for example, my printer can hold a
0.005" tolerance at best.  so generating a 40M file with 0.001 tolernace is
a waste of time.

Also, sometimes, the geometry is such that you just dont care all that much
about the accuracy.  you'd rather have a quick build.

jmwright, 2015-04-23:
> the short answer is time...you generally dont want to waste your time when the accuracy doesnt matter

@dcowden - Thanks, that's great to know. Is this a change that you have the interest and time to take on? It would take me a long time to get up to the level of understanding you're already at.

dcowden, 2015-04-23:
sorry, do not have the time :(

But i'll make you a deal-- if you do the coding i'll be happy to at least
take the time to answer all your questions promptly. I always have 10
minutes but i never have 2 hours, it seems!

jmwright, 2015-04-24:
This will have to be about 3rd or 4th on my to-do list for CadQuery.

1. Finish debugging the mirrorX().mirrorY() chaining problem. This may take awhile since it seems like there may be an underlying FreeCAD issue.
2. BSpline implementation. I think Mach 30 is going to need this sooner rather than later. This should be a quicker feature to implement though.
3. STL output quality setting implementation (this feature).
4. Sweep operation. Also something Mach 30 is going to need.

jgrizou, 2015-04-24:
Thanks for your quick answer. I agree with @dcowden, sometime slicing takes forever. Adding this option to FreeCAD would be nice but maybe not a top priority if it works in python.

I am really new to cadquery. I saw the tolerance parameter you mentionned. I tried to play with it but could not generate STL files of different quality. What does this tolerance parameters means?

I tried:
```
from cadquery import *
cylinder = Workplane("XY").circle(10).extrude(10)
with open('res10.stl', 'w') as f:
    f.write(exporters.toString(cylinder, 'STL', 10))
f.closed
```
I exported with different tolerances (100,10,1,0.1,0.01) in separate files, but when looking at the resulting 3D parts I could not see any visual difference. The files where the same in fact.

---
I have been developing simple OpenScad library ([robotis-scad](https://github.com/jgrizou/robotis-scad), [segment-scad](https://github.com/jgrizou/segment-scad)) that I want to share online in a browser but its not easy with OpenScad. Then, I saw cadquery and the very nice http://parametricparts.com/ (exactly what I was looking for!). Changing the resolution of the export would be a nice feature on that website too.

Again I am new to cadquery and I might be missing many points here..

dcowden, 2015-04-24:
yeah, the tolerance parameter is unforunately ignored for STL output,
because we're using the FreeCAD function, which doesnt accept it.  In
defense of the code, the parameter makes sense when you do AMF or a
three.js. mesh.

Just a thought-- does your slicer support AMF?  AMF is a much better format
-- in fact its 'STL 2.0'.  In addition to being more accurate and less
verbose, the exporter will honor the tolerance parameter for it, and its
available right now.

Many slicers support AMF at this point.  Give it a try if yours does.

jmwright, 2015-04-25:
@jgrizou Do you have an immediate need for this feature, or are you just noting that CadQuery needs to have it in the future?

jmwright, 2015-04-27:
Ok, so I've gotten back around to this issue. I'll take a look at the exporter code that @dcowden mentioned, and start experimenting.

jgrizou, 2015-04-28:
I tried AMF with Cura, it works. But it is not supported by makerbot for example. But my personnal case does not matter, I am looking for a way to share my designs with people that might have various 3D printers or even use online services. STL is still a widely use format.

Thanks for your time and quick acting guys. @jmwright I do not have an immediate need for this feature, I was asking if it was possible. Yet, I think it is a useful option for the future.

jmwright, 2015-04-28:
@jgrizou Thanks for the suggestion. We count on the community to help us guide CadQuery's development process.

jmwright, 2015-08-04:
I just had some experience doing this from within FreeCAD, which doesn't directly support resolution settings on STL files. https://opendesignengine.net/boards/50/topics/954?r=1020#message-1020

You can convert a solid to a mesh in the Mesh Workbench and tweak the conversion settings. 

I'm not advocating for this over what we've talked about above, but it might be an interesting option.

dcowden, 2015-08-08:
I think we should just implement our own stl generator.

We already have the framework, which supports creation of the AMF format.
That code generates a triangular mesh, which does support a resolution, so
all that's needed is to generate stl instead of amf. Most of the code
should be the same.

Of course as usual I'm all talk and no action. I don't have the time to do
it :(

jmwright, 2015-08-08:
@dcowden - Do you have a time estimate for how long it will take? Do you still have time to offer guidance?

dcowden, 2015-08-08:
Always time to offer guidance. I always have 10 minutes but never a couple
hours to code :(

Since most of the code is already written and stl is easy to produce, I
would guess it would take about 6-8 hours to do. That would include writing
some tests, which if I recall are lacking for the amf generator.

I can provide some more specifics on how I would do it when I get back home

dcowden, 2015-08-08:
Ok I went and looked a bit to remind myself of how I did it for AMF files.
I think here's what the recipe would be.

All of the changes would be here:

https://github.com/dcowden/cadquery/blob/master/cadquery/freecad_impl/exporters.py

(1) add elif exportType == ExportTypes.STL block after line 87, and remove
the freecad based implementation for exporting STL on lines 99 and 100.

(2) add a class called STLWriter, which is very similar to AMFWriter.  It
should take a tessellation in its constructor ( like AmfWriter  does).

(3) STLWriter needs a writeStl method(fileLike, binary=true). But instead
of using element tree to generate AMF XML output, we create STL. the binary
argument toggles STL binary mode on/off . The resolution is already
specified when the tessellation is created.

Of course creating STL in python has already been done many times. Here's
one example:

https://github.com/thearn/stl_tools/blob/master/stl_tools/numpy2stl.py

the only challenging bit of code is to do the same logic using the
tessellation as an input instead of a numpy array as the above does.

If the above were all there is to it, i'd estimate about 3 hours.

There is a 'GOTCHA', though, with STL generation.

STL, repeats the vertices for each face, which makes the file larger, but
also introduces the need to pay particular attention to numerical rounding
and formatting.

If a given vertex is the same, all three of its coordinates must be output
in the file EXACTLY the same. Otherwise, slight rounding errors will result
in 'holes' in the mesh. For example:
1.99999999999  2.000000000 are not the same, and will create issues when
used to represent the same point.

AMF is a much nicer format because each vertex is created exactly once
time, and then used by reference in each face, like so in concept:
 V1 = ( 1.0, 2.0, 1.0 )
V2 = (2.0,2.0,2.0)
V3 = ( 1.0,1.0,1.0)
Face1= ( V1, V2, V3 ).

So, in short, when formatting the output, make sure to fix the number of
decimals upfront. This way, even if the original model has slight numerical
rounding errors in it, the resulting mesh will not have holes in it.

But what to choose? Too few, you lose resolution. Too many, and you could
end up creating holes in the mesh when they were not wanted.  One clever
solution might be to use the input tessellation resolution as a guide. For
example, if the resolution is 0.1 units, then 0.01 units should be more
than adequate for the decimal point. Another option would be to fix it very
small always, like 5 decimal points. Its important to round, not truncate.

This saves space in the