dcowden/cadquery #89: _calcTransforms in freecad_impl/geom.py

jmwright, 2015-05-26:
@dcowden - Could I get an explanation of what this function is doing in geom.py, and why it expands the matrix to 4x4? I'm having trouble relating it to it's equivalent in PythonOCC.

dcowden, 2015-05-27:
Hi, 

A 4x4 transformation matrix is commonly used in 3d, because it represents, simultaneously, translation and rotation in 3d space.  You take a 3x1 vector ( and put a 1 at end, like x y z 1 ), and then multiply it by the 4x4 to get a new vector.  

Here's a decent write up:
http://www.euclideanspace.com/maths/geometry/affine/matrix4x4/

jmwright, 2015-05-27:
@dcowden Great, thanks. That's the reference I needed.

closed 2015-05-27 by jmwright

dcowden, 2015-05-27:
no problem! thanks for hacking away!
How is pythonOCC coming?  Are you getting more or less discouraged? I
remember when i was writing my slicer, it took severl months to work
through all the in's and outs of their API. mostly not their fault, since
the wrappers are just swig wrappers, it was mostly OCC itself.

jmwright, 2015-05-27:
It started off really fast, but I've gotten bogged down recently. I went through and ported the classic OCC bottle example over to pythonocc-core 0.16 to help me get more familiar with the API, and that helped a little. Surprisingly, I'm apparently the first person who's done that and offered to contribute it. The example was missing from pythonocc-core.

I'm going to continue to chip away at it, but it will be a long road. I was naively hoping for a quick (nearly) direct port from FreeCAD to PythonOCC, but that's not going to happen. 

I need to circle back around to some issues related to the FreeCAD implementation and the CadQuery module. I'm planning to do that and then return to the PythonOCC implementation when things quiet down again.

dcowden, 2015-05-27:
Yeah, now you can see what i mean about how big a deal porting it really
is.  That's why i abandoned the idea of having a transparent adapter layer
between CQ and FreeCAD and PythonOCC-- the two are just too different.   CQ
needs to use one or the other.

jmwright, 2015-05-27:
Yep, we're on the same page now :-)