dcowden/cadquery #90: add ruled argument to loft

hyOzd, 2015-06-08:
Can you add `ruled` argument to `loft()` function? It allows creating loft with straight edges, instead of a smooth surface.

This is the shape I want to create:
[image]

Without ruled argument it will look like this :(
[image]

This looks like a simple change to me. I already added ruled argument to `FreeCADPart.makeLoft()` call and it worked. I can do a pull request if you agree.

jmwright, 2015-06-08:
@hyOzd I think that will work fine to add a Ruled argument. The only thing I would suggest is to have it default to False at the [Shell.makeLoft level](https://github.com/dcowden/cadquery/blob/17250d9e1c48e04427af42e9570781e84256509e/cadquery/freecad_impl/shapes.py#L619) so that existing scripts will act as they did before.

jmwright, 2015-06-08:
Merged

closed 2015-06-08 by jmwright