dcowden/cadquery #92: line close() problem with workplanes XZ and YZ

hyOzd, 2015-06-12:
Below code works as expected:

    r = cq.Workplane(cq.Plane.XY()).move(5, 2).lineTo(10,0). \
        lineTo(10,10).lineTo(0,10).close().extrude(5)

[image]

But when I use plane XZ this happens:

    r = cq.Workplane(cq.Plane.XZ()).move(5, 2).lineTo(10,0). \
        lineTo(10,10).lineTo(0,10).close().extrude(5)

[image]

With Plane.YZ this:
[image]

It seems that when using a plane other than XY, close() operation takes X or Y coordinate as 0.

jmwright, 2015-06-12:
It seems that the close function just always assumes you are working in the XY plane.

https://github.com/dcowden/cadquery/blob/master/cadquery/CQ.py#L1639

If I change `self.ctx.firstPoint.y` to `self.ctz.firstPoint.z` and then run your XZ example, it works fine.

@dcowden Is there a reason for the assumption of the XY plane, or is this just a bug?

dcowden, 2015-06-12:
i think that's just a bug... :\

jmwright, 2015-06-12:
Is there a better way to handle that then testing what the plane is and then altering which coords get used?

hyOzd, 2015-06-12:
A ha!

```python
    def _addPendingEdge(self,edge):
        """
            Queues an edge for later combination into a wire.

        :param edge:
        :return:
        """
        self.ctx.pendingEdges.append(edge)

        if self.ctx.firstPoint is None:
            self.ctx.firstPoint = self.plane.toLocalCoords(edge.startPoint())
```

This worked!

jmwright, 2015-06-12:
All of the tests run fine and so I've pushed this fix to master. Thanks @hyOzd

jmwright, 2015-06-12:
Fixed in commit 1769eac353469a9023850b7374ace7f557c52e5c

closed 2015-06-12 by jmwright

dcowden, 2015-06-12:
thanks guys for cleaning up my busted code!

hyOzd, 2015-06-12:
Thank you for creating this software! I love it!