dcowden/cadquery #97: how to copy?

hyOzd, 2015-06-20:
Is there a way to create duplicate of an object? I'm creating models with lots of repeating parts. And I think it takes unnecessarily too long.

For example here is a script that I use to create PCB pin header models.

https://bitbucket.org/hyOzd/freecad-macros/src/812a55fdd4457d760bfeea81df88d87175d2c5dd/make_pinheaders.py?at=master

In this script, `make_pin()` function always produces the same shape, only difference is the position. I think being able to `copy` instead of running the loft operation each time, would speed my script.

jmwright, 2015-06-20:
There's no copy function that I'm aware of.

I'm AFK, but what happens if you do the following?

```python
pin1 = make_pin(...)
pin2 = pin1.translate(...)
```

hyOzd, 2015-06-20:
This worked. But it didn't speed my script as I thought :( 

But this is from documentation of `translate()` function:

> WARNING: the underlying objects are modified, not copied.

Actually, I noticed now, there seems to be a conflict in documentation. Just above that warning it says:

> Returns a copy of all of the items on the stack by the specified distance

Which one is correct?

jmwright, 2015-06-20:
I'll have to take a look when I get back in front of a computer. You can usually tell by what the function returns. If it returns a new CQ object, I think that would mean the second comment is correct.

In regards to the speed, can you put a print statement in your make_pin function to make sure that it's only running once for both pin1 and pin2?

jmwright, 2015-06-20:
Also, as I think about it, pin2 should be a copy of pin1 at that point, so pin1 should be safe from modification. In any case, we need to fix the documentation.

hyOzd, 2015-06-21:
> In regards to the speed, can you put a print statement in your make_pin function to make sure that it's only running once for both pin1 and pin2?

I did that; there is a significant improvement using `translate`:

    Pin1 in 0.011089
    Pin2 in 0.001097

jmwright, 2015-06-21:
> Pin1 in 0.011089
> Pin2 in 0.001097

That looks like an encouraging improvement to me. Is it good enough?

The FreeCAD implementation's  [translate function](https://github.com/dcowden/cadquery/blob/08791e1e31e0e7dabef5cf901eb4be0adc75df67/cadquery/freecad_impl/shapes.py#L244) (which is used by CQ.translate) makes a copy before applying its operation, so the original is not modified. This also reminded me of another way to make a copy of an object. You can make a copy of the wrapped object, and then create a new CQ object again from that. Making a copy that way is pretty ugly in an end-use script though.

If the original object had been modified/mutated, I would have expected the following code to work, but it doesn't.
```python
import cadquery as cq
from Helpers import show

r = cq.Workplane('XY').box(5, 5, 5)
r.translate((0, 0,20))

# Use the following to render your model with grey RGB and no transparency
show(r, (204, 204, 204, 0.0))
```

This code, however, does work because it makes use the returned copy.

```python
import cadquery as cq
from Helpers import show

r = cq.Workplane('XY').box(5, 5, 5)
r = r.translate((0, 0,20))

# Use the following to render your model with grey RGB and no transparency
show(r, (204, 204, 204, 0.0))
```
I'll get the documentation fixed. Thanks for bringing attention to it.

jmwright, 2015-06-21:
Documentation fixed in commit 292a9935d736a5b97ba826552ae1a728107271c3

hyOzd, 2015-06-21:
>>    Pin1 in 0.011089
>>    Pin2 in 0.001097

> That looks like an encouraging improvement to me. Is it good enough?

That's pretty good. But in the grand scheme of things it didn't matter much. It seems like fillet is taking most of the time in my script. I'm trying to implement chamfer API. Chamfer is good enough for my needs and I'm expecting it to be faster.

By the way, thanks for fixing the doc.

Could you have a look at the `rotate()` function as well? I haven't tried it, but its implementation very similar to `translate()`, and it has the similar text in documentation.

jmwright, 2015-06-21:
Yep, good catch. Fixed in commit abce43b274de2d5e768a80e089e8befb28be3072

Will you be doing a PR for your chamfer implementation? I've been meaning to implement a chamfer operation for a long time, and just haven't been able to get to it.

hyOzd, 2015-06-21:
Most likely. It seems to be the same as `fillet` operation.

jmwright, 2015-06-21:
Yeah, that's my thought too. My plan was to basically copy the fillet implementation and modify it to do the chamfer instead.

hyOzd, 2015-06-22:
> Chamfer is good enough for my needs and I'm expecting it to be faster.

For the record, it turns out chamfer is usually slower than fillet. In my case doing chamfer instead of fillet (highlighted below) was 2x slower.

[image]

I think you can close this issue now.

dcowden, 2015-06-22:
dude that is an _awesome_ model you have there! any chance we could put it
an a cadquery example?

hyOzd, 2015-06-23:
@dcowden you can find the script [here](https://bitbucket.org/hyOzd/freecad-macros/src/60da25069e71e2ee47eeb5bba39d0ef095f6ad47/make_qfp.py?at=master). Final version probably will be a batch script to create many models at once. Similar to [this one](https://bitbucket.org/hyOzd/freecad-macros/src/60da25069e71e2ee47eeb5bba39d0ef095f6ad47/make_pinheaders.py?at=master). But I think, it would be a better idea to add a simplified version as an example script.

jmwright, 2015-06-23:
> For the record, it turns out chamfer is usually slower than fillet. In my case doing chamfer instead of fillet (highlighted below) was 2x slower.

Wow, totally didn't expect that.

closed 2015-06-23 by jmwright

dcowden, 2015-06-23:
awesome, thanks for letting us include this!