dcowden/cadquery #99: sphere is not centered correctly

hyOzd, 2015-07-11:
```python
import cadquery as cq
from Helpers import show

sphere = cq.Workplane("XY").sphere(1, centered=(True, True, True))
show(sphere)
```

Creates this:

[image]

But if I call it with, `centered=(False, False, False)`, sphere is centered correctly. I'm suspecting that freecad's behavior is changed after this API is introduced?

jmwright, 2015-07-13:
Sorry for the slow reply. Thanks for reporting this. Sphere was a fairly recent addition to CadQuery (with FreeCAD 15.x), so I'm guessing it's just a mistake on my part. I'll take a look though.

jmwright, 2015-07-13:
This has been fixed in commit 682294409145c993118940af05b4656a5a23f915

Please let me know if it doesn't work still for you in some cases. I have also updated the FreeCAD module, so this functionality will be fixed there too.

Thanks again for reporting it.

hyOzd, 2015-07-13:
I'm not sure if this is the correct fix. X, Y positions seem to be fixed. But Z did not change.

Also when called with centered=False;

    sphere = cq.Workplane("XY").sphere(1, centered=(False, False, True))

output doesn't change at all.

jmwright, 2015-07-14:
> But Z did not change.

Looking at the vertex you mentioned in the first post, the Z changes for me when running variations of your code snippet above. With `centered=True` Z is 0.0, and with `centered=False` Z is 1.0. Is it not changing for you, or is this behavior different than you expected?

hyOzd, 2015-07-15:
> With centered=True Z is 0.0, and with centered=False Z is 1.0. Is it not changing for you, or is this behavior different than you expected?

That's the case for me as well. The vertex that I highlighted in above screenshot is not the center point of the sphere, it's the *top point* of it. So when `centered=(X,X,True)` I expect it to be in `Z=1` position but when  `centered=(X,X,False)` I expect it to be in `Z=2` position.

jmwright, 2015-07-17:
Ok, for some reason it was hard to wrap my head around this one. It seems to work intuitively now, and should work properly in other planes as well. Commit f42a63c7d1f02f0f9f0f60ae651ba9fc146d2425

Please let me know if it's working as expected now.

hyOzd, 2015-07-17:
Ok. Z is working fine. But X/Y is broken. X and Y are always centered, regardless of `centered` arguments.

jmwright, 2015-07-17:
Work you expect the X and Y positions to be (2.0, 2.0, X) for (False, False, X)?

hyOzd, 2015-07-17:
I hope this diagram will clear things up:

[image]

jmwright, 2015-07-17:
Thanks for the diagram, that did help. Please check the latest commit. 14262e744dbeeaafcf5c5bdc7a8983a45fa82e40

I've also kept the FreeCAD module updated, so it should have this fix.

hyOzd, 2015-07-17:
Finally, it all works as expected. Thanks for the fix.

jmwright, 2015-07-17:
Thanks for working through it with me. I haven't had a concentrated amount of time to sit down with this, so the diagram helped me identify quickly what needed to be done before I got interrupted again.

closed 2015-07-17 by jmwright

hyOzd, 2015-07-17:
You are welcome. I like drawing :)