dcowden/cadquery #100: fillet fails with combined shape

hyOzd, 2015-07-16:
Sample code to reproduce:

```python
import cadquery as cq
from Helpers import show

b = cq.Workplane("XY").box(10,10,2,centered=(False,False,False)).\
    union(cq.Workplane("YZ").box(2,10,10,centered=(False,False,False)))
    
BS = cq.selectors.BoxSelector
b = b.edges(BS((-10, -1, -1),(10,1,1))).fillet(3)

show(b)
```

Error:
```
Running the Python command 'CadQueryExecuteScript' failed:
Traceback (most recent call last):
  File "/home/heyyo/.FreeCAD/Mod/CadQuery/Gui/Command.py", line 178, in Activated
    imp.load_source('temp.module', tempFile.name)
  File "/tmp/tmpKBl3gr", line 13, in <module>
    b = b.edges(BS((-10, -1, -1),(10,1,1))).fillet(3)
  File "/home/heyyo/.FreeCAD/Mod/CadQuery/Libs/cadquery/CQ.py", line 794, in fillet
    s = solid.fillet(radius, edgeList)
  File "/home/heyyo/.FreeCAD/Mod/CadQuery/Libs/cadquery/freecad_impl/shapes.py", line 813, in fillet
    return Shape.cast(self.wrapped.makeFillet(radius, nativeEdges))

BRep_API: command not done
```

If I set the fillet radius below `2.0` (thickness of the shape) it works. But it fails when `r>=2.0`. Why this happens? I believe it's these lines. Below screenshot is before fillet.

[image]

Result of the union operation is not clean. It contains edges from both shapes that shouldn't exist anymore. This is a limitation of FreeCAD. In a similar situation fillet tool will fail. But FreeCAD has a menu item "Refine Shape", that removes these kind of edges and generally cleans up the shape. It is in the menu of "Part" workbench. Actually if you have these options (see below) enabled; it will automatically refine the shape after each operation.

[image]

I would expect cadquery to have a similar option, maybe have this behavior enabled by default.

dcowden, 2015-07-16:
I guess we can look to see what exactly FreeCAD does when those checkboxes
are marked, and then do whatever they do.

I've really struggled with this sort of thing-- FreeCAD ( and the
underlying OpenCascade ) is simply not a bulletproof kernel. When i was
writing CQ, and also on a couple of pythonOCC project, it was pretty common
to have to 'clean up' from the results of operations that should have
worked right the first time.

This leaves me wondering, though, if this is a FreeCAD issue or an
OpenCascade issue.  I would be interested to see if a direct pythonOCC
implementation of this same operation results in the same broken solid or
not.

hyOzd, 2015-07-16:
This: http://www.freecadweb.org/wiki/index.php?title=Part_RefineShape/en#Scripting ?

dcowden, 2015-07-16:
haha! of course, shape.removeSplitter(), just as i would have guessed :)

yeah, if that magically fixes things, we could run it by default after
operations, as long as its not too expensive... that shouldnt be too
difficult to add.

hyOzd, 2015-07-16:
This fixed my problem:

```diff
diff --git a/cadquery/freecad_impl/shapes.py b/cadquery/freecad_impl/shapes.py
index e0f2ecf..6f54848 100644
--- a/cadquery/freecad_impl/shapes.py
+++ b/cadquery/freecad_impl/shapes.py
@@ -800,7 +800,7 @@ class Solid(Shape):
         return Shape.cast(self.wrapped.cut(solidToCut.wrapped))
 
     def fuse(self, solidToJoin):
-        return Shape.cast(self.wrapped.fuse(solidToJoin.wrapped))
+        return Shape.cast(self.wrapped.fuse(solidToJoin.wrapped).removeSplitter())
 
     def fillet(self, radius, edgeList):
         """
```

I can do a pull requests. But I'm not sure if this is the right way. Maybe there should be another method named `refine` or `removeSplitter`? To be honest I don't see a use case for this. Maybe performance?

dcowden, 2015-07-16:
I think this implementation is fine.

The negative is that pretty much every solid operation needs to do this
same thing.  And ( if performance is worse) then very operation would also
need a flag like "clean=true|false" to disable it.

Its annoying to have to add removeSplitter() to every operation and an
optional argument to every operation, but i dont think there's a way to
factor the code to make it any better.

So the question is: what other solid operations could possibly result in
the need to call the cleanup? should we play that game, or should we just
add it to most operations with the idea that its not too expensive?

jmwright, 2015-07-17:
It would be interesting to time the execution to get a benchmark of how much time is lost. As long as it's not more than 2% loss, it might be worth it.

hyOzd, 2015-07-17:
> It would be interesting to time the execution to get a benchmark of how much time is lost. As long as it's not more than 2% loss, it might be worth it.

I plan to do that exactly. Tomorrow...

hyOzd, 2015-07-18:
Hi, I did run some quick and dirty benchmarks. It seems that, usage of `removeSplitter()` can make a significant impact on processing time depending on the shape or workflow. Sometimes good sometimes bad :)

First I benchmarked this:

```python
import cadquery as cq

def make():
    b = cq.Workplane("XY").box(10,10,10, centered=(False, False, False))

    for i in range(50):
        b= b.union(cq.Workplane("XY").workplane(offset=i).\
                   box(10,10,10, centered=(False, False, False)))

    return b

if __name__ == "temp.module":
    from Helpers import show
    show(make())

if __name__ == '__main__':
    make()
```

It creates a shape like this (`removeSplitter`=OFF):
[image]

When `removeSplitter`=ON, this script creates a simple box like shape as you would expect:
[image]

By removeSplitter=ON/OFF, what I mean is `fuse()` method includes a call to `removeSplitter()` or not. So it's called in each step when it's ON. Which one was faster in this case? Interestingly the case of `removeSplitter`=ON was much faster:

`removeSplitter`=OFF : 0m4.328s
`removeSplitter`=ON : 0m1.240s

Why did this happen? Because at each step shape is simplified, so that next fuse operation is always between two simple box shapes instead of a shape that is getting more complex. This shows us that there might be actaully an advantage of using `removeSplitter()` function.

Then I tried this:
```python
import cadquery as cq

def make():
    b = cq.Workplane("XY").box(1000,10,10, centered=(False, False, False))

    for i in range(50):
        b= b.union(cq.Workplane("XY").center(i*20,0).\
                   box(10,10,20, centered=(False, False, False)))

    return b

if __name__ == "temp.module":
    from Helpers import show
    show(make())

if __name__ == '__main__':
    make()
```

It creates this shape, `removeSplitter`=OFF:
[image]

When `removeSplitter`=ON:
[image]

Timing results:
`removeSplitter`=OFF : 0m4.964s
`removeSplitter`=ON : 0m10.052s

As you can see, usage of `removeSplitter()` doubled the processing time!

### Adding `clean()` method
Then I wanted to see if adding a `clean()` method to cadquery and calling this at the end would be beneficial compared to running `removeSplitter()` at each step.

```diff
diff --git a/cadquery/CQ.py b/cadquery/CQ.py
index 20d190c..22d31db 100644
--- a/cadquery/CQ.py
+++ b/cadquery/CQ.py
@@ -2043,6 +2043,14 @@ class Workplane(CQ):
         else:
             return self.newObject([newS])
 
+    def clean(self):
+        solidRef = self.findSolid(searchStack=True, searchParents=True)
+        if solidRef:
+            t = solidRef.clean()
+            return self.newObject([t])
+        else:
+            raise Exception("There is no solid for cleaning!")
+
     def cut(self, toCut, combine=True):
         """
         Cuts the provided solid from the current solid, IE, perform a solid subtraction

diff --git a/cadquery/freecad_impl/shapes.py b/cadquery/freecad_impl/shapes.py
index e0f2ecf..1d0336f 100644
--- a/cadquery/freecad_impl/shapes.py
+++ b/cadquery/freecad_impl/shapes.py
@@ -802,6 +802,9 @@ class Solid(Shape):
     def fuse(self, solidToJoin):
         return Shape.cast(self.wrapped.fuse(solidToJoin.wrapped))
 
+    def clean(self):
+        return Shape.cast(self.wrapped.removeSplitter())
+
     def fillet(self, radius, edgeList):
         """
         Fillets the specified edges of this solid.
```

I have no idea if this implementation is correct or what it does exactly. But it worked :D 

Then I run this slightly modified version of 2nd script:

```python
#!/usr/bin/python
import cadquery as cq

def make():
    b = cq.Workplane("XY").box(10,10,10, centered=(False, False, False))

    for i in range(50):
        b= b.union(cq.Workplane("XY").workplane(offset=i).\
                   box(10,10,10, centered=(False, False, False)))

    return b.clean()

if __name__ == "temp.module":
    from Helpers import show
    show(make())

if __name__ == '__main__':
    make()
```

Only difference is instead of calling `removeSplitter()` at each step, we call it at the end with `clean()` method. This produces the same output but takes only:

0m4.438s

If we compare this to `removeSplitter()` always on case, **0m10.052s**, we can see that it can be really beneficial to have a seperate `clean()` method.

### Conclusion
It's hard to come to a conclusion with just these tests but it gives an idea. 

I think having `removeSplitter()` always ON, is beneficial generally, since it will make resulting shape simpler at each step thus next operation will be cheaper. But if you are generating shapes with lots of features (possibly duplicating features, like the one above) it will cause a serious processing time increase. 

We have 3 options:

1. `removeSplitter()` always on
2. `removeSplitter()` always OFF, adding a `clean()` method
3. add some way to enable `removeSplitter` during runtime, something like; `set(autoclean=True)` method

I'm not sure if it's possible but I would go with the **option 3**, `autoclean/removeSplitter` being ON by default. Reason is; not having it ON can actually break some operations such as fillet operation as described in the first post in this page. It's easy to see why such operation fails when you have a visual tool. Actually I discovered the underlaying reason of this problem quickly, because I already had some trouble with fillet operation using FreeCAD. A user of cadquery who is not familiar with FreeCAD may have some trouble understanding what's wrong. In case of implementing this option, it should be noted in documentation that to increase performance autoclean can be disabled.

dcowden, 2015-07-18:
Wow! Thanks for this awesome analysis!

I agree with your recommendation to add autoclean=True to all methods.

You hint at, but do not give ample credit to your reasoning. Our priorities
must be:

(1) reliable operation
(2) intuitiveness
(3) performance

Given these priorities, option 3 is best. It prevents frustrations like you
had, and it is also the most intuitive.  People will be willing to do
non-intuitive things like setting autoClean=False if they seek to improve
performance, but already have what they want-- but the opposite is not true.

Option 3 will require adding autoClean=true to most of the operations that
could possibly benefit from it. But as you have proven, its a relatively
simple fix, so I think its the right thing.

Would you be willing to make the fix ?

hyOzd, 2015-07-18:
Instead of adding an extra parameter to each method, can we implement it like a setting that you call it once and then it effects the later operations. Example usage would be this;

     b = cq.Workplane("XY").set(autoclean=False).box((10,10,10))

or maybe;

     b = cq.Workplane("XY").autoclean(False).box((10,10,10))

Just an idea. I'm not sure if this will be easy to implement. Can we store this option in `CQContext`?

dcowden, 2015-07-18:
yeah, i like that idea. the 2nd form is better, i think. althought
set(authclean=False) contemplates future options, i think even if other
options exist, they would still be more clear chained on their own:

cq.Workplane("XY").autoclean(False).smudge(True)

hyOzd, 2015-07-18:
> cq.Workplane("XY").autoclean(False).smudge(True)

Okay, I will work on this one.

But I'm not sure about the word "clean". It may imply removing something, like "clear the stack", "clear the drawing" instead of "remove unwanted edges". Some ideas instead of "clean":

* cleanFaces / autoCleanFaces
* cleanEdges / autoCleanEdges
* simplify / autoSimplify
* freecad's way: removeSplitter / autoRemoveSplitter

What does @jmwright think about this?

dcowden, 2015-07-18:
i like simplify.

jmwright, 2015-07-18:
I don't think we've set a precedent yet for adding settings to the chain, which makes me a little nervous. However, if @dcowden (as the architect of CadQuery) doesn't think it will break anything, I'm fine with it.

FreeCAD also uses "refine" in relation to removeSplitter in some contexts, but I also think simplify is good.

jmwright, 2015-07-18:
Oh, and I second @dcowden on his compliment. That was an awesome analysis! Thanks for doing that.

dcowden, 2015-07-18:
well, Hmm. @jmwright, after i thought about it more, i think maybe you're
right.

Early on, when i layed out the CQ methods, i used JQuery as an inspiration,
since it works very well. Just now i went to look at JQ to see if it allows
any sort of set operations on the stack.  It looks like it doesnt.

The whole notion of having a property that applies to all of the operations
afterwards is a little weird... it might be difficult to anticipate the
impact on all of the operations that can occur.

Another thing i just now thought of is that setting it in the chain doesn't
clearly indicate which operations it applies to. some operations do not
create solids, so it doesnt apply to those ( or does it? )

Why dont we do this:

(1) implement a clean() function. That way, when people want to do it
explicitly, they can
(2) add clean=True as a parameter to all functions that will call the
method, and default it to True. that makes it very easy to see which
methods it gets called in, and which ones it doesnt.

I also think that while its more places to change, it will be easier to get
this change right than introducing a new property that introduces global
affects...

jmwright, 2015-07-19:
This sounds like the way to go then. If it defaults to True as you've suggested, our existing test cases should tell us if anything breaks badly. I have some ideas to make our tests for solid operations more robust to catch more subtle geometry problems that this might introduce too, but that's a different conversation.

hyOzd, 2015-07-25:
Guys, I somehow missed above 2 messages. And already did an implementation of

> cq.Workplane("XY").autoSimplify(True)

It's in this branch https://github.com/hyOzd/cadquery/commits/add-simplify

I even wrote tests for it :S 

I can remove `autoSimplify` features, instead add a `simplify=True` parameter to each function that has autoSimplify. Should I use word `clean` instead?

----

>Another thing i just now thought of is that setting it in the chain doesn't clearly indicate which operations it applies to. some operations do not create solids, so it doesnt apply to those ( or does it? )

In theory a line or polyline can be simplified. Think of this;

    .line(10,0).line(10,0)

This will create a two part straight line, `simplify` can make this 1 single part which is equivalent to this;

    .line(20)

Bu we don't have an API for this and I don't think there would be any practical use for it. Just something that came to my mind.

jmwright, 2015-07-28:
> (1) implement a clean() function. That way, when people want to do it
> explicitly, they can
> (2) add clean=True as a parameter to all functions that will call the
> method, and default it to True. that makes it very easy to see which
> methods it gets called in, and which ones it doesnt.

> Should I use word clean instead?

Based on the feedback from @dcowden I would. Do you need help making this change? I'm not sure how many places clean=True would have to be added, but it seems like there could be quite a few.

hyOzd, 2015-07-29:
> but it seems like there could be quite a few.

Indeed there is. Almost all solid operations may cause splitters. But in some cases `removeSplitter()` can't do its job. Such as cylindrical faces. It doesn't fail but it doesn't remove the splitter. I'm not sure if we should still call `removeSplitter()` in these operations, hoping that freecad someday fixes this.

dcowden, 2015-07-29:
I would vote to call it. It might be a function of both the operation and
the details of the geometry , so maybe it would sometimes work...

jmwright, 2015-07-29:
> I would vote to call it.

As long as it doesn't break geometry in some cases, I'd also say to go ahead and call it.

hyOzd, 2015-08-02:
Guys, work is almost done. But I my implementation of `clean` method has a problem. It breaks the `rotateAboutCenter()` method. You can find the code in my [add-simplify](https://github.com/hyOzd/cadquery/tree/add-simplify) branch. And here is a small snippet to reproduce the issue I'm having:

```python
import cadquery as cq
from Helpers import show

b = cq.Workplane("XY").line(10,0).line(0,10).line(-10,0).close().extrude(5,clean=True).\
    rotateAboutCenter((1,0,0),90)

show(b)
```

`rotateAboutCenter` fails when accessing the `Center` method of the object in the stack. This object is returned from `clean` method and when trying to access it's `Center` method (original code ignores the exception for some reason) an exception is raised:

    'Part.Shape' object has no attribute 'CenterOfMass'

Could you have a look at this?

jmwright, 2015-08-02:
Awesome, thanks for working on this. I'll take a look at this tomorrow morning, unless @dcowden can look at it sooner. Maybe it has to do with the Shape class  itself not inheriting from [TopoShape](http://www.freecadweb.org/wiki/index.php?title=TopoShape_API).

jmwright, 2015-08-03:
Just started looking at this. I'm getting a different error using your sample code, but it seems to be directly related.
```python
Running the Python command 'CadQueryExecuteScript' failed:
Traceback (most recent call last):
  File "/usr/lib/freecad/Mod/CadQuery/Gui/Command.py", line 177, in Activated
    imp.load_source('temp.module', tempFile.name)
  File "/tmp/tmp11mOJx", line 5, in <module>
    rotateAboutCenter((1,0,0),90)
  File "/usr/lib/freecad/Mod/CadQuery/Libs/cadquery/CQ.py", line 691, in rotateAboutCenter
    return self.each(_rot, False)
  File "/usr/lib/freecad/Mod/CadQuery/Libs/cadquery/CQ.py", line 1481, in each
    r = callBackFunction(obj)
  File "/usr/lib/freecad/Mod/CadQuery/Libs/cadquery/CQ.py", line 688, in _rot
    endPt = startPt + endVec

unsupported operand type(s) for +: 'NoneType' and 'Vector'
```

hyOzd, 2015-08-03:
This is the same error I get. The reason `.Center()` returns `None` is that; original code ignores the exception. I had posted the ignored exception. Sorry for confusion.

jmwright, 2015-08-03:
Ok, got it. Shame on us for putting `pass` in an `except` clause.

https://github.com/hyOzd/cadquery/blob/add-simplify/cadquery/freecad_impl/shapes.py#L191

jmwright, 2015-08-03:
Ok, so I modified the code to see the underlying object we're dealing with before the rotateAboutCenter call.
```python
b = cq.Workplane("XY").line(10,0).line(0,10).line(-10,0).close().extrude(5,clean=False)#.\
#    rotateAboutCenter((1,0,0),90)
print(b.val().wrapped)
```
I get the following output using clean=True first, and clean=False second. I also tried it with `combine` being true and false and found that combine didn't influence the type.
```python
Executed /usr/lib/freecad/Mod/CadQuery/Templates/script_template.py
<Shape object at 0x51e7500>

Executed /usr/lib/freecad/Mod/CadQuery/Templates/script_template.py
<Solid object at 0x4416cd0>
```
FreeCAD Shape objects don't have the `centerOfMass` function, and it looks like the new `clean` function is converting solids to shapes with `Shape.cast`. https://github.com/hyOzd/cadquery/blob/add-simplify/cadquery/freecad_impl/shapes.py#L806

It looks to me like you're doing the right thing based on the existing codebase, and I'm wondering if we would be able to reproduce this centerOfMass error in other ways too. If we can, you've found a pretty significant a bug.

jmwright, 2015-08-03:
Here's my suggestion. The current `rotateAboutCenter` implementation doesn't distinguish between Shapes and Solids. I would suggest adding a check for Shape vs Solid and then grabbing the solids from the Shape if needed. There's an example in the last post [here](http://forum.freecadweb.org/viewtopic.php?t=2355). However, that post doesn't show that you have to call `Solids` as a function and then grab the 0th solid. 

The [following line in CQ.py's rotateAboutCenter implementation](https://github.com/hyOzd/cadquery/blob/add-simplify/cadquery/CQ.py#L687) (after Shape-Solid check) would then become:
```python
startPt = obj.Solids()[0].Center() 
```
I did a quick modification of your branch and this works. Before I make that change to master, I wanted to see if that seems like a reasonable fix.

@hyOzd and @dcowden - Any thoughts?

jmwright, 2015-08-03:
Hold on, I may have missed something here...

jmwright, 2015-08-03:
Yeah, you have to check the wrapped FreeCAD object's type to see whether it's Shape or Solid. The fixed line that I gave above will work with both wrapped Shape and Solid objects, as long as they don't have more than one solid bundled in them. You don't have to check the type. I'm not sure if there would be any cases in CadQuery where calling `Solids` would result in multiple solids being returned anyway.

jmwright, 2015-08-03:
Looks like I've come to the same place @dcowden was awhile ago.

https://github.com/dcowden/cadquery/blob/master/cadquery/freecad_impl/shapes.py#L854

The weighted average mentioned in the TODO is the correct way to do it. Grabbing the first Solid is bound to break eventually.

jmwright, 2015-08-03:
Seeing that @dcowden had already done a similar fix elsewhere in the codebase, I went ahead and made this change. The `Solids()[0]` fix is not optimal though. I also made it so that the usage of CenterOfMass will not silently fail. I will file issues to go back and fix both of these properly.

@hyOzd You should be able to fetch and merge the upstream master now so that we can move forward on getting your code merged. 04a753e152cd903cb325a757d998783e1d0d5581

hyOzd, 2015-08-03:
@jmwright Thanks for the fix : )

hyOzd, 2015-08-03:
: ( aaand I found another bug. Stacking of `box()` fails with `clean`:

    b = cq.Workplane("XY").box(5,5,5,clean=True).box(10,20,2)

```
Running the Python command 'CadQueryExecuteScript' failed:
Traceback (most recent call last):
  File "/home/heyyo/.FreeCAD/Mod/CadQuery/Gui/Command.py", line 178, in Activated
    imp.load_source('temp.module', tempFile.name)
  File "/tmp/tmpMEB1r5", line 10, in <module>
    b = cq.Workplane("XY").box(5,5,5).box(10,20,2)
  File "/home/heyyo/.FreeCAD/Mod/CadQuery/Libs/cadquery/CQ.py", line 2301, in box
    boxes = self.eachpoint(_makebox, True)
  File "/home/heyyo/.FreeCAD/Mod/CadQuery/Libs/cadquery/CQ.py", line 1518, in eachpoint
    pnts.append(v.Center())
  File "/home/heyyo/.FreeCAD/Mod/CadQuery/Libs/cadquery/freecad_impl/shapes.py", line 188, in Center
    return Vector(self.wrapped.CenterOfMass)

'Part.Shape' object has no attribute 'CenterOfMass'
```

This works:
 
    b = cq.Workplane("XY").box(5,5,5,clean=False).box(10,20,2)

Same problem occurs with `sphere()` method as well. This is turning into a mess. Maybe we should find a way to make `clean()` method act like ‒return the same type of object‒ as other operations. I still don't know the difference between Shape and Solid types. Is it possible to convert between them? What information is needed or what information is lost if we try to convert between those?

jmwright, 2015-08-04:
> I still don't know the difference between Shape and Solid types. Is it possible to convert between them?

I think that Shape is a higher level abstraction than a Solid, and can encompass different types of topology. The class diagram [here](http://www.freecadweb.org/wiki/index.php?title=Topological_data_scripting#Class_Diagram) gives some context.

I need to look at your example for a bit. I'm not sure what the best way to proceed is. If @dcowden has time to add some thoughts, that would be great.

jmwright, 2015-08-04:
The problem seems to revolve around the Center call. What if, for functions like Center, we just make the function more intelligent? It could check to see if it's given an appropriate object, and do what's needed to convert it if it's not.

jmwright, 2015-08-06:
Nobody cried foul about my idea above, so I went ahead and implemented it. @hyOzd could you merge the upstream master again and give it a try? The change worked for me when I tried it to your branch, so if it doesn't work I missed something in translation. By modifying Shape.Center() it should work for all cases.

Now that I've implemented this, I think I understand something that you mentioned earlier @hyOzd . Could we use the [cast_to_shape function](http://www.freecadweb.org/wiki/index.php?title=Part_API) of the Part API to convert the wrapped FreeCAD shape back to a solid after the [removeSplitter](https://github.com/dcowden/cadquery/blob/ea26397b8394ac1f72fafe6b8c61dc1ebaf4bca9/cadquery/freecad_impl/shapes.py#L806) call? This would make for an implementation that fits more cleanly with the existing code.

hyOzd, 2015-08-06:
Thanks for the fix @jmwright . I hope to work on this tonight. 

Documentation of the `cast_to_shape` says "Cast to the actual shape type "; are there two different types named 'Shape'? Have you investigated this?

jmwright, 2015-08-06:
I could be looking at it wrong, but I was thinking that since Part.Solid is a TopoShape, you could just do something like `obj.cast_to_shape(Part.Solid)`. I haven't been able to try it yet though, so I could be totally off. Part is imported as FreeCADPart in the shapes.py file too I think, just a heads-up. That line would actually be `obj.cast_to_shape(FreeCADPart.Solid)`.

jmwright, 2015-08-06:
I got a chance to experiment with `cast_to_shape` a little bit, and here are my findings.

My modified version of the clean function.
```python
def clean(self):
 	print self.wrapped
	print self.wrapped.removeSplitter()
	print FreeCADPart.cast_to_shape(self.wrapped.removeSplitter())
    return Shape.cast(self.wrapped.removeSplitter())
```
The output I get in FreeCAD.
```python
<Solid object at 0x3f59740>
<Shape object at 0x3b4d0e0>
<Solid object at 0x3b48be0>
```
You can see that by using `cast_to_shape`, the wrapped object is converted back to a Solid, which I believe solves the problem.

My thought is that you should check the type of the incoming wrapped object, and it should leave the clean function the same as it came in.

Hopefully that helps.

hyOzd, 2015-08-06:
Thanks a lot for example code. After looking at the [implementation](https://github.com/FreeCAD/FreeCAD/blob/5676c632f63b75955d3376f2fda09f4e892d490c/src/Mod/Part/App/AppPartPy.cpp#L1655) of the `cast_to_shape`, I think I'm starting to understand this. `TopoShape` seems to be a generic container, that may be holding different object types. `cast_to_shape` converts this `TopoShape` to a specific object type. If TopoShape contains a solid it's converted to a `Solid`, if it contains a wire it's converted to a wire etc. Please correct me if I'm wrong.

@jmwright In this case, are you sure a type check is required? I think we should just call `cast_to_shape` and return whatever it returns. It should be returning what went into the `removeSplitter()`, but if it doesn't; there isn't much we can do.

jmwright, 2015-08-06:
I think you're correct on how `cast_to_shape` operates, but I didn't test it to make sure. If that is correct, then I think you're right that you can skip the type check, especially since you're controlling where clean gets called.

dcowden, 2015-08-08:
Hey, @jmwright, can you fill me in on the latest? My head is spinning after
reading this thread, im just getting back from vacation.

jmwright, 2015-08-08:
@dcowden - In short, removeSplitter() changes the underlying type, converting a Part.Solid to a Part.Shape. This breaks function calls like CenterOfMass, which only exist on a Part.Solid object.  My first instinct was just to have the Center function concert the type if needed, but then we figured out that we could revert (safely it seems) the Solid-> Shape cast with cast_to_shape inside the Clean function. That seemed like the better way to go. I'm probably going to revert my change to Center even though it's not hurting anything.

@hyOzd - I saw that you referenced this issue in the pull request. Does that mean you're ready to have it merged?

hyOzd, 2015-08-08:
@jmwright It's ready for merge.

dcowden, 2015-08-08:
Ah ok thanks. Probably the fact that they return a shape means that in some
cases we may get a compound solid instead of a single solid. In the case
that the cast returns something other than the solid what would we do?

jmwright, 2015-08-08:
@dcowden The cast_to_shape function casts to whatever the underlying object should be, even if it's something like a wire. I haven't tested it, but I would expect it to handle a compound solid correctly too.

dcowden, 2015-08-08:
oh ok then that should work well then!

jmwright, 2015-08-09:
Ok. I squeezed in a few minutes to get the pull request merged and to do some manual testing. I am unable to get it to fail, and so I've deployed this to the latest commit of the FreeCAD module. https://github.com/jmwright/cadquery-freecad-module

Thanks @hyOzd for all your work on this!

closed 2015-08-09 by jmwright

dcowden, 2015-08-09:
+1 thanks for contributing!