dcowden/cadquery #102: Polyline Doesn't Allow Line To Be Continued or Closed

jmwright, 2015-07-29:
With the latest commit, the following code...
```python
result = cq.Workplane('XY').lineTo(0, 1).polyline([(0, 1), (1, 1), (1, 0)]).lineTo(0, 0)#.close()
```
...throws the following error, whether I use lineTo or close.
```python
Traceback (most recent call last):
  File "/usr/lib/freecad/Mod/CadQuery/Gui/Command.py", line 177, in Activated
    imp.load_source('temp.module', tempFile.name)
  File "/tmp/tmpRSyNaM", line 6, in <module>
    result = cq.Workplane('XY').lineTo(0, 1).polyline([(0, 1), (1, 1), (1, 0)]).lineTo(0, 0)#.close()
  File "/usr/lib/freecad/Mod/CadQuery/Libs/cadquery/CQ.py", line 1087, in lineTo
    startPoint = self._findFromPoint(False)
  File "/usr/lib/freecad/Mod/CadQuery/Libs/cadquery/CQ.py", line 979, in _findFromPoint
    raise RuntimeError("Cannot convert object type '%s' to vector " % type(obj))

Cannot convert object type '<class 'cadquery.freecad_impl.shapes.Wire'>' to vector
```
[Here's](https://github.com/dcowden/cadquery/blob/master/cadquery/CQ.py#L951) the _findFromPoint code. line, lineTo and spline all seem to work fine, the problem is localized to polyline.

jmwright, 2015-10-12:
It seems like this error is caused by [_addPendingWire](https://github.com/dcowden/cadquery/blob/74b4f0057d38ef6ec09d55054e0083c86e11f2b0/cadquery/CQ.py#L1379) not setting ctx.firstPoint like [_addPendingEdge](https://github.com/dcowden/cadquery/blob/74b4f0057d38ef6ec09d55054e0083c86e11f2b0/cadquery/CQ.py#L1367) does. Seems like _addPendingWire should select the correct edge from among its edges, and then set ctx.firstPoint to it.

@dcowden Any thoughts on this?

dcowden, 2015-10-12:
haha, well you are not going to like my thoughts.

I hate _addPendingWire. in fact, the whole _notion_ of pending wires has
always been a splinter in my brain. it always felt like there was a better
way to solve the problem.

I woulnt be opposed to having addPendingWire set ctx.firstPoint, but i
think there are a couple of other things to think about.

When addPendingWire was created, it was not designed to ensure that the
wires were iterable, or to track the first and last point, etc.

A wire might consist of edges that are in the wrong order. OCC does that
from time to time, so i assume FreeCAD could do it also.

If our goal is to set the firstPoint on a wire, then we need to make sure
that the edges are in head-to-tail order.  This would probably end up
needing to be a 'clean' kind of operation: we'd probably want to sort a
wires edges head-to-tail first, then set first point.   Hopefully we could
find a method in freecad to sort the wire so that its edges are in the
right order.

Another thing to think about is what happens when you have two wires that
are NOT connected, and you add them both using addPendingWire.  which one's
head becomes ctx.firstPoint? is it the most recently added one?  in some
sense, the firstPoint might only make sense to evaluate when the wires are
combined.  I _think_ i remember that all of the pending wires are combined
at the time you try to do a 3d operation like cut or extrude. So we could
compute the firstPoint at that point...

Is any of this making sense at all?

jmwright, 2015-10-12:
> his would probably end up needing to be a 'clean' kind of operation: we'd probably want to sort a wires edges head-to-tail first, then set first point.   Hopefully we could find a method in freecad to sort the wire so that its edges are in the right order.

I think there is one, and it was something I looked at when we were trying to fix a mirror problem awhile back. I'd have to dig through my FreeCAD forum posts to be sure I think.

To take a step back, the core problem I'm trying to solve is that you can't continue from, or close, a polyline. The polyline function uses Wire.makePolygon(), which returns a wire. When creating a polygon normally it's fine if it's a wire without any kind of firstPoint set because it's probably closed. In the case of the polyline function, people are likely going to want to close or continue it. I'm wondering for polyline, if I should just find the vectors and iterate over them with a call to lineTo for each vector.

dcowden, 2015-10-12:
yeah i agree.  it sounds like dealing with wires is the wrong layer of
abstraction here.  what if polyline just pushed points onto the stack, and
then we could use a close() method to connect them all? that way it coudl
be interoperable with pushPoints too

jmwright, 2015-10-12:
If I understand correctly, we would call the CQ.pushPoints function inside the polyline function and then close would call CQ.lineTo using self.eachpoint. Is that correct? How do we differentiate between the case where we're just closing from the last point to the firstPoint vs connecting a bunch of points, or do we even need to?

dcowden, 2015-10-12:
yeah you understood correctly.

But now that i reviewed more code, i can see that my suggestion would be a
pretty dumb implementation.

looks like pretty much all of the '2d'-ish methods: lineTo, spline, 3 point
arc, hline, vline, etc-- push an edge onto the stack.

so maybwe should just change polyline to iterate over the points and push a
set of edges. that way, its interoperable with the other 2d functions,
which is what i was really going after with using points instead of a wire.

jmwright, 2015-10-12:
Sounds good. I've run out of time for today, but I'll tackle this the next time I get some CadQuery time.

jmwright, 2015-10-13:
Haha, I just noticed this in the comment section for the polyline function.

>This should probably yield a list of edges, not a wire, so that  it is possible to combine a polyline with other edges and arcs

I've got a solution worked up and it's waiting for comment in PR #121

dcowden, 2015-10-13:
wow. ok so at some point i agreed with myself. i guess that's good.... :)

jmwright, 2015-10-14:
Fixed as of commit 76b6ce10a55b7a2ed62feda7f46b701601ae2114

closed 2015-10-14 by jmwright