dcowden/cadquery #105: add Color property for solid objects

easyw, 2015-08-08:
It would be useful to add Color property for solid objects and keep color for all functions (e.g. union cut etc.)
thanks

hyOzd, 2015-08-09:
I was going to create an issue for "exporting STEP with colors". Since I know we are both here for the same thing I thought I would just add this here : ) Yeah we want to export colorfuls STEPs from cadquery.

jmwright, 2015-08-09:
@easyw and @hyOzd - Thanks for the ideas. The CadQuery module for FreeCAD allows you to specify the color and transparency for an object when it's rendered. However, it sounds like you are both saying that you want the color to be embedded as an intrensic part of a solid. My thinking is that the color property would only matter when exporting the solid (STLs/STEPs with color), or displaying it (in FreeCAD). Are these the only use cases you both see for it?

On the surface this doesn't sound too bad, but operations that result in multiple solids (split), or operations on compound solids might get a little interesting. I'd need to research it a bit.

hyOzd, 2015-08-09:
Let me correct; I myself only care for colors when exporting or displaying. But it would be nice to be able to fuse parts, and conserve the colors of different faces. Like freecad does;

[image]

I know this has some limitations. For example refine shape operation of FreeCAD can merge different colored faces;

[image]

But I think such glitches should be expected. User can avoid them by carefully designing the part, cutting before fusing for example;

[image]

easyw, 2015-08-09:
@jmwright and @hyOzd 
we are using cadquery to generate a parametric lib of ICs models for kicad...
may be you could also consider to merge hyOzd fork that seems to have some enhancement...
https://github.com/hyOzd/cadquery
thank you
Maurice

hyOzd, 2015-08-09:
@easyw Sorry for confusion. I guess you have seen some of my earlier notes somewhere : ) My repository is already merged to upstream. You should be able to run my scripts with the latest version of cadquery.

dcowden, 2015-08-09:
Stl doesn't support color, at least in a standard way. Amf and step do.

easyw, 2015-08-09:
@dcowden also vrml does support color...

@hyOzd thanks ... you are right ... it does work with cadquery-freecad-module  0.1.8 :)

dcowden, 2015-08-09:
is vrml a commonly needed export?

One of CQ's missions has always been to make it easier to create models
that are higher quality and more re-usable than the current world-- STEP
supports this goal directly.  I'd favor adding color to STEP first for this
reason.

easyw, 2015-08-09:
@dcowden 
me and @hyOzd are doing a repository for kicad, generating 3D parametric models
kicad would need vrml models for its internal 3d-viewer, and STEP AP214 models for MCAD exchanges
FC can export in VRML with colors, so STEP with colors would be the first and maybe the only needed target.
For the moment I've done some python libs/functions to manage the objects generated by hyOzd script, to export STEP and VRML models, appropriate to the kicad environment.
https://github.com/KiCad/kicad-library/issues/186
thank you for your nice framework
Maurice

dcowden, 2015-08-09:
> KiCad/kicad-library#186

that's awesome! i'm glad to hear its helpful!

hyOzd, 2015-08-10:
I have created an X3D exporter for freecad objects which I used in my generator scripts. It's [here](https://bitbucket.org/hyOzd/freecad-macros/src/e85defa66322bad7f2ef5df6fac17b5e94268eb8/export_x3d.py?at=master). As you can see it's rather primitive, but it works. VRML and X3D are very similar specifications. Actually X3D is the standard replacing VRML. I have already spent some time in the specification documentation. So I can lend a hand for VRML/X3D export if you are interested.

By the way I agree with @dcowden. I would love to see colorful STEP export first.

hyOzd, 2015-08-11:
[Here](https://gist.github.com/hyOzd/0d75fc98cccfae92a641) is a snippet I've created to export a simple cadquery object to STEP (with color), using FreeCADs GUI module (inspired by @easyw's export scripts). Thought you would be interested.

Note that this looks too much like a hack. I'm creating all the GUI of FreeCAD hidden. And I'm not calling the main event loop, which is required for a qt application to perform. This works now, but may not work in the future. I wouldn't count on such implementation. I wouldn't put this code in cadquery. Maybe as a utility/helper module, until a better option is implemented.

dcowden, 2015-08-11:
I see. Why does something like this not work to export step ( much simpler
):
(not tested, i only have access to email ATM )

import cadquery as cq

box = cq.Workplane("XY").box(10,10,10)
box.exportStep( box , fileName)

This method uses the underlying freecad implementation. For more control,
you can use something like this:

from cadquery import exporters
exporters.exportShape(box, ExportTypes.STEP, fileLikeObject, 0.1 )

the source is here:
https://github.com/dcowden/cadquery/blob/master/cadquery/freecad_impl/exporters.py:

​it contains code to export to SVG, STEP, AMF, and STL.  ( as per other
discussions, i think FreeCAD's implementation of STL export should be
replaced with our own so we can control the tolerance of the tessellation.

looking at it, i think the code in Shape should always delegate to the
exporters routines-- that way the user can control the tessellation and
other stuff.

​

hyOzd, 2015-08-11:
AFAIK STEP exported this way -`Shape.exportStep()`- won't have color.

My example was very simple. It's possible to color different faces of shape to a different color this way. That requires access to `ViewObject` attribute which is not provided by `FreeCAD` module but `FreeCADGui` module.

dcowden, 2015-08-11:
ah i see-- your sample creates color ok, gotcha.

jmwright, 2015-08-13:
@hyOzd @easyw and @dcowden - I'm not ignoring these conversations, I've just had almost no time for CadQuery this week. I will likely tackle issue #85 first, and then circle back to this. I do see it being a challenge though because of the FreeCAD View boiler plate code that @hyOzd mentioned. We'll have to figure out a way to get around having to do that.

Please keep in mind that I'm also trying to round out our test suite (and fixing the bugs that brings to light) to set us up to take even more great contributions from the community. I'm spread pretty thin, so it could be awhile before I can get to this.

dcowden, 2015-08-13:
no problem, thanks for weighing in!

iromero91, 2018-08-05:
Is there a chance this could sneak into the CQ 2.0 cycle? It would be really nice to incorporate CQ into Kicad without having freecad in the middle...

jmwright, 2018-08-07:
@iromero91 It's hard to say. It depends on the resources available, and everybody is pretty busy right now. Maybe once we get past the summer the team can re-evaluate where we're at and the resources we have.

+1 on incorporating CQ into KiCAD without FreeCAD in the mix.

adam-urbanczyk, 2018-08-07:
@iromero91 what is the exact use case? Would it be something like:

1) define individual shapes and their materials/colors,
2) assembly into a compound solid,
3) export to e.g. STEP?

Or is it something more involved with colors per face within individual solids?

iromero91, 2018-08-11:
As described above, to generate models that have multiple colors (representing multiple materials) for use in kicad pcb assemblies, as kicad can read step files natively now.

adam-urbanczyk, 2018-08-16:
Clear! I think it will be reasonably easy to add color per solid and use that while exporting STEP files. I'd rather not work on color per face and fuse using colors though.