dcowden/cadquery #106: Union Problem

jmwright, 2015-08-17:
While trying to round out some unit tests on CQ.py, I got confused by the behavior of union.

The following code results in only one of the two boxes being shown instead of a combination of both.

```python
s = cq.Workplane('XY')
box1 = s.rect(2.0,2.0).extrude(0.5)
box2 = s.rect(1.0,1.0).extrude(1.0)

box1.union(box2.val(), combine=False)
```
Gives:
[image]

Whereas, setting combine to true gives the other box:
[image]

However, shouldn't the result of the union operation look something like this:
[image]

dcowden, 2015-08-17:
yeah the semantics of combine=False are not exactly obvious when you are
using a union function, right? I mean union with combine=False doesnt make
any sense.

i cannot immediately explain the behavior-- except to say that its probably
not even be possible to use the combine flag with the union operation....

jmwright, 2015-08-17:
Ok, thanks. I'll put it on the list of things that need to be investigated.

jmwright, 2015-10-12:
The following code works as expected, closing this.
```python
import cadquery as cq
from Helpers import show

s = cq.Workplane('XY')
box1 = s.rect(2.0,2.0).extrude(0.5)
box2 = s.rect(1.0,1.0).extrude(1.0)

box3 = box1.union(box2)

show(box3, (204, 204, 204, 0.0))
```

closed 2015-10-12 by jmwright