dcowden/cadquery #107: extrude/cutBlind with multiple overlapping rectangles is broken

hyOzd, 2015-08-31:
If I push two *overlapping* rectangles to the stack then extrude them, resulting shape is broken. Result is okay if rectangles do not overlap. 

```python
import cadquery as cq
r = cq.Workplane("XY").box(10,10,10)
r = r.faces(">Z").workplane().rect(1,5).rect(5,1).extrude(1)
from Helpers import show

show(r, (204, 204, 204, 0.0))
```
[image]

Similar problem occurs with `cutBlind()`. 

I was trying to create a plus shape the easy way. Please let me now if this is not a valid use case.

jmwright, 2015-08-31:
What happens if you combine and simplify the plus before extruding or cutting?

hyOzd, 2015-08-31:
You mean like this;

    r = r.faces(">Z").workplane().rect(1,5).rect(5,1).combine().extrude(1)

This one fails with;
```
Running the Python command 'CadQueryExecuteScript' failed:
Traceback (most recent call last):
  File "/home/heyyo/.FreeCAD/Mod/CadQuery/Gui/Command.py", line 177, in Activated
    imp.load_source('temp.module', tempFile.name)
  File "/tmp/tmpy5VWnu", line 3, in <module>
    r = r.faces(">Z").workplane().rect(1,5).rect(5,1).combine().extrude(1)
  File "/home/heyyo/.FreeCAD/Mod/CadQuery/Libs/cadquery/CQ.py", line 2023, in combine
    if clean: s = s.clean()

'Wire' object has no attribute 'clean'
```

I've disabled that line, and it worked without `clean`, but the result is the same.

By the way is `combine()` a valid operation for 2d drawings, in this case I guess I should fix the above error : )

dcowden, 2015-08-31:
Yeah, i would not have expected this to work. I dont think it will work
even in pretty advanced CAD apps like SolidWorks.

I think the easiest way to make this shape would be to to use  polyline.

hyOzd, 2015-09-01:
I understand that two overlapping rectangles is not a regular shape that can be extruded without problems. But since some cadquery methods operate on stack, my expectation was this; extrude rectangles separately then combine them by fusing (union). After that combine with (or cut from in case of cutBlind) the parent solid. What do you think?

dcowden, 2015-09-01:
Yes I understand.  The concept of fusing in 2d is a great idea. However,
there is no support for this in freecad or even in OCC that I know of.

This feature would take some hard core development to implement, because it
would need to be done completely in cq itself.

hyOzd, 2015-09-01:
No, I meant; first extrude both rectangles separately, then fuse the extruded solids...

dcowden, 2015-09-01:
oh, ok sorry I misunderstood. I would expect fusing the two overlapping
extruded ones to work. though as per earlier i wouldnt expect it to work
without the clean() method, which you had to disable.

so maybe the path forward is to figure out why the clean() method threw an
exception...

jmwright, 2015-09-01:
> By the way is combine() a valid operation for 2d drawings, in this case I guess I should fix the above error 

If I remember right, the clean function was only implemented by @hyOzd for Solid objects. I would expect that's why the exception is thrown. I looked around for a few minutes and had trouble finding the equivalent of removeSplitter for Wires. I'm not sure how clean would be implemented in this case.

hyOzd, 2015-09-01:
How about an empty implementation that does nothing?

dcowden, 2015-09-01:
yeah that sounds like a good idea.....

hyOzd, 2015-09-02:
I've tried something [here](https://github.com/dcowden/cadquery/compare/master...hyOzd:fix-extrude). This fixed the issue and created the result I want. But I'm not exactly sure with these code changes. 

Why did `cast` return a Solid even if compound contains more than 1 solids? Note that 3 tests fail probably related to this change. This is one of the failing lines;

```
Traceback (most recent call last):
  File "/home/heyyo/working/cadquery/tests/TestCadQuery.py", line 966, in testBoxPointList
    self.assertEquals(1, s.solids().size())
AssertionError: 1 != 4
```
I'm guessing that, since Compound was casted to Solid, `solids()` were returning only 1 Solid object. But now that I've changes this, it's returning 4 Solid object. Comments in the test code are conflicting. I'm not sure which one is the correct behaviour.

jmwright, 2015-09-03:
I've talked to @dcowden about this sort of thing before. My guess is that the test just needs to be made more robust (to handle compounds as a special case), but I'll need to dig into it a little more. I'll try to do that tomorrow.

dcowden, 2015-09-03:
Yeah don't put too much stock in my old tests. There were a lot of late
nights involved when I wrote them.

jmwright, 2015-09-03:
@hyOzd I've looked at your change and think it looks good. I like how you've set it up to default to preserving the separate solids rather than combining them.

I feel that the comment in that test is actually correct, and that the assert is wrong. The following line...
```python
self.assertEquals(1, s.solids().size())
```
...should be...
```python
self.assertEquals(4, s.solids().size())
```
as you've noted. Feel free to roll the change to that test into the pull request for your fix, or I can probably get to the test change tonight.

hyOzd, 2015-09-03:
I am planning to add a test for my changes. I can fix those as well.

jmwright, 2015-09-03:
Awesome, thanks!

dcowden, 2015-09-03:
Thanks so much for contributing!

hyOzd, 2015-09-06:
Guys, my changes triggered another bug. `Compound::Center()` method is broken, it makes a call to itself:

```python
    def Center(self):
        return self.Center()
```

This causes a fail due to endless recursion. Here is an example code to trigger the issue (this happens with my changes only):

```python
import cadquery as cq
from Helpers import show

boxes = cq.Workplane("XY").box(10,10,10).\
        workplane(offset=20).box(10,10,10).\
        workplane(offset=20).box(10,10,10)
    
show(boxes)
```
Error message:
```
...........
...........
  File "/home/heyyo/.FreeCAD/Mod/CadQuery/Libs/cadquery/freecad_impl/shapes.py", line 871, in Center
    return self.Center()
  File "/home/heyyo/.FreeCAD/Mod/CadQuery/Libs/cadquery/freecad_impl/shapes.py", line 871, in Center
    return self.Center()
  File "/home/heyyo/.FreeCAD/Mod/CadQuery/Libs/cadquery/freecad_impl/shapes.py", line 871, in Cent
maximum recursion depth exceeded
```
I think the reason my changes triggered this bug is; before we were always casting to "Solid", even if it was a compound. I don't understand how it worked out before. But currently I'm trying to find/implement a valid `Center` method for `Compound` class. Any tips/help appreciated.

hyOzd, 2015-09-06:
I think, Compound.Center was supposed to call Shape.Center, but there was a typo involved. So I fixed it by pointing to Shape.Center. Then I implemented weighted average method for calculating Center. Please have a look at it; https://github.com/hyOzd/cadquery/commit/52c4e0aadbfb7877078f7977f90702a3cc6b5ce0 .

This change will fail if Compound doesn't contain any solids. Do you think we should do something about this?

jmwright, 2015-09-06:
@hyOzd I didn't test it directly, but I think your code looks fine. You could add a check that throws a descriptive error if there are no solids in the compound. The user likely won't get a good error message on that otherwise.

hyOzd, 2015-09-06:
@jmwright I will add the warning.

Code looks fine but it isn't : )

This fails:

    box = cq.Workplane("XY").pushPoints([(0,0),(10,0)]).box(5,5,5).\
          edges(">Z").fillet(0.2)

```
  File "/home/heyyo/.FreeCAD/Mod/CadQuery/Libs/cadquery/CQ.py", line 793, in fillet
    s = solid.fillet(radius, edgeList)

'NoneType' object has no attribute 'fillet'
```
It's not shown in the error message but the problem is originated from this line of the `fillet()` function:
    
    solid = self.findSolid()

`findSolid` looks for an object of type `Solid`, but cannot find it because resulting shape of the above operation is a `Compound`.

I'm stuck.

jmwright, 2015-09-07:
@hyOzd  I'll have to look at this in the morning, but I see what you're saying. The existing codebase would have never worked correctly in this case anyway, but didn't throw the error. I don't feel like you should have to combine the solids of a compound to get something like this to work, but fixing it properly is going to be an interesting challenge. I'll post updates as I learn more in the morning.

jmwright, 2015-09-07:
@dcowden We're kind of in a tough spot here I think.

I ran into this same problem a few months ago without realizing it. I was doing a braille example and wasn't able to fillet the tops and bottoms of the dots properly because of this. I didn't dig into it at the time because it wasn't a critical feature of the example.

Part of the issue is that the selectors don't seem to handle this compound case correctly. So, when `edges()` is called on the compound, you get far from the proper behavior. You can see this if you run the example above in the CadQuery module.

```python
box = cq.Workplane("XY").pushPoints([(0,0),(10,0)]).box(5,5,5).\
      edges(">Z").fillet(0.2)
```

This would be a pretty big change/fix.

1. We'd have to fix the selectors to handle Compounds.

2. We'd have to change the findSolid function so that it returns a list of solids if there are more than one (i.e. in a Compound).
```python
def findSolid(self, searchStack=True, searchParents=True):
        if searchStack:
            for s in self.objects:
                if type(s) == Solid:
                    return s
                elif type(s) == Compound:
                    return s.Solids()

        if searchParents and self.parent is not None:
            return self.parent.findSolid(searchStack=True, searchParents=searchParents)

        return None
```
We should probably also raise a descriptive error, rather than just passing `None` back.
3. We'd have to update the code everywhere that `findSolid` is called. It seems to only be used internally in CQ.py so this is doable, but will require stepping through each of the solids individually and applying the and operation. Looking at the example from @hyOzd above, I'm not sure yet how to keep the solids matched up with their edges that were selected when dealing with Compounds.

dcowden, 2015-09-08:
i agree with your assessment.  Compounds are annoying, but are generally
not something we should make the user deal with i think-- at least _most_
of the time.  I agree that edges() should return all edges-- if a compound,
we'd want all of the edges of all the solids, etc.

But sometimes i want to deal with a single solid underneath the compound--
if we hide the presence of compounds, by making them look like a list of
solids, there still needs to be a way to select one of the child solids.
If compounds were put onto the stack as a list of solids, then it would be
trivial to select one, but how to know which one? maybe a selector matches
solids by coordinate, etc would work there?

jmwright, 2015-09-14:
@hyOzd and @dcowden Where do you two feel we're at on this?

jmwright, 2015-09-17:
The Center fix is in PR #114

jmwright, 2015-09-18:
Got curious about the original problem that sparked this issue, and worked through the following.
```python
import cadquery as cq
from Helpers import show

r = cq.Workplane("XY").box(10, 10, 10)

# We're going to use this workplane in multiple places
wp = r.faces(">Z").workplane()

# Build the cross shape out of two extruded rectangles
rect1 = wp.rect(1, 5).extrude(1)
rect2 = wp.rect(5, 1).extrude(1)
r = rect1.union(rect2)

show(r, (204, 204, 204, 0.0))
```
That yields the following.

[image]

jmwright, 2015-09-18:
I think the only thing that's left from this thread is making findSolid() so that it handles compounds properly. Below is my code from above, but there are a couple of different ways this could go. The first is that it returns a list of all solids in a compound. I'm thinking that this is the preferred method, and as far as I can tell won't break any tests. The second option is to fuse all solids before returning just the one, monolithic solid. Any preferences?

```python
def findSolid(self, searchStack=True, searchParents=True):
        if searchStack:
            for s in self.objects:
                if type(s) == Solid:
                    return s
                elif type(s) == Compound:
                    return s.Solids()

        if searchParents and self.parent is not None:
            return self.parent.findSolid(searchStack=True, searchParents=searchParents)

        return None
```

dcowden, 2015-09-18:
I definitely think the first is preferred. we don't know for sure if the
user wants to fuse several different solids into one. they can do that
themselves if they want, but doing it for them removes the ability to undo
it easily.

hyOzd, 2015-09-18:
There is also that when you fuse two Solids, if they don't intersect (completely separate shapes) FreeCAD will return a compound. AFAIK it's not possible to get rid of Compound in this case. I'm guessing freecad's engine doesn't allow such shape to exists as a single Solid object.

dcowden, 2015-09-18:
correct! neither does the underlying kernel OCC-- the idea of a Compound
exists for precisely this reason, even at the lowest possible level.

jmwright, 2015-10-12:
Added proper handling of compounds in findSolid() in commit 74b4f0057d38ef6ec09d55054e0083c86e11f2b0

closed 2015-10-12 by jmwright