dcowden/cadquery #108: `clean()` -in some cases- fails with shapes other than `Solid`

hyOzd, 2015-09-01:
Prior discussion was in #107.

This fails:

    r = cq.Workplane("XY").rect(10,10).rect(10,10).combine()

with:
```
"/home/heyyo/.FreeCAD/Mod/CadQuery/Libs/cadquery/CQ.py", line 2023, in combine
    if clean: s = s.clean()
'Wire' object has no attribute 'clean'
```

Reason: when I implemented `clean` functionality, I've added `clean()` method only for `Solid` class. An easy solution seems to be implementing an empty `clean()` method for other classes as well. 

But we also have this;

    r = cq.Workplane("XY").rect(10,10).clean()

```
"/home/heyyo/.FreeCAD/Mod/CadQuery/Libs/cadquery/CQ.py", line 2417, in clean
    raise ValueError("There is no solid to clean!")

There is no solid to clean!
```

I want to solve this issue by adding `clean()` method for other classes in `shapes.py` but above error implies the user is in the wrong by calling `clean()` API on an unsupported object. My question is should I remove this check, and let the users call the `clean()` API which will have no effect?

dcowden, 2015-09-01:
+1. I don't think there's much harm in empty clean() methods, especially
since in the future i can see a possibility they may not do nothing.

jmwright, 2015-09-01:
I'm fine with this solution. We define when it's wrong to call clean on an object, and you guys have made a good argument for wires. The only thing I would suggest is to leave the check in place for objects that we'd never want to call clean on (like edges or vertices).

hyOzd, 2015-09-01:
@jmwright thanks for answering before I ask ;)

But the thing is, I don't know how should I call `clean()` method of objects other then `Solid`. Current implementation uses `findSolid()` to find a Solid object, how can I access other object types? Or just the latest object in the stack?

jmwright, 2015-09-01:
I've pushed this change out into the CadQuery module for FreeCAD. It's available in the latest commit of master.

closed 2015-09-02 by hyOzd