dcowden/cadquery #116: cleanup freecad_impl

galou, 2015-10-04:
I would like to clean up the freecad_impl files.
For example;
- this is not a bug that FreeCAD modifies the vectors in place, this is a choice. The functions are called `normalize` and not `normalized`. On my opinion, `normalized` should return a copy and `normalize` should modify in place (can possibly also return a copy of the modified version). I would like to rename `normalize` to `normalized`.
- Plane.zDir should the original direction, Plane.invZDir should the inverse. With the current code, Plane.zDir is Plane.invZDir (same reference). I think this is what is expected (what is invZDir used for by the way?).

Is this OK?
What about PEP8 (at least for whitespaces, I would not touch variable names)?

Gaël

jmwright, 2015-10-06:
@galou Do you anticipate the normalize/normalized chain would break the existing implementation? I don't think we'd have to worry about breaking things for the users since normalize is used internally almost exclusively, but it might require quite a few internal updates to make this change.

> What about PEP8 (at least for whitespaces, I would not touch variable names)?

I have actually been cleaning up the whitespace in accordance with PEP8 as I go. I would be in support of making this update across the codebase.

jmwright, 2015-10-14:
> I would like to rename normalize to normalized.

I vote to go ahead with this change. It would bring CadQuery in line with the way OpenCASCADE defines normalize and normalized. http://dev.opencascade.org/doc/refman/html/class_n_collection___vec3.html#ad1f196a05f81122700566e74b2df344a

jmwright, 2015-11-24:
The normalize/normalized change has been addressed in commit 9ce84fb5d67a91f67574ae75df98c7b2deff1399

closed 2015-11-24 by jmwright