dcowden/cadquery #117: Vector.__hash__()

galou, 2015-10-05:
According to Python documentation (https://docs.python.org/2.7/reference/datamodel.html?highlight=__hash__#object.__hash__),
" If a class defines mutable objects and implements a __cmp__() or __eq__() method, it should not implement __hash__(), since hashable collection implementations require that a object’s hash value is immutable (if the object’s hash value changes, it will be in the wrong hash bucket)"

What is the reason why you implemented the __hash__ function?

jmwright, 2015-10-05:
I think that's a question only @dcowden can answer.

jmwright, 2015-10-23:
@galou I have removed `__hash_` from `Vector` as of commit 146ecd589bf721976d20784c50ee49a086c6fd4b

closed 2015-10-23 by jmwright