dcowden/cadquery #123: About planned STEP import function

huskier, 2015-11-22:
Dear Dave, the STEP import function is planned in the roadmap. I just wonder how it is going. I'd like to help a little bit for this.

With this function, maybe we don't need do the reverse engineering things in some cases, and we just got a seed part, and then grow from it.

dcowden, 2015-11-22:
Hi there:

I'm afraid other projects ( mainly kid activities ) are taking most of my
time at the moment, there's been no progress on STEP import that i know of.

I'm not sure i understand exactly how STEP import would work. STEP geometry
is much, much better than STL, but of course it still does not contain any
parametric information. What, more specifically, would STEP import do?

Perhaps we could discuss a bit more how it would work, and then we could go
from there once we have agreement?

huskier, 2015-11-22:
@dcowden What I thought is very simple. If we have a portion of DIN_rail(like http://stevenengineering.com/ord_center/Images/Featured/67_0801733.gif), we could extrude its length as we want. We don't need to do the 2D sketch first, then extrude.

Maybe we need to convert the imported STEP file into a CQ object, then we can do other things with the power of CADQuery.

dcowden, 2015-11-22:
Ok I see, the value would be to import 2d geometry. I can see the value
there. When it is imported, it cannot be parametic, but it can be the basis
for other stuff

CadQuery already supports import of 3D step objects as a solid. For
example, you could import a pulley with no hole in its center from s step
file, and then use cq to make a parametric center hole. Or you could import
a solid and use it as a part of a scripted solution,  for example import
screw threads.

But today we cannot import pure 2d geometry and use it as a set of
wires/edges, which would be nice. That's doable . Is this the primary value
for you?

One other comment about parametic 2d.  CQ is really lacking completely the
features required to make 2d sketching parametric. We have basic procedural
tools, but we don't have a geometric solver and constraints, which would
allow much better sketching experience. For example, you cannot draw two
parallel lines and set the distance they are apart. Or create a centerline
and use it to define symmetry. Or draw a line and trim it at another
entity.

I feel like importing 2d geometry from step would be a good feature, but I
also think that it's because cq's 2d offering is not easy enough.

huskier, 2015-11-23:
@dcowden 

>Ok I see, the value would be to import 2d geometry. I can see the value
>there. When it is imported, it cannot be parametic, but it can be the basis
>for other stuff

YES, that's what I want.

>But today we cannot import pure 2d geometry and use it as a set of
>wires/edges, which would be nice. That's doable . Is this the primary value
>for you?

Partly. I also want to import a solid and use it as a part of a scripted solution.

>CadQuery already supports import of 3D step objects as a solid.

Dave, you mentioned that CadQuery already supports import of 3D step objects as a solid, however, I could not find the API in the document. Could you please show me an example?

I am thinking of mixing FreeCAD and CadQuery scripting, and employing the power of these two excellent library. There is already an example there "Ex024_Using_FreeCAD_Solids_as_CQ_Objects1".  I prefer the syntax of CadQuery to FreeCAD, and CQ's is much simpler.

dcowden, 2015-11-23:
hmm. looks like the code is not linked into the main CQ api, but here's the
code:

https://github.com/dcowden/cadquery/blob/master/cadquery/freecad_impl/importers.py

here is a live example over on parametricparts.com: ( it uses a slightly
different syntax, but the approach is the same as if you use the
open-source CQ version):

http://parametricparts.com/parts/8xk8z5n2/

in that model you can see I imported a step model, and then applied  a hole
and filleted the edges. The model demonstrates that you can select edges
and faces of an imported STEP model as if it were created in CQ.

we just need to add a convenience method into the CQ object for import, but
for now you can probably do something like:

from cadquery.freecad_impl import importers
myShape = importers.importStep(<filename>)
myShape.faces(">Z").hole(0.2)

huskier, 2015-11-23:
@dcowden  Thank you very much for the example. Now the first goal, importing a solid and use it as a part of a scripted solution, is achieved.

Let's talk about the second goal: extracting the imported STEP model's profiles for further use.
[Reference image] https://github.com/huskier/images/blob/master/STEPModelProfile.png
The profile in the above image is captured from a section bar model (https://github.com/huskier/images/blob/master/SectionBar.step). The profile is quite complex. We prefer extracting the profile from model to drawing ourselves. I've been stuck at myShape.faces("<Y").wires(). Any ideas how to do this? Thanks.

```
import cadquery as cq
from Helpers import show
from cadquery.freecad_impl import importers
myShape = importers.importStep("E:/siom-work/FreeCAD/SectionBar.step")
print myShape.faces("<Y").wires()
```

dcowden, 2015-11-23:
That looks like a good start. What happens when you run your code sample?

huskier, 2015-11-24:
@dcowden  It returns "cadquery.CQ.Workplane object at 0x0BE5E930", and no further information.
and 
print myShape.faces("<Y").wires() 
returns 1.

I am thinking of that : circle is a 2D sketch, so should wires. 

Maybe we need the syntax of CQ.Workplane("XY").sketch2D(circle(10.0)), which should be equal to CQ.Workplane("XY").circle(10.0). This syntax will make our goal doable easily, like:

myWire = myShape.faces("<Y").wires()
CQ.Workplane("XY").sketch2D(myWire).extrude(100)

Is it possible?

dcowden, 2015-11-24:
hmmm. i dont compltely understand whats' going on. i didnt exptect
myShape.faces("<Y").wires() to return 1, i expected a list of wires.

I can see part of the problem here, though: we need a way to select a face
that has existing wires on it, but then actually start with those wires on
the stack. that's currently not possible, i think.

You may be right that we need to separate the notion of a sketch from a
workplane.  I debated this many times when writing CQ.  in solidworks, the
two notions are different, and that would be an advantage here.

I need to do some prototyping on this.  Im in the process of getting my dev
environment up and running, and additionally we're starting discussions
about CQ 2.0-- which might be based on a completely different backend.  If
we do something as radical as splitting sketches from workplanes, that'd be
a CQ 2.0 thing.

huskier, 2015-11-24:
>hmmm. i dont compltely understand whats' going on. i didnt exptect myShape.faces("<Y").wires() to >return 1, i expected a list of wires.

My fault. myShape.faces("<Y").wires().size() returns 1.

>additionally we're starting discussions about CQ 2.0

I would like to participate in the discussion, and learn from you gurus. So where does this happens? Google group? Unfortunately, we(China) are under a great protection from Google things.

dcowden, 2015-11-24:
> myShape.faces("return 1, i expected a list of wires.

sure, glad to include you!

jmwright, 2015-11-24:
Unfortunately, we do have a Google Group for general discussion which you probably won't be able to access. https://groups.google.com/forum/#!forum/cadquery

You're always welcome to post general discussion topics here on Github if that's what you have access to.

huskier, 2015-11-25:
@dcowden 
@jmwright 
Thanks for your kindness!

closed 2015-11-25 by huskier