dcowden/cadquery #126: About mutiple parts exporting into a single STEP file

huskier, 2015-11-26:
At present, the Shape class has a member function called exportStep(fileName). This function can only support a single part exporting. When we open the exported file, the label appeared as "Open CASCADE STEP translator 6.8", not meaningful.

If we create a box with CQ, we need to create six parts: top_prt, bottom_prt, left_prt, right_prt, front_prt and back_prt. However, CadQuery cannot support export six parts simultaneously. We could use union fuction to union the six parts all. However, this way we only get a whole part, and we cannot edit individual part in other CAD environment later.

dcowden, 2015-11-26:
Ok there are two issues here.. the label, and the behavior of multiple
parts.

The label is something we cannot control without a patch by Occ.

The multiple parts is something we can perhaps work with. The way to handle
this case is with a compound object. It is a single object which composes
more than one separate solid body.  In occ, using a compound is the only
way to achieve a step file containing multiple,  separate bodies.

CadQuery generally handles this using the union =true parameter , but when
you choose false, it creates separate solids that are neither unioned nor
part of the same compound.

We may need to add a method that allows intentionally grouping several
objects into a compound.

At present, the Shape class has a member function called
exportStep(fileName). This function can only support a single part
exporting. When we open the exported file, the label appeared as "Open
CASCADE STEP translator 6.8", not meaningful.

If we create a box with CQ, we need to create six parts: top_prt,
bottom_prt, left_prt, right_prt, front_prt and back_prt. However, CadQuery
cannot support export six parts simultaneously. We could use union fuction
to union the six parts all. However, this way we only get a whole part, and
we cannot edit individual part in other CAD environment later.

hyOzd, 2015-11-26:
@huskier I've developed a small python module for exporting multiple part cadquery objects to STEP (and VRML/X3D/FreeCAD) with colors. It uses FreeCAD api to do this. So it's not developed on top of cadquery. That means it may not be compatible with cadquery in the future if cadquery switches to another backend as planned. But, it works at the moment.

Here is the module: https://github.com/hyOzd/ecad-3d-model-generator/blob/master/e3dmg/exporters/export.py

Main entry point of this module is the `export()` function. If you are interested in a temporary solution I can show you a small example on how to use it.

dcowden, 2015-11-26:
We're going to be having a discussion in the google group very soon on the
future/roadmap of CQ-- but in short Jeremy and I are thinking that for now
staying with a freecad backend is still our best bet.

I'd like to consider how to provide the functionality you guys need in CQ.

I have another set of code buried in the CQ exporters to make it simple to
save a shape as an SVG file, which is very convenient for generating
documentation and debugging. I'd like to make that much easier to use too.

huskier, 2015-11-27:
@dcowden  I'a looking forward to your effort for more functions. I'd like to help a little bit if needed.

@hyOzd  I'm very glad if you could show me an example for your module.

Maybe a general Philosophy is combine all the available tools together, but it is very complex!

hyOzd, 2015-11-27:
You will need these two files.

* [export.py](https://github.com/hyOzd/ecad-3d-model-generator/blob/master/e3dmg/exporters/export.py)
* [componentmodel.py](https://github.com/hyOzd/ecad-3d-model-generator/blob/master/e3dmg/componentmodel.py)

But here I've put these to into [a single file](https://gist.github.com/hyOzd/a0a2b089cd6a001c9aa0).

A small example with two cubes:

```
import cadquery as cq
from mpexport import ComponentModel, export

box1 = cq.Workplane("XY").box(10,20,10)
box2 = cq.Workplane("XY").box(20,10,10)

model = ComponentModel()
red = (1,0,0)
green = (0,1,0)
model.addPart(box1, red, "box1")
model.addPart(box2, green, "box2")
componentName = "model"  # parts are labeled as componentName_partname
export("STEP", basename, model, "boxes.step", fuse = False)
```

This will generate a step file with two cubes in it. One is red and labeled as "model_box1" and the other one is green and is labeled "model_box2".

`componentName` argument can be annoying, you can change it [here](https://gist.github.com/hyOzd/a0a2b089cd6a001c9aa0#file-mpexport-py-L147).

Let me know if you need any more help.

dcowden, 2015-11-27:
Cool! Thanks for figuring out how to do this!

What would be the best way to integrate this into the cq core?

hyOzd, 2015-11-27:
Hmm, integrating this into cq core... Maybe as a plugin.. Like a bonus feature. That's because I really didn't like how this code works. And I tried everything. That's the only way it worked.

cadquery only uses the `App` module of the freecad, it doesn't use the `Gui` module which isn't much useful outside of FreeCAD application. But the thing is, if you want any operation that requires colors, you will need the `Gui` module. `App` module doesn't carry or deal with any sort of color information whatsoever. 

Just importing the `Gui` module as `FreeCADGui` isn't enough. Some portions of the `Gui` module are not initialized unless you call the global `FreeCADGui.showMainWindow()` function. See [here](https://gist.github.com/hyOzd/a0a2b089cd6a001c9aa0#file-mpexport-py-L131). As you can see I deal with the mainWindow of the freecad by simply hiding it. And I think this is an ugly hack.

There is also a limitation. I couldn't find how to color faces of a part different colors. Freecad has a dialog that can do this. So it's certainly supported. But there seems to be no API access for this. As a solution I've added the `fuse` parameter. `export()` function fuses the different solids together to create a single solid. Freecad preserves the colors of the faces after a fuse operation, so we get a multi colored single solid.

I don't like the idea of integrating this code into cq, because it creates a dependency to an ugly and unpredictable API. On the other hand it's an important feature that people are looking for. I think it should be added as a carefully isolated module of cq.

dcowden, 2015-11-27:
Yeah I didn't understand that integrating it would result in a dependency
on the gui object. I agree, that would be a nasty dependency.
​ You're right that we shouldnt integrate that unless we can figure out how
to support color another way.

i spent a bit looking at the opencascade documents. I think it is possible,
but as usual there is really no documentation at all on how to do it. The
class to add color to a step file is here:

http://www.opencascade.com/doc/occt-6.9.1/refman/html/class_s_t_e_p_construct___styles.html

but its not at all obvious how to use it. Best i can tell, you construct it
and attach it to a
XSControl_WorkSession
<http://www.opencascade.com/doc/occt-6.9.1/refman/html/class_x_s_control___work_session.html>
​, and then you use that worksession when exporting using
STEP_ControlWriter:

http://www.opencascade.com/doc/occt-6.9.1/refman/html/class_s_t_e_p_control___writer.html
​
​

​but to do that we'd be needing to bypass FreeCAD and use OCC directly.​

huskier, 2015-11-28:
@hyOzd  The module works perfectly for my application. Thank you very much for your endeavour.

@dcowden 
>but to do that we'd be needing to bypass FreeCAD and use OCC directly.​

The following page says that pythonocc's functions can be called from FreeCAD.
http://www.freecadweb.org/wiki/index.php?title=PythonOCC

PythonOCC is really a good thing to learn.

dcowden, 2015-11-28:
Indeed, there has been a lot of talk about having cq use pythonocc
directly. But that is a lot of work that Jeremy and I cannot tackle alone.

huskier, 2015-11-29:
@dcowden  I hope that I could help a little bit. I like CQ.

dcowden, 2015-11-29:
Ok awesome! Thanks for helping. We are currently getting our documentation
done, so we have a stable base before we start developing in a major way.
The documents have been lacking for a long time,  and I'm just now getting
some time to address thrm

huskier, 2015-12-01:
@hyOzd  Thanks for your tricky solution.
Once we initiated the FreeCADGui, how do we Close/Quit/Exit the GUI?
FreeCADGui.showMainWindow()
FreeCADGui.getMainWindow().hide() # prevent splash of main window

On WinXP, the command window got frozen after doing the export operation, and we have to force closing the frozen window. There is no problem on Linux(I use Debian Jessie.).

hyOzd, 2015-12-01:
Hmm, I don't have access to WinXP. So I cannot reproduce your issue. 

* Do you ever see the FreeCAD window appear? If so `getMainWindow().hide()` isn't effective at all.
* Is your version of FreeCAD up to date? Have you tried daily builds?
* Which interpereter are you using? Are you using FreeCAD itself to run the python file?

You can try adding this line at the end of script to close the window;

    FreeCADGui.getMainWindow().close()

huskier, 2015-12-01:
@hyOzd   Thanks for your help.
>Do you ever see the FreeCAD window appear? If so getMainWindow().hide() isn't effective at all.

Just a splash. I think the getMainWindow().hide() is effective.
>Is your version of FreeCAD up to date? Have you tried daily builds?

It's 0.15, which is the latest stable version.
>Which interpereter are you using? Are you using FreeCAD itself to run the python file?

I use Anaconda-2.1.0, and the python version is 2.7.8. FreeCAD is just used as a library.

I will try FreeCADGui.getMainWindow().close() tomorrow, since my home computer runs Debian.

hyOzd, 2015-12-01:
> Just a splash. I think the getMainWindow().hide() is effective.

In my case I don't even see the splash. Can you try daily build of the freecad?

huskier, 2015-12-02:
@hyOzd 
FreeCADGui.getMainWindow().close() could let the program terminate, but an "Unsaved Document" appears at the end of the program, and we have to click the mouse to close the dialog. Maybe we need to close the window in a force way, and discard the "Unsaved Document" automatically.

I use the daily build V0.16. The splash is still there, very shortly.

closed 2015-12-05 by huskier