dcowden/cadquery #129: Modify CadQuery Script Syntax to support consistent execution from different environments

dcowden, 2015-12-05:
We are planning to change the syntax of cadquery scripts, so that we can better leverage common script syntax between cadquery-freecad-module, our documentation, and parametricparts.com

This document describes the motivation for the changes.  We want to involve the community on this change, so please feel free to comment on this thread after reading the specification below.

These changes will NOT affect the syntax of the CQ objects themselves. It will only affect the elements of the script involved with the 'container', such as import statements, the show() method in cadquery-freecad-module scripts, the result variable in sphinxdoc examples, and the build() method on parametricparts scripts.

The 'open comment period' is from now until 12/12/2015. Please comment by then if you are interested in how this will work.

https://docs.google.com/document/d/1pcWnjkSAHZPoFU_4oMJRbEDpTynAYHgPRIVA7qo-1S4/edit?usp=sharing

dcowden, 2015-12-06:
Just to clarify: the document contains a bit of info, but the discussion should focus on the questions raised in the "Questions" section:

1. Should the CQGI automatically do “import cadquery” or “from cadquery import *” ?
2. Should CQGI-compatible build scripts separate build logic into a build() method, or should the script use a build_result() method to export objects?
3. Should the CQGI include validation and parameter metadata, or should it be limited to just the build_result method

jmwright, 2015-12-07:
Regarding question 1, if we ever want to rebrand the CadQuery FreeCAD Module (CFM) to be a general use scripting module (it can already be used that way), we wouldn't want to be automatically importing cadquery for every user. However, we could address this by adding a cadquery-autoimport setting for the module and default it to false if we rebrand.

dcowden, 2015-12-08:
alternatively, we could make this a responsibility of the CQGI...

jmwright, 2015-12-08:
How would that work? Would there just be a property of the CQGI that the user could set?

dcowden, 2015-12-08:
Since cqgi is cadquery specific,  it could just always import cadquery for
the user.

Though right now cq has the freecad provider hard coded, I think the cqgi
can also provide registration for cq implementations also. Not sure yet how
that would work exactly,

jmwright, 2015-12-08:
If there are validation errors when the CQGI implementation reaches build_object, should it refuse to execute build_object and just report the errors?

dcowden, 2015-12-08:
good question.  in short, i think yes.

if CQGI supports the notion of validation, then there are two ways
validation is run:

(1) *'Validation Mode'. *CQGI's intention is ONLY to validate. In this
case, the container is wanting to give feedback to a user about whether a
set of inputs is valid, and wishes to do this as quickly as possible-- IE,
avoiding the time required to run the entire script.

(2)*'Execution Mode'* CQGI's intention is to execute a build. Since user
inputs were already validated, this will, in nearly if not all cases,
result in redundant validation. But since validation is assumed to be fast,
and since it actually takes _extra_ work NOT to run the validation ( since
it is part of the script), then its best to accept the redundant
execution.  This provides another benefit: the execution environment is
allowed to skip validation in some cases, if it wishes, since they will be
run later anyway when the model builds.

In validation mode, add_error will not stop execution, because we want to
present all of the errors to the user at once. The portion of the script
that represents initialization and validation runs in its entirety, and
accumulates all of the errors.

In contrast, in execution mode, any call to add_error should stop further
execution. That's because we know building the entire model is pointless.
Continued execution would only provide benefit in gathering more errors.
But if that was the intent, the container should have used validation mode
first anyway.

jmwright, 2015-12-08:
So, there would now be a Validate menu option in addition to the Execute option that already exists in CFM?

dcowden, 2015-12-08:
right. it definitely makes a bit more sense when coupled with a 'form'
where the user is supplying parameter inputs. It seems much less likely to
be useful for the script author-- which is almost certainly the role you
have using CFM.

hyOzd, 2015-12-12:
Guys, sorry I'm a little late. I read this the day (my night :D ) @dcowden  published. But never had time to response.

If I understood it right, this won't have any effect on usage of cadquery from regular python code. Can you confirm? At the moment I can run a python script from command line and produce output, I don't have to abide by any syntax (other than python). Am I going to have to call a `build_object()` method when this proposal is implemented? I just want to understand if I will have to change my current scripts.

--------

About the "Implementation Details" section; I don't like the idea of separating the script into two sections, or doing some sort of parsing. I would like it if we could stay on python, and not implement sctrict rules on top of it. I'm worried this will put some constraint on script developer. I can't name it but it looks like this may prevent us from using some basic features of python.

I think having separate `validate()` and `build()` functions in the script is a cleaner/better idea.

---------

I don't like the naming of `define_config()`. As a non-native speaker I always get confused when I see "configuration" used as a "preset of parameter"s. "configuration" and "parameter" sound too similar to me. On the other hand; I think just using the word "preset" (for ex: `define_preset()`) seems like a better idea to me. What do you think?

-------

How about OOP? Have you guys seen my [e3dmg](https://github.com/hyOzd/ecad-3d-model-generator) project in which I code "paremetric generator scripts" for electronic component models? Let me tell you about it's structure, which is pretty simple.

"generator"s are classes that inherit the [`Generator`](https://github.com/hyOzd/ecad-3d-model-generator/blob/master/e3dmg/generator.py) class. At the moment `Generator` class doesn't provide any methods itself. It's just there to determine if ,a class defined in a module, is a generator. A generator must implement 2 methods:

* `__init__()` : should accept the generator specific parameters
* `generate()` : builds the model and returns a `ComponentModel` instance

`ComponentModel` is a simple container that holds multiple cq objects and colors associated with them. Even a list of tuples would do.

Here is a simple box model generator example [from e3dmg](https://github.com/hyOzd/ecad-3d-model-generator/blob/master/e3dmg/generators/box.py):

```
class BoxGen(Generator):

    def __init__(self, l, w, h, color):
        self.l = l
        self.w = w
        self.h = h
        self.color = color

    def generate(self):
        """Returns a ComponentModel."""
        b = cq.Workplane("XY").box(self.l, self.w, self.h)
        model = ComponentModel()
        model.addPart(b, self.color, "Body")
        return model
```

An example of how to use this from a python console:

    from boxgen import BoxGenerator
    box = BoxGenerator(10,20,30, (1,0,0))
    model = box.generate()
    export(model)

One thing I like about this is that I can use python objects (instances of generator classes as "preset configs". Please see the [example database file](https://github.com/hyOzd/ecad-3d-model-generator/blob/master/e3dmg/database/test.py#L37) from e3dmg.

How would this architecture be turned into CGQI: just add a method named `validate()` that will check the parameters and some form of error reporting and you are good to go.

Don't get me wrong I'm not saying do it my way. Let me list some advantages and disadvantages from my point of view:

Advantages:
* cleaner script files from a python developer perspective, you can even define multiple generators/builders in a single file
* power of inheritance allow you to subclass an existing generator, and create a modified version of it without duplicating code. For example look at my [qfn generator](https://github.com/hyOzd/ecad-3d-model-generator/blob/master/e3dmg/generators/qfn.py). `BaseQfn` is a usable generator, `QFNGen` and `MQFNGen`are just sub generators, that make it slightly easier to define `BaseQfn` paremeters.
* easier implementation, python's `import` module will do most of the job

Disadvantages:
* it's ugly. I'm worried some beginners would just run away at the sight of `class` keyword.
* it's really ugly. At least my way of doing it (*) is really ugly. See [this](https://github.com/hyOzd/ecad-3d-model-generator/blob/master/e3dmg/generators/qfn.py#L68). Since the `__init__()` method stores the generator parameters as "object attributes", the `generate()`  method has to access them via the object instance: aka `self`. This will fill the script with the instances of `self.x`, `self.y`, `self.width` etc. As you can see I try to solve this -partly- by assigning all parameters to function local variables at the beginning of the script. Typing this isn't an issue for me. I do it quickly with some keyboard trickery (yay emacs!). But it's still ugly, though at least "it's ugly in one place" : )

(*) there are some alternatives that I didn't have time to look into, but they require setting some rules on generator defining

I'm sorry, this turned out to be longer than I expected. Thank you if you read it. I would be glad if I'm even a little bit of help.

dcowden, 2015-12-12:
​Hi, Hasan, thanks for chiming in!  I've put responses below.​

On Sat, Dec 12, 2015 at 10:59 AM, Hasan Yavuz ÖZDERYA <
notifications@github.com> wrote:

> Guys, sorry I'm a little late. I read this the day (my night :D ) @dcowden
> <https://github.com/dcowden> published. But never had time to response.
>
​Thanks for jumping in!​ Your e3dmg libary is cool.

> If I understood it right, this won't have any effect on usage of cadquery
> from regular python code. Can you confirm? At the moment I can run a python
> script from command line and produce output, I [...]
​You are correct. if you use cadquery as a library, you can do whatever you
want. build_object() must be called only when you are running cadquery
scripts inside of an environment that needs to know about an object.  For
cadquery-freecad-module, for example, build_object replaces the show()
method.​

> ------------------------------
>
> About the "Implementation Details" section; I don't like the idea of
> separating the script into two sections, or doing some sort of parsing. I
> would like it if we [...]
​I understand that view point. In fact, that's how parametricparts.com
works now. The concern is that requiring a build() method is a fairly large
hurdle for people to get over.  From a programmers' viewpoint, clearly
having a build() method and a validate() method is better. But the concern
is that lots of people won't make that first hurdle at all.

Have a look, for example, at the format for the lego example there. You'll
see that the format is very, very similar to what you've done with
generators:  http://www.parametricparts.com/parts/vqb5dy69/
​
There we don't require that scripts subclass anything, but we require a
build() method, which is the same as the generate() method in ed3mg.  We
compile the script as a module, and then gain access to the variables
defined that way. This approach has all of the advantages and disadvantages
that the e3dmg approach does.

> ------------------------------
>
> I don't like the naming of define_config(). As a non-native speaker I
> always get confused when I see "configuration" used as a "preset of
> parameter"s. "configuration" [...]
preset is fine with me. How do you like describe_parameter() for adding
extra metadata about a variable? or perhaps describe_variable() is clearer?
​

> ------------------------------
>
> How about OOP? Have you guys seen my e3dmg
> <https://github.com/hyOzd/ecad-3d-model-generator> project in which I
> code "paremetric generator scripts" for electronic [...]
Interesting! There are many similarities between what you iterated to and
what we are proposing. In fact, your export() method works exactly the same
as the proposed build_object() method. in the end, you have to have a way
to pass a completed object to the build environment.

You have used the 'build method' strategy for the script-- though you have
additionally made it more obvious that the author must have a 'build'
method'( in your case, generate()), because you have required that the
author create a subclass of Generator.

> Don't get me wrong I'm not saying do it my way. Let me list some
> advantages and disadvantages from my point of view:
>
> Advantages:
>
>    - cleaner script files from a python developer perspective, you [...]
​You have stated the advantages and disadvantages perfectly. I agree with
all of your statements. In fact, for parametricparts.com, i selected the
path you recommend, because of the advantages you listed.

​I think the big question is: who are cq scripts operating in a CQGI
environment really targeted for? Clearly, programmers using cadquery as an
embedded library can be expected to be solid programmers, but in that case,
we are not applying any constraints.

That brings up the question: how technical are the target audience for
cadquery scripts inside of an execution environment (
cadquery-freecad-module or parametric parts) likely to be?

If the answer is 'they are all experienced python programmers', then i
think a solution like you describe is best. In fact, I used to agree with
you, which is why parametricparts.com scripts look just as you suggest.
But then Jeremy launched cadquery-freecad-module, and used very simple
script format that's more approachable for less competent developers.  To
be honest, he seems to have gotten more uptake than parametric parts has,
so i'm thinking that the average experience level of the 'typical' author
creating scripts is lower than i first thought.

In short, your technical arguments are right-- the question is whether they
are the right fit for the target audience  ( and, perhaps, who we _want_
the target audience to be).  My assumption has been that for really good
programmers, they will use cadquery however they like, so a more
approachable, easier script format is a better move so we can help the less
skilled users.
​

hyOzd, 2015-12-12:
> preset is fine with me. How do you like describe_parameter() for adding extra metadata about a variable? or perhaps describe_variable() is clearer?

If it's an input variable, it should be defined with `describe_parameter` in my opinion. How about dropping the `describe` part. This naming set looks more intuitive and cleaner to me:

* build( shape ) # output a shape
* error( message ) # indicate an error and add it to a list for handling later
* parameter( param, metadata ) # define parameter metadata for a param
* preset( name, dictionary ) # define a pre-set configuration

> That brings up the question: how technical are the target audience for cadquery scripts inside of an execution environment ( cadquery-freecad-module or parametric parts) likely to be? If the answer [...]

I understand your concerns. But I'm not sure the reason parametricparts.com didn't take up was its interface. Let me share my experience: 

1. I've discovered parametricparts.com. 
2. Got really excited by the idea (jquery based syntax and python). 
3. Opened an example
4. Clicked build
5. It didn't work. Maybe it was just taking too long. I wasn't sure. I thought it wasn't a well maintained project. Decided "Fail" and moved on. (*)

Later on I discovered cadquery as a python library and loved its freecad integration. It just worked. It was running local, so I didn't have to wait for download times. I really like the idea of parametricparts from a user perspective. You go to a page enter your parameters and you download your model. You don't have to install anything, you don't have to chase down library dependencies. It's great! But from a script developer perspective. Not so much. You have to frequently make changes and try them out during development. A development environment running on cloud isn't viable in this case. Note that by download times I don't mean minutes. Thank god our connection isn't that slow. But when you are making iterative changes, waiting more than a couple seconds can be distracting, aka: "let me check reddit a second..... sh.t it's been 30 minutes"

That's just my experience though. I certainly understand and agree with you that even a function definition requirement can scare people away. 

I'm wondering how do you plan to advertise cadquery if these changes are implemented? Do you sell it as a script language of its own (maybe you point that it's a subset of python so you can use python syntax, but there are some limitations/rules) or do you advertise it as a python library?

(*) PS: just tried [gear generator](http://www.parametricparts.com/parts/8npxd6ur/) script on parametric parts; I'm getting "Error Rendering: Authentication Required"

-----

Is using comments to define script parameters etc. an option? [Like thingiverse does](http://customizer.makerbot.com/docs). So that it's not required to run the script to get the parameter information, just parse it with some regex... Not sure if this is a good idea, just pointing out an alternative.

dcowden, 2015-12-12:
Hi,

* yes, a local IDE is important, as you point out. I got several comments,
as you described, that using PParts as an IDE is frustrating.  the
iterations are too slow.

Ironically, that's exactly why we need to do this CQGI standardization. The
goal is to allow using the cadquery-freecad-module for development, and
then upload a script to parametric parts.  For that to work, though,
 changing the syntax of the script isnt really going to work.

I think at this point we're on the same page for what's needed, though.  If
CQGI standardizes script syntax, you can use cadquery-freecad-module for
local development, and then, when complete, upload it for use on a web
based platform, so that others can easily customize objects.

the extra metadata definitions are not needed for authors-- they become
necessary when the script is uploaded for more use by others.

regarding defining parameters, yes, we definitely discussed using source
code comments like thingiverse does.  That's a possibility, but i really
don't like that.  its actually more work for the author to understand the
magic syntax that will work, and it requires a bunch of custom parsing on
the server.  The proposed CQGI solution is more friendly to authors, i
think.

Regarding future plans:
cadquery is and always will be open source., and will always be designed to
be a library for use from within python.  the CQGI, a new module, will be
advertised as the preferred way to build tools that allow users to run
cadquery scripts. Two such tools are cadquery-freecad-module and
parametricparts.com.   In the long term, the plan is to provide an upload
feature in cadquery-freecad-module to upload a model directly to
parametricparts.com.

hyOzd, 2015-12-13:
> Regarding future plans: cadquery is and always will be open source., and will always be designed to be a library for use from within python. the CQGI, a new module, will be advertised as the [...]

Maybe I shouldn't have used words "sell, advertisement" up there. I was looking for the word "present" : )

Note: @dcowden  some of your comments are lost in the formatting. They are shown as 'quote' in github page, so they are easy to miss. They look fine on the e-mail  though.

closed 2016-05-20 by jmwright