dcowden/cadquery #131: Incorrect Center Of BoundBox

huskier, 2015-12-26:
Considering the following code to calculate the Center Of BoundBox, the correct result should be 14.0 in X direction, but the code gives 13.938211451097892. I tracked the CQ's BoundingBox, and find it is directly wrapped from FreeCAD's BoundingBox. I've lost here, and don't know how to debug this problem. Any ideas? Thanks.

```
import cadquery
from Helpers import show
import logging

logging.basicConfig(level=logging.ERROR,
                format='%(asctime)s %(filename)s[line:%(lineno)d] %(levelname)s %(message)s',
                datefmt='%a, %d %b %Y %H:%M:%S',
                filename='c:/CenterOfBoundBox.log',
                filemode='w')

result0 = (cadquery.Workplane("XY")
                 .moveTo(10,0)
                 .lineTo(5,0)
                 .threePointArc((3.9393,0.4393),(3.5,1.5))
                 .threePointArc((3.0607,2.5607),(2,3))
                 .lineTo(1.5,3)
                 .threePointArc((0.4393,3.4393),(0,4.5))
                 .lineTo(0,13.5)
                 .threePointArc((0.4393,14.5607),(1.5,15))
                 .lineTo(28,15)
                 .lineTo(28,13.5)
                 .lineTo(24,13.5)
                 .lineTo(24,11.5)
                 .lineTo(27,11.5)
                 .lineTo(27,10)
                 .lineTo(22,10)
                 .lineTo(22,13.2)
                 .lineTo(14.5,13.2)
                 .lineTo(14.5,10)
                 .lineTo(12.5,10 )
                 .lineTo(12.5,13.2)
                 .lineTo(5.5,13.2)
                 .lineTo(5.5,2)
                 .threePointArc((5.793,1.293),(6.5,1))
                 .lineTo(10,1)
                 .close())
result = result0.extrude(100)

logging.error('debuging center of boundbox......')
logging.error('xmin = ' + str(result.BoundingBox().xmin))
logging.error('xmax = ' + str(result.BoundingBox().xmax))
logging.error('xlen = ' + str(result.BoundingBox().xlen))
logging.error('CenterOfBoundBox is ' + str(result.BoundingBox().center))

show(result)
```

jmwright, 2015-12-28:
I had to add the `val()` calls shown below to get your code to work, but I get the same result. There's also a bug in FreeCAD's mirror functionality that I'm going to have to dig into at some point. The only recourse we have right now is to report these bugs to the FreeCAD devs. At some point I hope to set up a FreeCAD development environment and fix some things like this, but I don't know when I'll get around to doing that.

```python
Mon, 28 Dec 2015 12:27:50 tmp5rTd7R[line:39] ERROR debuging center of boundbox......
Mon, 28 Dec 2015 12:28:12 tmpIx8OPn[line:39] ERROR debuging center of boundbox......
Mon, 28 Dec 2015 12:28:12 tmpIx8OPn[line:40] ERROR xmin = -0.123577097804
Mon, 28 Dec 2015 12:28:12 tmpIx8OPn[line:41] ERROR xmax = 28.0
Mon, 28 Dec 2015 12:28:12 tmpIx8OPn[line:42] ERROR xlen = 28.1235770978
Mon, 28 Dec 2015 12:28:12 tmpIx8OPn[line:43] ERROR CenterOfBoundBox is Vector (13.938211451097892, 7.499999999999999, 50.0)
```
```python
import cadquery
from Helpers import show
import logging

logging.basicConfig(level=logging.ERROR,
                    format='%(asctime)s %(filename)s[line:%(lineno)d] %(levelname)s %(message)s',
                    datefmt='%a, %d %b %Y %H:%M:%S',
                    filename='c:/CenterOfBoundBox.log',
                    filemode='w')

result0 = (cadquery.Workplane("XY")
           .moveTo(10,0)
           .lineTo(5,0)
           .threePointArc((3.9393,0.4393),(3.5,1.5))
           .threePointArc((3.0607,2.5607),(2,3))
           .lineTo(1.5,3)
           .threePointArc((0.4393,3.4393),(0,4.5))
           .lineTo(0,13.5)
           .threePointArc((0.4393,14.5607),(1.5,15))
           .lineTo(28,15)
           .lineTo(28,13.5)
           .lineTo(24,13.5)
           .lineTo(24,11.5)
           .lineTo(27,11.5)
           .lineTo(27,10)
           .lineTo(22,10)
           .lineTo(22,13.2)
           .lineTo(14.5,13.2)
           .lineTo(14.5,10)
           .lineTo(12.5,10 )
           .lineTo(12.5,13.2)
           .lineTo(5.5,13.2)
           .lineTo(5.5,2)
           .threePointArc((5.793,1.293),(6.5,1))
           .lineTo(10,1)
          .close())
result = result0.extrude(100)

logging.error('debuging center of boundbox......')
logging.error('xmin = ' + str(result.val().BoundingBox().xmin))
logging.error('xmax = ' + str(result.val().BoundingBox().xmax))
logging.error('xlen = ' + str(result.val().BoundingBox().xlen))
logging.error('CenterOfBoundBox is ' + str(result.val().BoundingBox().center))

show(result)
```

huskier, 2015-12-29:
@jmwright Thanks for your trying.
After extruding the wires, a solid is obtained. When I calculate the boundbox with FreeCAD GUI, the result is correct(28 in X direction). 

I am trying to create an example to reproduce this issue in FreeCAD, and then report the bug to the FreeCAD devs if it is really bug in FreeCAD.

jmwright, 2015-12-29:
Are you using FreeCAD scripting to obtain the bounding box, or the GUI?

huskier, 2015-12-29:
@jmwright  The GUI. Maybe I should try to use the script to obtain the bounding box.

huskier, 2015-12-30:
There is an answer from here http://www.freecadweb.org/tracker/view.php?id=2382.

They said that it is not a bug. Now the problem is to integrate tessellate function into the BoundingBox function.

huskier, 2015-12-30:
I've found the tessellate function.
tessellate() : Gui::Tessellator , Part::TopoShapePy

The implement is like following:
PyObject* TopoShapePy::tessellate(PyObject *args)
{
    //implementing......
}

Still don't know how to integrate into the CQ.

huskier, 2015-12-30:
Modifying the function BoundingBox in the Shapes.py as following could solve the problem. Just add a line of code before return BoundBox. Do we need to pass the precision parameter into the BoundingBox function?

```
    def BoundingBox(self):
        self.wrapped.tessellate(0.0001)
        return BoundBox(self.wrapped.BoundBox)
```

dcowden, 2015-12-31:
that implementation looks right to me, assuming that it produces the
expected behavior.

closed 2016-01-21 by huskier