dcowden/cadquery #135: Test error for function testMultiFaceWorkplane

huskier, 2016-01-20:
I've got a test error when I run "python runtests.py". It is said that Part.Face has no attribution of Mass.

I will paste the traceback error here tomorrow.

dcowden, 2016-01-20:
that's odd. the tests pass for me and also in travis with master.
what freecad version are you using?

huskier, 2016-01-20:
0.15 On Windows XP.

dcowden, 2016-01-20:
Hmm ok. I don't have a way to duplicate that environment easily atm.

Can you try freedcad 0.16? That's what I ran a my tests with

huskier, 2016-01-21:
The test passed under FreeCAD_0.16.5958_x86 on Windows XP, but not under Version 0.15.

However, I still get some errors output at the end of the test.

...Error Executing Script:
Traceback (most recent call last):
  File "E:\programming\github\cadquery\cadquery\cqgi.py", line 101, in build
    exec (c, env)
  File "<cqscript>", line 2, in <module>
ValueError: ERROR
Full Text of Script:

raise ValueError("ERROR")

...Error Executing Script:
Traceback (most recent call last):
  File "E:\programming\github\cadquery\cadquery\cqgi.py", line 95, in build
    self.set_param_values(build_parameters)
  File "E:\programming\github\cadquery\cadquery\cqgi.py", line 125, in set_param
_values
    p.set_value(v)
  File "E:\programming\github\cadquery\cadquery\cqgi.py", line 250, in set_value

    .format(str(new_value), self.name))
InvalidParameterError: Cannot set value 'a string' for parameter 'h': parameter
must be numeric.
Full Text of Script:

h = 20.0
build_object(h)

.Error Executing Script:
Traceback (most recent call last):
  File "E:\programming\github\cadquery\cadquery\cqgi.py", line 95, in build
    self.set_param_values(build_parameters)
  File "E:\programming\github\cadquery\cadquery\cqgi.py", line 122, in set_param
_values
    raise InvalidParameterError("Cannot set value '%s': not a parameter of the m
odel." % k)
InvalidParameterError: Cannot set value 'w': not a parameter of the model.
Full Text of Script:

h = 20.0
build_object(h)

....Error Executing Script:
Traceback (most recent call last):
  File "E:\programming\github\cadquery\cadquery\cqgi.py", line 105, in build
    raise NoOutputError("Script did not call build_object-- no output available.
")
NoOutputError: Script did not call build_object-- no output available.
Full Text of Script:

h = 20.0

...
----------------------------------------------------------------------
Ran 139 tests in 9.028s

OK

dcowden, 2016-01-21:
> File "", line 2, in

yeah those are ok. they are traces from the tests when i intentionally
create error states. I probably should clean up those traces so they dont
appear during the tests, but they are harmless.

So, back to your issue, at some point along the way, we've introduced a FC
0.15 incompatibility. i have no idea when that happened. But i'm not sure
how important it is to support the older version at this point. what do you
guys think?

huskier, 2016-01-21:
I hope that FC iterate more faster, so most of the linux distros could include the most latest FC, since it is a really technical work to compile FC from the source code.

dcowden, 2016-01-21:
thats true. however, there is a ppa you can use to install FC 0.16 on
ubuntu without compiling. That's how i did it.  I can't remember where it
was, but i think @jmwright knows because he pointed it out to me.

jmwright, 2016-01-21:
The problem with dropping support for 0.15 is that the Windows version that you download and install is still 0.15. I don't think we can dump it yet.

dcowden, 2016-01-21:
hmm ok good call. when i get a chance, i'll downgrad to 0.15 and see if i
can duplicate this problem.

huskier, 2016-01-21:
@dcowden & @jmwright I just updated my pull request created twenty days ago. Please help review it if you get a chance.

BTW, I also updated the extruding issue.

closed 2016-01-22 by huskier