dcowden/cadquery #137: About CQ 2.0

huskier, 2016-02-29:
> In practice its not clear at all what the 2nd cad kernels would even be.

If we could have the 2nd kernel, we can develop the CQ with two kernels at parallel. I've googled the web, and  found that only OCE has a good shape as an open source CAD kernel. Could we get a kernel source from ACIS or Parasoid guys, even an very old one?

dcowden, 2016-02-29:
you are right-- ACIS and parasolid are the '2nd kernels' i had in mind.
there are no open source alternatives to OCE that are worth considering at
this time. I really _want_ to consider verbnurbs or solvespace, but they
are simply not anywhere close to what we need.

Developing an interface to parasolid or ACIS would be a move designed to
'commericalize' CQ technology. Since you have to pay for the kernel, i
assume that project would not be open source.

My thought was to license parasolid or ACIS, build a CQ wrapper, and then
build a web-based solution. OnShape has done precisely that.

huskier, 2016-02-29:
> My thought was to license parasolid or ACIS, build a CQ wrapper, and then build a web-based solution. OnShape has done precisely that.

I've tried OnShape, a kind of Cloud CAD, which is an interactive operational CAD like most of the desktop CADs. CQ, a scripting CAD or a batching CAD, is in a very different domain as those ones.

huskier, 2016-02-29:
@dcowden 
Do you have applied for a patent for the CQ technology? Or do you write some academic papers on this technology?

dcowden, 2016-02-29:
yes definitely. I dont really view OnShape as similar to CQ-- The
similarity was only in the licensing model. they have licensed the
parasolid kernel, and then used a web deployment to avoid the need to
charge users for a copy of parasolid.  We could do the same thing.

dcowden, 2016-02-29:
Hah, no academic papers-- i'm not that smart!  I guess i just happened to
be the 'right guy' to invent CQ-- in my day job i am a programmer, so i was
familiar with jquery and fluent apis and all that stuff, but my education
is actually in Mechanical Engineering-- where i worked for long enough to
learn CAD and actually making things in real life ( I worked at Pratt and
Whitney for a while ).  CQ seemed obvious to me.

I have in fact filed for a patent in the US. Software patents are really
hard to get though. I really doubt i'll be able to get it.

closed 2016-04-04 by jmwright