dcowden/cadquery #138: How to explicitly add a wire into a Workplane object?

huskier, 2016-03-12:
Here the question:

I have a Wire (CQ) object, and want to add it into a Workplane object, and then extrude from it.

I tried to add an "addWire(self, wire)" function into the Workplane class, the code looks like this:

```
    def addWire(self, wire):
        """
        """
        r = self.newObject([wire])
        r.ctx.pendingWires = []
        r._addPendingWire(wire)
        return r  
```

But I failed. I read the "consolidateWires(self)" function, tried many ways, still no success.

Any ideas?

dcowden, 2016-03-12:
addWire should work. What more specifically fails?

huskier, 2016-03-13:
@dcowden  OK, here is the whole thing. Several month ago, I asked you whether we can learn from a existing model, and then extrude from the learned wire. Now, I could do this with PythonOCC, but I still want to do the thing in CQ. Therefore, I want to import the leaned "Wire" from PythonOCC through 
Part.__fromPythonOCC__()...
CQ.fromFreeCAD()...

I could import the shape into CQ, but when I do future operations, error occurs.

Here is the code:
The code can also be found @ 
https://github.com/huskier/images/blob/master/Hybrid_CQ_FreeCAD_PythonOCC.py
https://github.com/huskier/images/blob/master/SectionBar_ZDir.step
I run the code on Fedora 23 (64bit) with 
FreeCAD 0.16
PythonOCC 0.16.3
and Anaconda 2.5.0

```
from __future__ import print_function

import sys
sys.path.append('/usr/lib/freecad/bin')

import cadquery as cq

from cadquery import *

import FreeCAD

import Part as FreeCADPart

from OCC.BRep import BRep_Tool

from OCC.TopAbs import (TopAbs_VERTEX, TopAbs_EDGE, TopAbs_FACE, TopAbs_WIRE,
                        TopAbs_SHELL, TopAbs_SOLID, TopAbs_COMPOUND,
                        TopAbs_COMPSOLID)
                        
from OCC.TopExp import TopExp_Explorer, topexp_MapShapesAndAncestors
from OCC.TopTools import (TopTools_ListOfShape,
                          TopTools_ListIteratorOfListOfShape,
                          TopTools_IndexedDataMapOfShapeListOfShape)
from OCC.TopoDS import (topods, TopoDS_Wire, TopoDS_Vertex, TopoDS_Edge,
                        TopoDS_Face, TopoDS_Shell, TopoDS_Solid,
                        TopoDS_Compound, TopoDS_CompSolid, topods_Edge,
                        topods_Vertex, topods_Wire, TopoDS_Iterator)                        

from OCC.STEPControl import STEPControl_Reader
from OCC.IFSelect import IFSelect_RetDone, IFSelect_ItemsByEntity
from OCC.Display.SimpleGui import init_display

from OCCUtils.Construct import (make_closed_polygon, make_n_sided,
                                make_vertex, make_face)
from OCCUtils.Topology import WireExplorer, Topo, dumpTopology

from OCC.BRepBuilderAPI import (BRepBuilderAPI_MakeEdge,
                                BRepBuilderAPI_MakeFace,
                                BRepBuilderAPI_MakeWire)
from OCC.gp import gp_Pnt, gp_Vec
from OCC.BRepPrimAPI import BRepPrimAPI_MakePrism                                

from OCC.Display.SimpleGui import init_display
display, start_display, add_menu, add_function_to_menu = init_display()

lstWires = []

# CQ2PythonOCC()......
# PythonOCC2CQ()......

def mydumpTopology(shape, level=0):
    """
     Print the details of an object from the top down
    """
    brt = BRep_Tool()
    s = shape.ShapeType()
    if s == TopAbs_VERTEX:
        pnt = brt.Pnt(topods_Vertex(shape))
        print(".." * level  + "<Vertex %i: %s %s %s>" % (hash(shape), pnt.X(), pnt.Y(), pnt.Z()))
    else:
        print(".." * level, end="")
        if s == TopAbs_WIRE:
            lstWires.append(shape)
            print("shape type is TopAbs_WIRE...")
    it = TopoDS_Iterator(shape)
    while it.More():
        shp = it.Value()
        it.Next()
        mydumpTopology(shp, level + 1)

step_reader = STEPControl_Reader()

status = step_reader.ReadFile('/home/huskier/Public/cad/SectionBar_ZDir.step')

if status == IFSelect_RetDone:  # check status
    failsonly = False
    step_reader.PrintCheckLoad(failsonly, IFSelect_ItemsByEntity)
    step_reader.PrintCheckTransfer(failsonly, IFSelect_ItemsByEntity)

    ok = step_reader.TransferRoot(1)
    _nbs = step_reader.NbShapes()
    aResShape = step_reader.Shape(1)
else:
    print("Error: can't read file.")
    sys.exit(0)

mydumpTopology(aResShape)

print(len(lstWires))

face = BRepBuilderAPI_MakeFace(topods_Wire(lstWires[-1]))

starting_point = gp_Pnt(0., 0., 0.)
end_point = gp_Pnt(0., 0., 100.)
vec = gp_Vec(starting_point, end_point)

prism = BRepPrimAPI_MakePrism(face.Shape(), vec).Shape()

freecadObj = FreeCADPart.Wire(FreeCADPart.__fromPythonOCC__(lstWires[-1]))

print(str(type(freecadObj)))

cqObj = cq.CQ(freecadObj)

cqObj = Wire(cqObj)

print(str(type(cqObj)))

wkplane = cq.Workplane("XY").addWires(cqObj).extrude(20)
#print(str(type(wkplane)))
#print(wkplane.wires().size())

display.DisplayShape(prism, update=False) 

start_display()

```

and the error reads:

> <type 'Part.Wire'>
> <class 'cadquery.freecad_impl.shapes.Wire'>
> Traceback (most recent call last):
>   File "Hybrid_CQ_FreeCAD_PythonOCC.py", line 114, in <module>
>     wkplane = cq.Workplane("XY").addWires(cqObj).extrude(20)
>   File "/opt/anaconda2/lib/python2.7/site-packages/cadquery-0.4.0-py2.7.egg/cadquery/cq.py", line 2008, in extrude
>     r = self._extrude(distance)  # returns a Solid (or a compound if there were multiple)
>   File "/opt/anaconda2/lib/python2.7/site-packages/cadquery-0.4.0-py2.7.egg/cadquery/cq.py", line 2302, in _extrude
>     thisObj = Solid.extrudeLinear(ws[0], ws[1:], eDir)
>   File "/opt/anaconda2/lib/python2.7/site-packages/cadquery-0.4.0-py2.7.egg/cadquery/freecad_impl/shapes.py", line 848, in extrudeLinear
>     f = FreeCADPart.Face(freeCADWires)
> Part.OCCError: shape is not a wire

huskier, 2016-03-13:
My ultimate goal is to separate the learning and the using task, i.e. we first learned the "wire" from a model, and then stored the learned info into a file, or a python variable, then when we use the info we don't need to load the model again.

The SectionBar_Wire.step is what we learned from the SectionBar.step model.
https://github.com/huskier/images/blob/master/SectionBar_Wire.step

Next step, we need to import it as a CQ Workplane object with the wires we need. My problem is here. I cannot do it right now.

dcowden, 2016-03-13:
Wow, ok you are really diving into the OCC side a lot!

It looks like you are doing the cq steps right.. According to the
traceback, all went well until it hit the freedcad code.

Freedcad complained that the shape it got isn't a wire. There are a couple
of possibilies:

1. There is an incompatibility between the wires freedcad is using and OCC.
Are you using the OCC bundled with freedcad? That should work, but I have
not tried it.

2. Somehow you are passing a cq wire object, not the underlying freedcad
wire

I can't see the bug right off, and I see that you have debug testing the
types...

huskier, 2016-03-14:
> Are you using the OCC bundled with freedcad? That should work, but I have
> not tried it.

No, I installed freecad and OCC separately. 
For FreeCAD, I am following the instructions here:
https://copr.fedorainfracloud.org/coprs/przemof/FreeCAD/

And for OCC, I used the command:
conda install -c https://conda.anaconda.org/dlr-sc pythonocc-core

So, I think they are not bundled.

dcowden, 2016-03-14:
ah ok, i see. in that case, i think it is possible that the 'Wire' object
that is used by the FreeCAD OCC, and the one used with the copy of OCC you
have installed, might not be the same. IE, you have a reference to a C++
object in both cases, but since they are not the same class, they are not
compatible.

huskier, 2016-03-14:
That's weird, not expected. 
FreeCAD <--- OCC
PythonOCC <--- OCC

Maybe we need to compile all the sources from scratch, and that will be solve the imcompatible problem?

dcowden, 2016-03-14:
to be clear, that's just an idea-- i dont know for sure. it is very hard to
troubleshoot problems at the C++ layer

jmwright, 2016-04-04:
I'm really late to the party on this, but one thing that's caused me a lot of trouble in the past is version mismatches with OCC/OCE. The PythonOCC and OCC versions have to match or things just don't work right, and I suspect it's the same way with FreeCAD. There's also the layer of abstraction that FreeCAD has probably added on top of OCC for wires (etc) that could cause problems as well.

jmwright, 2016-04-27:
@huskier Any progress on this?

huskier, 2016-04-29:
@jmwright  Not really.

I'm keeping busy with the work and my family. I'll get back if I've got a time slot.

Thanks @jmwright.

dcowden, 2016-12-28:
hey, @huskier , not expediting, but if you are interested, I've created the beginnings of CQ 2.0. It has FreeCAD and pythonOCC support, and tests working.  Its not much, but its a start. 

The code is available on the 2_0_branch.  
The project is organized here: https://github.com/dcowden/cadquery/projects/1

Feel free to jump in! the current work is on implementing low-level objects ( Vector, plane, bounding box, etc ) on top of OCC

jmwright, 2017-10-28:
@bustier This may be what you're looking for. It's not clean in that it involves the use of FreeCAD objects, but it does work.

https://github.com/fragmuffin/cqparts/blob/develop/src/cqparts/types/threads/base.py#L184

closed 2017-10-28 by jmwright