dcowden/cadquery #139: [suggestion] CQ object <==>  PythonOCC object

huskier, 2016-03-13:
At present, we could do the exchange through:
CQ.toFreeCAD()
CQ.fromFreeCAD()  (CQ support natively, don't need the function)
FreeCADPart.__toPythonOCC__()
FreeCADPart.__fromPythonOCC__()

I think we could add another pair into the CQ for now and into the future:
CQ.toPythonOCC()
CQ.fromPythonOCC() (it is better to be explicitly, and also we could preserve the implicit cqObj = CQ(the pythonocc obj))

dcowden, 2016-03-13:
Im not sure about this. Of course it won't harm anything programmatically,
and I can see that it would be helpful in what you are working on now.

But the worry I have is that OCC is a very low level API. I am afraid that
exposing methods to work at that level are going to encourage development
to go in the wrong direction.  I don't think we want people to have to deal
with OCC or freedcad at all. Rather, we want to simplify things.  It seems
that whenever one would use methods to directly manipulate an OCC wire,
they are probably working around a limitation on cq. So why not simply
address the limitation instead?

huskier, 2016-03-14:
Yes, I agree with you. I always like the simplicity of CQ.

closed 2016-03-14 by huskier