dcowden/cadquery #143: Recover face vertices (counter) clock-wise?

johanneshk, 2016-04-18:
Hi,

I was wondering about the following: I have a face and want to recover the vertices in a way that when drawing lines from vertex to vertex the face is drawn. My first approach is something like this:
```
for face in faces:
    x= []
    y= []
    for v in face.Vertices():
        x.append(v.X)
        y.append(v.Y)
    x.append(x[0] # close shape
    y.append(y[0] 
    plt.plot(x,y)
    plt.show()
```
This works in most cases, although sometimes the order of vertices is wrong.
So here is my question:

* is there a way to recover face vertices in such a way?
* why is it that sometimes it works/sometimes not? I guess it's not safe to assume ordering...

I guess I could also use face.Edges() to recover this 'by hand'

Thanks!
 Johannes

dcowden, 2016-04-18:
Hi Johannes,

Thanks for trying cadquery!

You should definitely use Wires to recover this information.

In general your proposed algorithm will have problems in two cases: when
edges have curves rather than straight lines, and when faces have more than
one loop.

I do not have the object reference handy, but I think there is an
'outerwire' property that will tell you which wire is the outer one. Given
a face,it is a surprisingly difficult problem to figure out which wire of a
list is the 'outer ' one.

Can you describe more about what you are trying to do, maybe I can help
more.

johanneshk, 2016-04-19:
Hi and thank you for your reply!

I want to write a script for 'separation' of solids in a stp file. I designed a clock which consists entirely out of 2d shapes that are of course extruded orthogonal to the sketch layer. All shapes have a thickness of 1mm and I want to create a dxf of all shapes oriented correctly and arranged in a rectangle (aspect ratio ~1) in a space saving way. See [also](http://forum.freecadweb.org/viewtopic.php?f=22&t=14916&sid=dda24989aae039213cadfdd73e09a038) . Then I can upload this dxf to some laser cutting provider.

This is standard procedure in the industry, and I try to implement my approach. What I've done so far (in a really dirty script):

1. determine orientation of a solid: Orientation is the direction in which the shape doesn't change. In the direction of orientation there are only two faces with the same orientation. Those faces have a distance of 1mm (or whatever thickness). These conditions did the trick so far.
2. rotate all solids, such that they are all oriented in the direction of (0,0,1). Also rotate solids around their orientation axis to find the orientation with smallest bounding box. (All shapes are roughly rectangular)
3. arrange solids in a space saving way. Of course there is no exact algorithm to do this (My testcase has 97 solids). Since I don't (for the time being) want to implement my own algorithm I tried: [this](http://projectdrake.net/blog/2008/12/29/bin-packing-rectangle-packing-and-image-atlasses/). Which works, but the results could be better. Now I want to try [this](http://projectdrake.net/blog/2008/12/29/bin-packing-rectangle-packing-and-image-atlasses/). And for this I need to export the outline of the shapes.

Indeed my parts also have cutouts which could makes things event more complicated. But for the time being I could indeed export the outerwire. Of course at some point it would be pretty cool to also arrange parts in the cutout of another part (if possible).

Maybe I need to implement my own packing algorithm... But so far the algorithm works and working with cadquery was a breeze!

dcowden, 2016-04-19:
Ah, interesting project!

What's interesting is that we have been speaking of doing the inverse of
what you are doing: allowing people to _import_ dxf files into cadquery, so
that they can extrude them.

Let me do a bit of research for you, and see if there is an easy way to do
this. It doesnt feel like this is code we should have to write-- i'm sure
its been done already.

jmwright, 2016-04-21:
Cool project @johanneshk 

There's absolutely no pressure to share anything, but I'd like to start encouraging members of the community to post any CadQuery work they feel like showing to the community. https://groups.google.com/forum/#!topic/cadquery/mS2ITd37sQA

Again, there's no obligation to share anything. @dcowden and I are just really interested in how people are using CadQuery, and want to give people an outlet to share their cool projects.

Thanks for using CadQuery!

jmwright, 2016-04-27:
@dcowden @johanneshk Any updates on this conversation? I'm curious what the research turned up.

dcowden, 2016-04-27:
Sorry, this is waiting on me, i have not looked into this yet.

dcowden, 2016-05-01:
Hi, ok over in the Google group I learned a bit more about this project. 

In short, there is currently no easy routine available to convert a wire or edge to a list of points, which is what is needed here.

in OpenCascade, there are functions available to do this, but they are not accessible via the FreeCAD api. 

If we can access the underlying OCC Wire object, and if we have pythonOCC installed, the class below contains a wrapper that will do what we want:

https://github.com/dcowden/emcfab/blob/master/applications/slicer2/Wrappers.py#L441

As you can see, generating a list of points is not trivial.  There are several considerations:
   1. Obviously, you must iterate over all of the Edges in each wire
   2. You need to make sure the edges are in head-to-tail order when you iterate over them. 
        suprisingly, this is not done for you by default, so there is some work necessary to do that too.
   3. Each edge has a basis curve, which may or may not be oriented along the edge direction
   4. You need to use GCPnts_QuasiUniformDeflection or other OCC api to convert an edge into a 
       set of discrete points. This function allows you to specificy the amount of deviation you are 
       willing to accept from the original curve. 

The library I shared above does much of this code, and more-- it is (as far as I know ) the only implementation around that can directly slice a STEP file for printing on 3d printers.

jmwright, 2016-05-02:
> You need to make sure the edges are in head-to-tail order when you iterate over them.

http://forum.freecadweb.org/viewtopic.php?f=22&t=10474#p84346

Not sure if this helps, but the connectEdgesToWires function in FreeCAD is supposed to connect edges head to tail, but I'm not sure that it does it correctly. That bug is still outstanding and is the reason chaining mirror operations in CadQuery doesn't work.

closed 2017-08-20 by jmwright