dcowden/cadquery #149: Infinite Windows

Coding-Badly, 2016-06-27:
Each time I Execute Script (F2) with one of my scripts a new FreeCAD document is created.  Even if the script is empty.

However, the example scripts do not exhibit the behaviour.  Execute Script with an example loaded refreshes the existing FreeCAD document.

How can I get Execute Script to refresh the existing document instead of creating a new document?

Note: I love CadQuery!  It is so very much better than OpenSCAD!

jmwright, 2016-06-27:
Hi @Coding-Badly 

This issue is actually related to the CadQuery module for FreeCAD, which is here: https://github.com/jmwright/cadquery-freecad-module

What version of FreeCAD are you running? If you use "Save As" from the CadQuery menu to save the script before executing it, does the module stop creating new documents?

Coding-Badly, 2016-06-27:
Thank you for the reply!

FreeCAD Help / About...
OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.6704 (Git)
Build type: Release
Branch: releases/FreeCAD-0-16
Hash: 0c449d7e8f9b2b1fb93e3f8d1865e2f59d7ed253
Python version: 2.7.8
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

Got it.  A space in the filename is the trigger.  If the filename contains a space, a new document is created with Execute Script.  If the filename does not contain a space, Execute Script refreshes the existing document.

Coding-Badly, 2016-06-27:
> This issue is actually related to the CadQuery module for FreeCAD...

Oh bugger!  Did I put this in the wrong place?

Coding-Badly, 2016-06-27:
The pound sign is also poison.

jmwright, 2016-06-27:
> Did I put this in the wrong place?

Things that are related to the core library usually go here, and issues with the FreeCAD module go at the link I posted above. Dave and I watch both streams though and so we see everything that's posted on either repo. It's not a big deal.

I've cross-linked this issue to one on the module's repo. https://github.com/jmwright/cadquery-freecad-module/issues/75

Thanks for the report and the research into the problem. I'll try to address this in the module soon.

closed 2016-06-27 by jmwright