dcowden/cadquery #154: [Possible Bug] global name 'toVector' is not defined

huskier, 2016-11-18:
When calling Edge.makeCircle(...) method, an error, "global name 'toVector' is not defined", is encountered.

I'm trying replacing toVector(pnt) with Vector(pnt) or FreeCAD.Base.Vector(pnt), but all failed.

So, how do we fix this problem? I think we need to define a "toVecotr" method, but where is the best place to do this.

In class Edge, the classmethod makeCircle(...)
'''
    @classmethod
    def makeCircle(cls, radius, pnt=(0, 0, 0), dir=(0, 0, 1), angle1=360.0, angle2=360):
return Edge(FreeCADPart.makeCircle(radius, toVector(pnt), toVector(dir), angle1, angle2))
'''

Thanks.

huskier, 2016-11-18:
I could make the code running as changing the code slightly.
`
@classmethod
def makeCircle(cls, radius, pnt=(0, 0, 0), dir=(0, 0, 1), angle1=360.0, angle2=360):
    return Edge(FreeCADPart.makeCircle(radius, FreeCAD.Base.Vector(pnt.x, pnt.y, pnt.z), FreeCAD.Base.Vector(0,0,1), angle1, angle2))
`

jmwright, 2016-11-18:
@huskier I suspect this may have been code that got lost in the conversion from the Parametric Parts codebase to the CadQuery GitHub repo. I've been pretty busy lately, but I'll check into this and the other issue you filled soon. Thanks for the feedback.

dcowden, 2016-11-19:
Hey guys, yes, there were some conversions where toVector was changed
around..

I will have a look at this tomorrow. @huskier I suspect you are correct
about the right fix, but let me look to see if I can figure out what I was
thinking ( other than 'man I need to release this' which is apparently the
case ;)

dcowden, 2016-11-19:
Hi, @huskier:

The easiest solution is to use Wire.makeCircle, but we do need to fix Edge.makeCircle and Face.makePlane since they are provided.

Of course you are right, FreeCADPart.makeCircle expects a FreeCAD.Base.Vector for both the center and normal argument ( as is in Wire.makeCircle).

toVector is left over from before the creation of geom.py / Vector.py.  I think the right solution here is to replace the toVector call with construction of a geom.Vector, followed by the .wrapped property (untested):

```
 @classmethod
    def makeCircle(cls, radius, pnt=(0, 0, 0), dir=(0, 0, 1), angle1=360.0, angle2=360):
        center = Vector(pnt)
       normal = Vector(dir)
        return Edge(FreeCADPart.makeCircle(radius, center.wrapped,normal.wrapped, angle1, angle2))

```

The solution for Face.makePlane should be similar. @huskier would you mind terribly making the fix? I've actually not got my laptop available ATM so though the change is simple it would take me some time.

huskier, 2016-11-20:
@dcowden there is a very weird thing, which is that center.wrapped is a Vector type, whereas normal.wrapped is a FreeCAD.Base.Vector type.

I put a print statement just before return statement,
`
print type(center), type(normal), type(center.wrapped), type(normal.wrapped)
`

and the printed result is as:
<class 'cadquery.freecad_impl.geom.Vector'> <class 'cadquery.freecad_impl.geom.Vector'> <class 'cadquery.freecad_impl.geom.Vector'> <type 'Base.Vector'>

Apparently, the type of the center.wrapped is unexpected, and it should be Base.Vector, but printed Vector instead. I really do not know the reason.

huskier, 2016-11-28:
This appeared my fault (old version of geom.py file), and @dcowden is correct. I have pulled a request to the master branch to fix this small bug.

I'll close this issue.

huskier, 2016-11-28:
I am really sorry for the untested test function testEdgeWrapperMakeCircle and testFaceWrapperMakePlane, and I will make them pass as soon as possible.

The detailed info from travis-ci.org, and I do not know why the code passed travis-ci.
======================================================================
ERROR: testEdgeWrapperMakeCircle (tests.TestCadObjects.TestCadObjects)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/travis/build/dcowden/cadquery/tests/TestCadObjects.py", line 47, in testEdgeWrapperMakeCircle
    self.assertTupleAlmostEquals((0.0, 5.0, 0.0), halfCircleEdge.CenterOfBoundBox(0.001).toTuple())
TypeError: assertTupleAlmostEquals() takes exactly 4 arguments (3 given)
======================================================================
ERROR: testFaceWrapperMakePlane (tests.TestCadObjects.TestCadObjects)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/travis/build/dcowden/cadquery/tests/TestCadObjects.py", line 52, in testFaceWrapperMakePlane
    mplane = Face.makePlane(10,10)
NameError: global name 'Face' is not defined

huskier, 2016-11-29:
I have fixed the two test errors above mentioned, and now I could close this issue.

Thanks @dcowden  & @jmwright.

closed 2016-11-29 by huskier

jmwright, 2016-11-29:
Thank you @huskier for all your work. It's greatly appreciated.