dcowden/cadquery #155: The idea of adding a sketchlet function into the 2D operations

huskier, 2016-11-18:
`
    def sketchlet(self):
        def makeSketchletWire(pnt):
            p0 = pnt
            p1 = p0.add(Vector(-23.5,0,0))
            e1 = Edge.makeCircle(1.6,p1)
            #p2 = p0.add(Vector(23.5,0,0))
            #e2 = Edge.makeCircle(1.6,p2)
	    listOfEdges = []
            listOfEdges.append(e1)
            #listOfEdges.append(e2)
           sketchWire = Wire.assembleEdges(listOfEdges)
            sketchWire.forConstruction = False
            return sketchWire       
        return self.eachpoint(makeSketchletWire, useLocalCoordinates=True)
`

huskier, 2016-11-18:
The problem is how do we know how many individual wires can be obtained from the listOfEdges?

`

    def assembleEdges(cls, listOfEdges):
        fCEdges = [a.wrapped for a in listOfEdges]
        wa = Wire(FreeCADPart.Wire(fCEdges))
        return wa

`

dcowden, 2016-11-19:
Hi, @huskier 

I love the idea of a sketchlet.  Its closely related to another feature I've been wanting to implement-- truly parametric 2d sketches.

Tell me more about what data format/data you envision inside of a sketchlet.  If you have 'static' geometry in mind, i suspect that using DXF would be the best strategy. DXF import would essentially be a 2-d sketchlet, but would be much easier for users to create because so many DXFs already exist, and they can be exported by nearly any program.

I have used this package before with great success: https://pypi.python.org/pypi/dxfgrabber

I would envision something like this:

`result=cadquery.Workplane("XY").import("/path/to/file.dxf").extrude(50)`

Now, things get a bit more interesting if you imagine that sketchlets are parametric. For example, i dont simply have a square defined by  coordinates at (0,0), (10,0), (10,10), and (0,10) with lines inbetween ( as would be done with existing CQ).

 Instead, I have a square, which is composed of 4 lines, with:
    * two lines having vertical constraints
    * two lines have horizontal constraints
    * all 4 lines having an equal length constraint (or alternatively, the vertices of each line endpoint having co-incident constraints)
    * one side having a length constraint set to 10
   * a co-incident constraint for one vertex of one line with the origin

Now _this_ is a different ball of wax-- because now the sketch itself can preserve design intent, while allowing the single value, 10, to be changed.

Implenting this would require using a geometry solver. But would be really cool!

dcowden, 2016-11-19:
@huskier ,@jmwright

should we be interested in doing parametric sketches, here's one candidate library for it:

http://geosolver.sourceforge.net/

Another possibility is to use the Cheetah solver, which is what FreeCAD uses ( read here)
http://www.slideshare.net/NickSidorenko/beyond-parametric-new-approach-to-geometric-constraint-solving

dcowden, 2016-11-19:
actually, maybe FreeCAD doesnt use Cheetah any more. Their solver can be found here. Maybe it is an older version of Cheetah, but since it is open source we could use it. Or we could write a wrapper around it since we already use FreeCAD

https://github.com/FreeCAD/FreeCAD/tree/master/src/Mod/Sketcher/App/planegcs

jmwright, 2016-11-20:
@dcowden I don't have a problem with 2D parametric sketching being added to CadQuery.

huskier, 2016-11-21:
Could we discuss this in two steps, first 'static' geometry, and then 'parametric' geometry? And for now, could we firstly focus on the 'static' geometry aspect?

huskier, 2016-11-21:
I like python library dxfgrabber. If we add 'static' geometry import function with dxf support, we need to use dxfgrabber, and the question is whether we include dxfgrabber library as a part or just make it as a dependent library?

BTW, maybe we need to add another function to expand the arc-making method, since the current makeThreePointArc(...) sometimes inconvenient.

dcowden, 2016-11-21:
Yes, we have discussed this before and decided that we are willing to begin allowing dependencies.  There are already so many that it's a problem needing a solution anyway. We need to provide requirements.txt for easy installation.

There are two ways to make cq easier to install initially, including freecad.  One would be to provide a conda package, the other to provide a docker container. I have started and mostly completed that work a few months back, but for side tracked on some other projects.

When that work is completed, extra libraries will not be any harder for the user to install

huskier, 2016-11-29:
Could we move into the next step?  Could you help me make the following simple demo workable in the current master branch? Then we could do the 'static geometry' importing things.

`

def sketchlet(self):
    def makeSketchletWire(pnt):
        p0 = pnt
        p1 = p0.add(Vector(-23.5,0,0))
        e1 = Edge.makeCircle(1.6,p1)
        #p2 = p0.add(Vector(23.5,0,0))
        #e2 = Edge.makeCircle(1.6,p2)
        listOfEdges = []
        listOfEdges.append(e1)
        #listOfEdges.append(e2)
        sketchWire = Wire.assembleEdges(listOfEdges)
        sketchWire.forConstruction = False
        return sketchWire       
    return self.eachpoint(makeSketchletWire, useLocalCoordinates=True)

`

dcowden, 2016-11-29:
@huskier i would rather work on just importing DXF next. 

I think we're going to run into problems quickly getting the example above to work nicely, due to the limits on the number of wires ( which you observed earlier ).

This might help: its the current 'best' way to link in new functions

http://parametricparts.com/docs/extending.html

huskier, 2016-11-29:
@dcowden With the help of dxfgrabber, it is not so difficult to import DXF file. However, DXF drawings are usually more complex than CQ's native drawing. I am afraid we have to deal with the limits on the number of wires.

Another issue is that the dependencies. For the scripting from CLI, maybe it is not a big problem, and we just install the dependencies. However, for the FreeCAD's plugin, we have to think a way to make it work before the conda packages and the docker containers.

Anyway, we could just implement the function first.

Meanwhile, I will think about a solution to the limits on the number of wires.

dcowden, 2016-11-29:
@huskier yes, I agree there are two main audiences here: 

(1) people who are using FreeCAD, and are using the CQ plugin ( and are thus also probably using FreeCAD's GUI ), and 
(2) people who are using CQ as  a stand alone tool.

A docker image and conda package are clearly ideal for audience (2).  I think its actually possible that it is ideal for a portion of audience (1), but I agree it is not suitable for people in group (1) who have already installed FreeCAD before they come to use CQ. They do not want to repeat/remove their install.

I assume that FreeCAD has DXF import. Since our implementation is based on FreeCAD, i assume we can easily test the feasibility and wire limits by importing DXF from within FreeCAD. If it works there, we can make it work too.

I think importing DXF will be, in practice, much easier to code than the sketchlets you have proposed.  With sketchlets we need to have the syntax fit in nicely with the CQ code flow ( the chainability, etc).  If we have a single new function called dxf('filename'), the implementation of this function can natively call FreeCAD libraries, create wires, etc, with impunity.  I would envision it creating a multitude of wires ( representing loops in the underlying DXF), and adding them all to the stack. I dont see it being a problem.

huskier, 2016-11-30:
OK, let's work on DXF importing.

dcowden, 2016-11-30:
Ok, I will start the process with:

  * a document describing how it should work
  * a branch with a start 

I have done some work already with dxfgrabber in the past.  Oddly enough, I also ended up using another dependency, networkx, for finding loops. And merging nodes that were subject to round-off error.  The main difference between a low and high quality DXF importer is whether it handles merging loops so that there are not geometry problems. Many DXFs are created with 'dumb' lines and arcs that do not meet exactly at the ends. 

I'll try to write some code that reads the DXF, organizes the entities into wires, and then outputs them

huskier, 2016-12-01:
Good to know the work plan. If there is something I can help, please fell free to tell me.

dcowden, 2016-12-01:
I'll definitely be able to use some help. My hope is to describe the way of working through either words or code, and get it flushed out to the point where others (you ;) ) could add functions, tests, etc.  

I have a some time off my day job coming up over the holidays, which is when I plan to do some of this work.

dcowden, 2016-12-28:
CQ 2.0 work started