dcowden/cadquery #162: Implement pythonOCC-based Topological objects

dcowden, 2016-12-11:
The new system should represent existing topological items.
New subclasses should be created for commonly used, special cases, for easier use

The below represents a class-hierarchy:
  
     Shape
        Vertex
        Edge
        Face
        Shell
        Solid

fragmuffin, 2017-09-30:
@dcowden 
What affect will `pythonOCC` implementation have over existing dependencies on `freecad`?
(also relevant: #161)

I ask because... while I'm developing with `cadquery` (both in contribution to `cadquery`, and in my own attempt at a parts library in [cqparts](https://github.com/fragmuffin/cqparts)), I want to know how much refactoring I'll need to do.

I often need to reference the `wrapped` instance of `cadquery.Edge`.
for example

```python
>>> import cadquery
>>> import FreeCAD
>>> import Part
>>> Part
<module 'Part' from '/usr/lib/freecad/lib/Part.so'>

>>> wire = cadquery.Workplane("XY").lineTo(1, 1).wire().val().wrapped
>>> isinstance(wire, Part.Wire)
True

>>> vertex = wire.Edges[0].valueAt(1)
>>> vertex
Vector (0.7071067811865475, 0.7071067811865475, 0.0)
>>> isinstance(vertex, FreeCAD.Base.Vector)
True
```

Nothing in the above is prefixed with a `_`, which implies they're not "private" (or what passes for _private_ in python)

dcowden, 2017-09-30:
@fragmuffin, when we move to pythonOCC, the objects you have now will be replaced with their pythonOCC topological equivalents, or possibly wrappers we can created for them.

The pythonOCC topological objects do not have many methods at all, they are more like structs.  Behaviors like you have above (moving to a parameter along an edge) are done with a separate class, passing the edge as a reference.

We can preserve compatibility by making wrappers that match freecads.   but the bigger question is what would be most idiomatic , were compatibility not a concern.

fragmuffin, 2017-10-01:
@dcowden,
> We can preserve compatibility by making wrappers that match freecads

I'd avoid this if possible; if they're designed to mimic FreeCAD's classes, then I feel like they'll prove to be a maintenance burden.

> what would be most idiomatic

I feel like pulling a wire out of a `Workplane` instance could be more idiomatic.
currently...

```python
>>> wire = my_cadquery_workplane.wire().val()
>>> isinstance(wire, cadquery.Wire)
True
```

Because I'm expecting to pull the `Wire` **out** of the `Workplane`, I feel this is more of a #167 implementation.

with respect to the `cadquery.Wire` & `cadquery.Edge` classes, and their , I think that's easy:

```python
# in cadquery
class Wire(object):
    ...
    @property
    def length(self):
        return self._wrapped.Length
    @property
    def edges(self):
        return [Edge.cast(e) for e in self._wrapped.Edges]
    def value_at(self, dist):
        for e in self.edges:
            edge_length = e.length
            if dist > edge_length:
                dist -= edge_length
                continue
            return e.value_at(dist)
        return None

class Edge(object):
    ...
    @property
    def length(self):
        return self._wrapped.Length
    def value_at(self, dist):
        return self._wrapped.valueAt(dist)
```

Note in the above code I've also:
* changed `wrapped` to `_wrapped`
* made all `cadquery` class object calls _snake case_ as opposed to _camel case_

dcowden, 2017-10-01:
@fragmuffin I'm ok with that.

I have noticed that most cad framework s, ( OCC, onshape, and parasolid), use separate classes to interrogate geometry. I think it's because it gets really complex, and in many cases ends up involving more than one object.

As an example, while getting the midpoint for an edge works well as an edge method, getting the distance between two objects, or the intersection doesnt.  You end up with query objects that accept topology and answer questions. 

Here is a look at what an edge class might look like implemented in pythonOCC, and leaving many methods on the edge object:

https://github.com/tpaviot/pythonocc-utils/blob/master/OCCUtils/edge.py

adam-urbanczyk, 2017-10-01:
@fragmuffin  Can you share which FreeCAD functions / methods are missing in the current API?

fragmuffin, 2017-10-02:
@adam-urbanczyk 
oh man... let me dig through my code...
actually, it seems I'm not using that many so far

```python
Part.Edge.valueAt(...)
Part.Geom*.value # for valueAt error mitigation in circular arcs, see link below
Part.Solid.sewShape()
Part.Solid()  # re-casting "sewn" shapes and pushing it back into the wrapped variable
```

for usage of "`Part.Geom*`" as `Curve`, see [implementation in profile_to_cross_section() for threads](https://github.com/fragmuffin/cqparts/blob/develop/src/cqparts/types/threads/base.py#L119-L133)

jmwright, 2017-10-08:
@adam-urbanczyk On your PythonOCC branch you seem to have the FreeCAD and PythonOCC implementations existing side-by-side. In your opinion, is this the direction we should go, or should the PythonOCC implementation be broken out separately? I ask because I'm wondering how practical it would be to merge your PythonOCC branch in the next release and make it an "experimental feature". I could bump CQ 2.0 support up to the top of the queue for the FreeCAD module so that people could start experimenting with the PythonOCC implementation more easily. Is this a good idea or bad idea?

dcowden, 2017-10-08:
@adam-urbanczyk @jmwright @fragmuffin

I think cq 2.0 should be pythonOCC only. It's simply too complex and too much surface area to simultaneously support both. 

Cq 2.0 can be released with Adams code and achieve alpha quality.  I'd rather iterate from there with a clear objective of making cq 2.0 more capable than 1.x ( direct API and so on).

On another note, this is a case where any decision is better than none. I vote for cq 2.0 to be pythonOCC only (that's what's reflected in the 2_0 branch now). But I would prefer a decision to support them both over talking about it a lot longer.

jmwright, 2017-10-08:
I'm ok with just jumping in and iterating from there, but I still have my concerns about ending up with a Python 2 vs Python 3 situation.

jmwright, 2017-10-08:
Assuming we move forward with a separate PythonOCC implementation, there are two main paths.

1. Separate repository
2. Long-running branch in the main repository

#2 would allow us to do parallel version releases for 1.x and 2.x, and would allow us to rebase on the 2.x branch when 2.x becomes the community standard. There will be confusion in the meantime, but we'd just have to be prepared to accept that as a necessary part of evolving CQ.

dcowden, 2017-10-08:
@jmwright @fragmuffin @adam-urbanczyk

what options do we have to avoid the python 2/3 situation, or at least make it not as bad?

Obviously one way is to support freecad based backend in the long term, but the question I suppose is how many users would be affected by the change in the backend.

If we committed to 100% cq1.x API compatibility, that would mean that only users who used a mix of freecad code and cq code are affected. Of course perhaps more importantly it leaves us with no GUI.

Between these two arguments, I think the strongest by far is the loss of cqfm.  If we don't have a roadmap to get GUI support we are happy with, thats a strong argument to stay with freecad support. 

If we wanted to stay with freecad, I'd recommend that we keep one stream, and plan on always being freecad based. Then, when we run into freecad limitations, we can then use the underlying OCC version distributed with freecad.  I would say we should start using Adams  code.

So with that approach we are supporting single backend, that's a mix between freecad and occ

jmwright, 2017-10-08:
> Between these two arguments, I think the strongest by far is the loss of cqfm. If we don't have a roadmap to get GUI support we are happy with, thats a strong argument to stay with freecad support.

My goal is to make CQFM (the FreeCAD module) work with CQ 2.0, so I don't know that it should be the driving factor in the decision. @adam-urbanczyk has a method to export Brep objects from PythonOCC that I am going to try to utilize to display CQ 2.0 objects in FreeCAD.

dcowden, 2017-10-08:
@jmwright how would CQFM work without FreeCAD though?  Or you're saying the user can still install freecad to get CQFM, and we'll assume that CQ2.0 uses the OCC bundled with FreeCAD in that case?

jmwright, 2017-10-09:
I haven't worked out all the details yet, but when a CadQuery script calls `show_object()` with PythonOCC selected as the implementation, the resulting PythonOCC object will be exported to Brep and then to a FreeCAD object for display. There might also be room there for code to be run in a mixed mode (FreeCAD and PythonOCC). I haven't figured out yet how/if we'll be able to use OCC/OCE bundled with FreeCAD, or if we'll have to embed specific versions. I'm going to bump CQ 2.0 support above the new code editor in the roadmap for CQFM if we can determine a definite path forward (I.E. start using Adam's branch as the CQ 2.0 codebase).

dcowden, 2017-10-09:
Ok that sounds good.

I'm not going to have much time to work on this over the next couple of
months, to much other stuff going on. but I'll try to support as much as
possible

adam-urbanczyk, 2017-10-09:
@jmwright @dcowden 

I did not think too much about the FreeCAD part of the code that is still left in my experimental branch. I think I am in favor of removing it from 2.0 (we have too many things to implement anyhow).

Regarding CQ workbench in FreeCAD it should be indeed possible to use .brep files to import geometries without any loss of precision.

I am not sure why are you so afraid of py2/3 compatibility. I think it is 90% there in the OCC branch. Did I miss something?

I think in order to start moving forward it is critical to have more people involved. Would nice to have e.g. a conda pacakge and ipython / jupyter integration. Those things are on my ambitions list. What do you think?

jmwright, 2017-10-10:
> I am not sure why are you so afraid of py2/3 compatibility.

I'm not worried as much about the compatibility as how it seems to have split the community, even now when Python 3.x is becoming the standard (at least in Linux).

> Would nice to have e.g. a conda pacakge and ipython / jupyter integration. Those things are on my ambitions list. What do you think?

I've only used Anaconda a couple of times, but it seems like it's really helpful to have a conda package for the scientific Python crowd, which CadQuery fits well with.

As for IPython/Jupyter integration, are you talking about what Adam Vermeer has done with Jupyter Notebook integration, or something else?

RustyVermeer, 2017-10-10:
@adam-urbanczyk I have just added the LICENSE to both avnb and CQNB.

jmwright, 2017-10-10:
@adam-urbanczyk I've never taken a codebase from Python 2 support to Python 2/3 support. Where do we start?

Peque, 2017-10-10:
@jmwright [2to3](https://docs.python.org/3/library/2to3.html) might be a good starting point.

@adam-urbanczyk  Do you think supporting Python 2 is worth the effort at this point? Specially if starting almost from scratch and taking into account that we are so close to Python 2 deprecation. Why not going for Python 3 directly?

jmwright, 2017-10-10:
@Peque If we go completely to Python 3 I think we lose support for any FreeCAD version but 0.17 (which hasn't been released yet). Someone can correct me if I'm wrong about that though.

Peque, 2017-10-11:
@jmwright Yeah, thought that would not be a problem, as the initial proposed back-end was PythonOCC.

Also thought being this new CQ project a probably-2018-project, FreeCAD 0.17 would be released when the time comes (if it ever comes) to support FreeCAD backend. :blush:

jmwright, 2017-10-11:
@Peque I guess I was thinking about the current CQ 1.x implementation, but you're right that PythonOCC (CQ 2.0) doesn't have that reliance on FreeCAD.

adam-urbanczyk, 2017-10-11:
@jmwright  @Peque  What I mean is that the experimental OCC branch at the moment passes all tests on 2.7 and all but 3 (CQIG related) on 3.6. So if we start iterating from there, would be nice to keep it. I don't think that this will be that complicated.

Peque, 2017-10-11:
@adam-urbanczyk Yeah, I agree with you, it might not be that complicated, although it is still some extra work.

Actually I was thinking more about the future... deprecating things is also time consuming (and painful for the users). If CQ 2 is to break CQ 1 compatibility in some way (and it seems it probably will), then I do not see the need to support Python 2 in CQ 2. Python 2 deprecation is happening sooner than it seems (time goes quickly). :joy:

In my own experience maintaining Python 2 and Python 3 compatibility can actually become a pain. Maybe not at first, but at some point you might find yourself wanting to use some new features included in the latest versions (i.e.: asyncio), and then it becomes harder. Also there are some projects already declaring their intention to drop Python 2 support soon, so you might find that some dependencies, maybe in a couple of years from now, no longer support Python 2. At that point you might want to deprecate Python 2 support as well, and users will suffer from not having used Python 3 from the beginning.

I am just thinking about minimizing efforts, as there is already a lot of work/hours required to develop CQ 2. Not that I have a strong opinion on this matter nor I hate Python 2... So whatever you decide is fine. Just wanted to share my opinion. :blush:

PS: https://wiki.python.org/moin/Python2orPython3

fragmuffin, 2017-10-12:
> In my own experience maintaining Python 2 and Python 3 compatibility can actually become a pain.Maybe not at first, but at some point you might find yourself wanting to use some new features included in the latest versions

@Peque i know what you mean; when code supports both py2&3, as far as the language goes, developers are blending the worst of both versions.
* native py2: developers have relatively clean code, that's supported everywhere, but is lacking modern advances
* native py3: developers get all the new toys, and the code is very clean.
* py2&3: messy code to accommodate both, tests must be run twice, and no new toys (from py3)

If it's possible, I vote for a clean py3 transition.

dcowden, 2017-10-12:
I agree-- my experience has also been really bad trying to support both.  For CQ, we are dependent on FreeCAD or on pythonOCC.  To me, the most obvious path is (hasbeen) to stay on 2 till all of our dependences are 'squarely' on 3 ( meaning, bugs worked out), and then move.

It seems reasonable to target python3 in any major new development effort (CQ 2.0).  If pythonOCC and FreeCAD will be python3 by the time CQ 2.0 is ready, it seems like its best to simply decide to use python3 to develop CQ 2.0

adam-urbanczyk, 2017-10-17:
I do not necessarily agree with all the statements, but if there is no interest from the community then indeed let's focus on py3.

jmwright, 2017-10-18:
> I do not necessarily agree with all the statements

@adam-urbanczyk Are there any statements in particular that you disagree with, or is it just the idea of dropping support for Python 2 in general?

adam-urbanczyk, 2017-10-18:
@jmwright  the latter

closed 2018-12-06 by dcowden