dcowden/cadquery #169: Create conda package

dcowden, 2016-12-11:
CQ should be available via conda package. That way, its possible to run CQ with a one-step install, even though there are many dependencies. 

pythonOCC already has a conda package-- ideally, we can simply mark that as a dependency so that we do not have to re-package that.

jmwright, 2017-10-28:
@adam-urbanczyk Has done this. Does that take care of this issue (other than documenting it)?

dcowden, 2017-10-28:
Is the package created from cq master Travis? I'm ok closing if the ci loop is creating the conda package. If not I'd like to do that first. It would be nice if each push to master creates a new (autonumbered) release

jmwright, 2017-10-28:
I don't think CI integration has been done yet.

adam-urbanczyk, 2017-10-29:
The conda package is for created based on my (OCC) branch and definitely without CI/CD.

dcowden, 2017-10-29:
What are the chances we could leverage the work you did in the main project? I'm sure building the conda package would be mostly the same with a freecad based version right? 

How complete is your OCC based version? If I recall, nearly all the tests pass, and you even have the svg exporter working.

I think the direction forward is to combine your OCC branch with a better direct API wrapped around it (as discussed in #167) to yield cq 2.0.

adam-urbanczyk, 2018-04-13:
@dcowden @jmwright coming back to the topic of conda:

I am setting up conda build on my OCC branch (for now on travis). If this experiment succeeds we can implement the same for CQ1 (Linux+windows and python3 [no freecad conda pkg with py2]) . What do you think?

BTW I made a CadQuery organization on anaconda website. I can add you to it if you have accounts there.

@dcowden  I think the OCC version is fairly complete actually. Do you still want to to reuse it?

adam-urbanczyk, 2018-04-19:
I am not saying cq would not be supported on py2 + win, but only that no cond pkg would be there for this use case.

You need to setup an account on anaconda,  I'll add you then to the Caquery org.

dcowden, 2018-04-19:
@adam-urbanczyk  @jmwright yes, i'd be in favor of using adam's branch to create a freecad free  version.  

i am open to ideas about the best way to pull it off.

adam-urbanczyk, 2018-06-24:
Great work @bollwyvl! Can conda-forge build be triggered on every release of CQ?

dcowden, 2018-06-24:
This is so awesome!
I merged those two prs. What other work is required? At a minimum i think we should update the readme, to reflect this installation method.

bollwyvl, 2018-06-24:
Oh, the PRs don't need to be merged, but that's great! We're handling both of them in patches to meet the various checks.

The recipe will be reviewed by someone on conda-forge core (this can take a few days), I might have to do some rework. Once all that gets squared, then it will get promoted to a feedstock, and will get built and pushed to anaconda.org.

closed 2018-12-06 by dcowden