dcowden/cadquery #172: Implement Parametric 2d sketching

dcowden, 2016-12-11:
Currently, sketching is not parametric.  Sketches explicitly define geometry, rather than the design intent. 

For example, instead of defining a square as four lines between points 0,10, 10,10, 10,0, and 0,10, we should define four lines with these constraints:
    each of the ends co-incident
    length of each line equal
   one line equal to 10
   one point co-incident with the origin

To do this, we will need to integrate a constraint solver.

dcowden, 2016-12-11:
This is the solution for the functionality proposed in #155.

adam-urbanczyk, 2016-12-29:
2D constraint solver implemented in js : https://github.com/xibyte/jsketcher

dcowden, 2016-12-29:
> 2D constraint solver implemented in js : https://github.com/xibyte/
> jsketcher

Too bad it's agpl, we can't use it.

That said, I really thing cq 2.0 needs to contemplate parametric 2d
sketching, even if we implement it later on. I'll at least include an idea
of how it might work in the 2_0 branch