dcowden/cadquery #179: No documentation of extended selector syntax

adam-urbanczyk, 2017-04-13:
We have this great feature, but seems that users don't know about it (cf #177)

adam-urbanczyk, 2017-04-13:
Adding this as an example could be a good starting point:

```
import cadquery as cq
from Helpers import show

result = cq.Workplane("XY").box(2, 2, 2).\
    faces(">Z").shell(-0.2).\
    faces(">Z").edges("not(<X or >X or <Y or >Y)").\
    chamfer(0.125)

show(result)
```

jmwright, 2017-04-13:
@adam-urbanczyk Great idea. I can add that example. Would you have the time and interest to add some documentation on the logical and DirectionNth selector features that you added? You understand them better than anybody at this point.

https://github.com/dcowden/cadquery/blob/master/doc/selectors.rst

adam-urbanczyk, 2017-04-13:
Yes, I'll write something

jmwright, 2017-04-14:
This example is now in the FreeCAD module's master: https://github.com/jmwright/cadquery-freecad-module/blob/master/CadQuery/Examples/Ex033_Shelled_Cube_Inside_Chamfer_With_Logical_Selector_Operators.py

adam-urbanczyk, 2017-04-23:
Fixed in #181

closed 2017-04-23 by adam-urbanczyk