dcowden/cadquery #182: Mismatch between rarray code and documentation

hackscribble, 2017-04-19:
Documentation for `rarray` says ...

```
        :param xSpacing: spacing between points in the x direction ( must be > 0)
        :param ySpacing: spacing between points in the y direction ( must be > 0)
```

However, the test in the code is ...

```
        if xSpacing < 1 or ySpacing < 1 or xCount < 1 or yCount < 1:
            raise ValueError("Spacing and count must be > 0 ")
```

So, pitch values 0 < x < 1 result in an error while still being > 0.

jmwright, 2017-04-19:
@hackscribble Thanks for reporting this.  I don't see any reason a pitch between 0 and 1 shouldn't be allowed. @dcowden Do you remember what the logic was on that?

jmwright, 2017-04-22:
I experimented with this, and the library is coded to keep users from entering an X or Y spacing less than 1.

https://github.com/dcowden/cadquery/blob/90acdbc440bd63347dfc546845f3fcf5a4ece3e1/cadquery/cq.py#L1050

I think this...

```python
if xSpacing < 1 or ySpacing < 1 or xCount < 1 or yCount < 1:
```

...should be changed to this...

```python
if xSpacing <= 0 or ySpacing <= 0 or xCount < 1 or yCount < 1:
```

A partial count doesn't make any sense in this context, but the spacing could be less than 1 mm. Since this change should not break backward compatibility, I propose that we incorporate it and then make the docs match.

jmwright, 2017-04-23:
Fixed in commit e3d8bacf9519fc575f9b00daa99e75c62b7d0db4

Thanks again to @hackscribble for pointing this out.

closed 2017-04-23 by jmwright