dcowden/cadquery #183: Fix FreeCAD 0.17 Issues

dcowden, 2017-04-23:
FreeCAD 0.17 is breaking several tests ( 9 at this point).

One issue is explained here:
https://github.com/jmwright/cadquery-freecad-module/issues/77

I have fix that one, but there are others.
I'd recommend waiting until freeCAD 0.17 is stable before digging in

dcowden, 2017-04-23:
to test against latest FreeCAD, do these steps:

sudo add-apt-repository ppa:freecad-maintainers/freecad-daily
sudo apt-get update
sudo apt-get install freecad-daily
export FREECAD_LIB=/usr/lib/freecad-daily/lib

This installs the latest freecad alongside the stable version, so that you can run tests with both.

dcowden, 2017-04-23:
pushed changes that make it possible to test against latest FreeCAD 0.17 daily build. 

The tests are closer to passing in 0.17, but still fail. They pass in 0.16, at at least we are at the point
where this version allows it.
This way we can iterate on getting 0.17 working while not breaking 0.16

dcowden, 2017-04-23:
As of 4/23, here are the errors:

```
ERROR: testChamferAsymmetrical (tests.TestCadQuery.TestCadQuery)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/dcowden/gitwork/cadquery/tests/TestCadQuery.py", line 941, in testChamferAsymmetrical
    edge = cube.edges("|Z").vals()[0]
IndexError: list index out of range

======================================================================
ERROR: testEnclosure (tests.TestCadQuery.TestCadQuery)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/dcowden/gitwork/cadquery/tests/TestCadQuery.py", line 1337, in testEnclosure
    oshell.edges("|Z").fillet(p_sideRadius)
  File "/home/dcowden/gitwork/cadquery/cadquery/cq.py", line 840, in fillet
    raise ValueError("Fillets requires that edges be selected")
ValueError: Fillets requires that edges be selected

======================================================================
ERROR: testFillet (tests.TestCadQuery.TestCadQuery)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/dcowden/gitwork/cadquery/tests/TestCadQuery.py", line 918, in testFillet
    c = CQ( makeUnitCube()).faces(">Z").workplane().circle(0.25).extrude(0.25,True).edges("|Z").fillet(0.2)
  File "/home/dcowden/gitwork/cadquery/cadquery/cq.py", line 840, in fillet
    raise ValueError("Fillets requires that edges be selected")
ValueError: Fillets requires that edges be selected

======================================================================
FAIL: testAndSelector (tests.TestCQSelectors.TestCQSelectors)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/dcowden/gitwork/cadquery/tests/TestCQSelectors.py", line 380, in testAndSelector
    self.assertEqual(2, len(el))
AssertionError: 2 != 0

======================================================================
FAIL: testParallelEdgeFilter (tests.TestCQSelectors.TestCQSelectors)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/dcowden/gitwork/cadquery/tests/TestCQSelectors.py", line 134, in testParallelEdgeFilter
    self.assertEqual(4, c.edges("|Z").size())
AssertionError: 4 != 0

======================================================================
FAIL: testPerpendicularDirFilter (tests.TestCQSelectors.TestCQSelectors)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/dcowden/gitwork/cadquery/tests/TestCQSelectors.py", line 105, in testPerpendicularDirFilter
    self.assertEqual(8,c.edges("#Z").size() ) #8 edges are perp. to z
AssertionError: 8 != 0

======================================================================
FAIL: testSumSelector (tests.TestCQSelectors.TestCQSelectors)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/dcowden/gitwork/cadquery/tests/TestCQSelectors.py", line 398, in testSumSelector
    self.assertEqual(8, len(el))
AssertionError: 8 != 0

======================================================================
FAIL: testNestedCircle (tests.TestCadQuery.TestCadQuery)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/dcowden/gitwork/cadquery/tests/TestCadQuery.py", line 435, in testNestedCircle
    self.assertEqual(14,s.faces().size() )
AssertionError: 14 != 9

======================================================================
FAIL: testWorkplaneOnExistingSolid (tests.TestCadQuery.TestCadQuery)
Tests extruding on an existing solid
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/dcowden/gitwork/cadquery/tests/TestCadQuery.py", line 678, in testWorkplaneOnExistingSolid
    self.assertEqual(10,c.faces().size() )
AssertionError: 10 != 6

----------------------------------------------------------------------
Ran 151 tests in 5.009s

FAILED (failures=6, errors=3)

```

jmwright, 2018-02-23:
With ask the changes to the CI system and the fact that all the tests are passing, I think we have this covered.

closed 2018-02-23 by jmwright