dcowden/cadquery #188: Intersecting Solids

fragmuffin, 2017-09-09:
:+1: for an `intersect` command.

**Example** a remake of [wikipedia's intersect CSG example](https://en.wikipedia.org/wiki/Constructive_solid_geometry#Workings)
```python
import cadquery as cq
from Helpers import show

sphere = cq.Workplane("XY").sphere(7.5).translate((5, 5, 5))
box = cq.Workplane("XY").box(10, 10, 10)
intersection = sphere.intersect(box)  # error, not implemented

show(intersection)
```

**Workaround** can be found [in this gist](https://gist.github.com/fragmuffin/814c2e89e7eabdfca2059b9a511695a1)

or am I missing something? (entirely possible :wink:)

dcowden, 2017-09-09:
This is easy to implement. The intersect method is already implemented at the low level in shapes.py.

App that's needed is to add a method into cq.py that's similar to cut, but uses Solid.intersect instead of Solid.cut

A PR would be welcome

fragmuffin, 2017-09-09:
I can't help but feel that you've set me up for success.
regardless, I feel warm and fuzzy.

dcowden, 2017-09-09:
ha, you are quite funny ( evidently, in addition to being fuzzy).

+10 points for adding a test case, and thanks for the contribution. Is it WIP or are you ready for me to merge it in?

easyw, 2017-09-09:
common is different from intersect?

dcowden, 2017-09-09:
They are the same. intersect is the name I'm most familiar with, and is commonly used with CSG. But 'common' is what FreeCAD calls it.

fragmuffin, 2017-09-10:
@dcowden 
it's ready to merge, just a simple change.

actually, the WS cleanup was my editor, sometimes it's useful, but other times I get a _to-and-fro_ between conflicting editors... either way, you're welcome :wink:

dcowden, 2017-09-10:
Awesome, ok thanks again for the contribution!

dcowden, 2017-09-10:
Available in  master

closed 2017-09-10 by dcowden