dcowden/cadquery #191: Documentation Examples are broken

dcowden, 2017-09-12:
Examples are not rendering, due to ShapeResult has no property Wrapped.

This is a side affect of changes to make examples CQGI compliant.

dcowden, 2017-09-12:
Traceback (most recent call last): File "/home/travis/virtualenv/python2.7.13/lib/python2.7/site-packages/cadquery-0.0.0-py2.7.egg/cadquery/cq_directive.py", line 41, in cq_directive exporters.exportShape(result.first_result, "SVG", _s) File "/home/travis/virtualenv/python2.7.13/lib/python2.7/site-packages/cadquery-0.0.0-py2.7.egg/cadquery/freecad_impl/exporters.py", line 63, in exportShape fileLike.write(getSVG(shape.wrapped))AttributeError: 'ShapeResult' object has no attribute 'wrapped'
result = cadquery.Workplane("front").box(2.0, 2.0, 0.5)
build_object(result)

jmwright, 2017-09-15:
Fixed in commit 0c40258e28c0fad08b699430bce34ea2613d8d1e

closed 2017-09-15 by jmwright