dcowden/cadquery #192: pip Installation References 0.5-SNAPSHOT

jmwright, 2017-09-13:
This is the last thing displayed in the console after running `pip install cadquery`

```
Successfully installed cadquery-0.5-SNAPSHOT
```

However, the version number that's installed is 1.0.0.

```python
cadquery.__version__
'1.0.0'
```

This may have something to do with the Travis CI deployment of the package to PyPi.

fragmuffin, 2017-09-13:
https://github.com/dcowden/cadquery/blob/master/setup.py#L19
there's ya problem

jmwright, 2017-09-13:
@fragmuffin Thanks! I need to create a doc showing all the places we need to change the version number before doing a release.

jmwright, 2017-09-13:
Fixed in commit https://github.com/dcowden/cadquery/commit/9db2328a1e2114350fd66571f7da46e103c4f82f but I'll hold off on tagging it as a release. We'll fix it when v1.1.0 is released.

closed 2017-09-13 by jmwright

fragmuffin, 2017-09-14:
alternatively pull the version directly from the `cadquery/__init__.py` file.
I was reading about PyPi best practices and ended up combining advice in 2 articles:
* http://peterdowns.com/posts/first-time-with-pypi.html and
* https://hynek.me/articles/sharing-your-labor-of-love-pypi-quick-and-dirty/

I've documented it all [here](https://github.com/fragmuffin/pygcode/tree/master/deployment). `setup.py` implementation is [here](https://github.com/fragmuffin/pygcode/blob/master/setup.py).
It works beautifully, and I've got almost zero redundancy across the entire module.