dcowden/cadquery #203: Python 3 support

Grawp, 2017-10-10:
Is Python 3 ever going to be supported?

jmwright, 2017-10-10:
There's some discussion going on about this in #162 

We're not philosophically opposed to Python 3 support, it just hasn't happened yet. It's a little bit of an unknown quantity to @dcowden and I too, which doesn't help.

fragmuffin, 2017-10-11:
It's probably a good idea to support it, 3 has some advantages, but ultimately python3 support will have to happen at some point, so it's definitely better to introduce it earlier rather than later.

I've previously set up a testing suite for py2 / py3, so I could make a PR to at least make py3 testing easier.

I've got generally less time over the next few weeks, but I'll try to allocate some time to make a PR.

dcowden, 2017-10-12:
What version of python does FreeCAD bundle?  I think a good time to switch is when they switch.

jmwright, 2017-10-12:
Both 0.16 and 0.17 show Python 2.7.12 on my Ubuntu system, but I think FreeCAD uses the system Python version in Linux.

On Windows with 0.16 r6706 the python version is 2.7.8.

adam-urbanczyk, 2017-10-26:
I just found out that there is FreeCAD conda package with py3.6 for windows and linux:

https://anaconda.org/freecad/freecad/files

If it works properly we can support python3 easily. I will test this once I have some time.

jmwright, 2017-10-26:
Awesome, that will be very helpful.

jmwright, 2018-02-09:
Thanks to the work of @adam-urbanczyk and a few other community members, we now have CI integration for Python 3 testing, and all the tests pass.

closed 2018-02-09 by jmwright