dcowden/cadquery #208: Parallel Direction Selector Broken In FreeCAD 0.17 and CadQuery > 1.0.0

jmwright, 2017-12-07:
@easyw Brought this to my attention.

On FreeCAD 0.17 with CQFM (the FreeCAD module) versions including CadQuery commits newer than [release 1.0.0](https://github.com/dcowden/cadquery/commit/731efe57e7f8fe2fb3b23108df28b174eda8f5d3), any script that uses the parallel direction selector will fail. [This example](https://github.com/dcowden/cadquery/blob/master/examples/FreeCAD/Ex020_Rounding_Corners_with_Fillets.py) shows what happens. The error is as follows.
```python
Error executing CQGI-compliant script. Fillets requires that edges be selected
```
I have started looking into this as I get time. I suspect it's something with one of the changes we've made to the selectors since the 1.0.0 release. 

Here is a test script that I've started using for experiments to try to track down root cause.
```python
import FreeCAD
import cadquery as cq
from cadquery import selectors
from Helpers import show

def selectEdges(selectee):
    selector = '|Z'
    objType = 'Edges'

    toReturn = selectee._collectProperty(objType)

    if selector is not None:
        if isinstance(selector, str):
            selectorObj = selectors.StringSyntaxSelector(selector)
        else:
            selectorObj = selector

        toReturn = selectorObj.filter(toReturn)

    return toReturn #cq.CQ.newObject(toReturn)

result = cq.Workplane("XY").box(3, 3, 0.5)#.edges("|Z").fillet(0.125)
fc_result = result.val().wrapped.Edges
FreeCAD.Console.PrintMessage("Edges: " + str(len(fc_result)) + "\r\n")
FreeCAD.Console.PrintMessage("Edges: " + str(result.edges().size()) + "\r\n")
FreeCAD.Console.PrintMessage("Edges: " + str(len(selectEdges(result))) + "\r\n")

# Displays the result of this script
show(result)
```
Somewhere after version 1.0.0 was released, the last FreeCAD.Console line went from displaying "4" to displaying "0". I'll continue working on this, but my time right now is pretty limited. If anyone has any insights into what might be happening, I'd be glad to hear them.

jmwright, 2017-12-08:
FreeCAD is failing silently here, and it seems to be the `Part.Line` versus `Part.LineSegment` problem we've seen before with FreeCAD 0.17. This line of code here is the problem.

https://github.com/dcowden/cadquery/blob/731efe57e7f8fe2fb3b23108df28b174eda8f5d3/cadquery/selectors.py#L161

If I output `o.geomType()` to the console, FreeCAD gives the following message.
```
Unknown Edge Curve Type: <type 'Part.Line'>
```
There's a suggestion in the [last post here](https://forum.freecadweb.org/viewtopic.php?style=4&f=22&t=24885) that gives an idea of how to make code compatible with 0.16 and 0.17.

@dcowden @adam-urbanczyk What are your thoughts on this?

adam-urbanczyk, 2017-12-08:
@jmwright  I do not understand the root cause. What changed in 0.17?

jmwright, 2017-12-08:
I think they got rid of Part.Line in favor of Part.LineSegment in 0.17.

jmwright, 2017-12-09:
@adam-urbanczyk See the second response down (from Chris_G) [here](https://forum.freecadweb.org/viewtopic.php?f=22&t=25252).

easyw, 2017-12-09:
A discussion here

https://staging.freecad.io/forum/viewtopic.php?f=10&t=18767&start=20

adam-urbanczyk, 2017-12-09:
@jmwright  @easyw  thanks!

Looks like this line needs fixing (in "my" 0.17 both Part.Line and Part.LineSegment exist):

https://github.com/dcowden/cadquery/blob/bdb87b4ed3bdc56251458bb97b7713113b345869/cadquery/freecad_impl/shapes.py#L430

For completeness:
```
OS: Ubuntu 16.04.3 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.12619 (Git)
Build type: None
Branch: master
Hash: b3631e134be8bfade0534f2dc0ee668ae71a223f
Python version: 2.7.12
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
```

jmwright, 2017-12-10:
@adam-urbanczyk Thanks. If my suggested change below looks reasonable, I'll put it into a PR. I tested in both FreeCAD stable (0.16) and FreeCAD daily (0.17) and everything worked fine.
```python
if hasattr(FreeCADPart,"Line"):
    self.edgetypes[FreeCADPart.Line] = 'LINE'

if hasattr(FreeCADPart,"LineSegment"):
    self.edgetypes[FreeCADPart.LineSegment] = 'LINE'

# if hasattr(FreeCADPart,"LineSegment"):
#     #FreeCAD <= 0.16
#     self.edgetypes[FreeCADPart.LineSegment] = 'LINE'
# else:
#     #FreeCAD >= 0.17
#     self.edgetypes[FreeCADPart.Line] = 'LINE'
```
```
OS: Ubuntu 17.10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.6712 (Git)
Build type: None
Branch: releases/FreeCAD-0-16
Hash: da2d364457257a7a8c6fb2137cea12c45becd71a
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17
```
```
OS: Ubuntu 17.10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.12772 (Git)
Build type: None
Branch: master
Hash: c8c9b76f43d2c78c4afb65b756f3f07d626aac85
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/UnitedStates (en_US)
```

jmwright, 2017-12-11:
Added a PR [here](https://github.com/dcowden/cadquery/pull/209)

jmwright, 2017-12-11:
That PR was checked out by @adam-urbanczyk and merged, so I'm going to close this issue.

closed 2017-12-11 by jmwright