dcowden/cadquery #211: cadquery consumes KeyboardInterrupt

fragmuffin, 2017-12-19:
importing `cadquery` seems to consume the `KeyboardInterrupt` exception.

consider the following script

```python
import time
#import cadquery

try:
    while True:
        time.sleep(1)
        print('zzz...')

except KeyboardInterrupt:
    print("\n[keyboard interrupt]")

finally:
    print('[done]')

```

prints `zzz...` every second until a user presses `[ctrl+c]`.
however, if the `import cadquery` line is uncommented, pressing `[ctrl+c]` does nothing.

I'm assuming this is actually a freecad issue, but is there anything we can do to stop it?

fragmuffin, 2017-12-20:
**workaround** with `signal`

```python
#!/usr/bin/env python

import time
import cadquery

try:
    import signal
    def _handler_sigint(signum, frame):
        raise KeyboardInterrupt()
    signal.signal(signal.SIGINT, _handler_sigint)

    while True:
        time.sleep(1)
        print('zzz...')

except KeyboardInterrupt:
    print("\n[keyboard interrupt]")

finally:
    print('[done]')
```

jmwright, 2017-12-20:
@fragmuffin It probably is a FreeCAD thing. One of the first things CadQuery has to do is import FreeCAD as a library, and I'm not aware of anything that CadQuery does with the console.

fragmuffin, 2017-12-20:
@jmwright 
Yeah, that makes sense.

I've used the above workaround to get it working on my system, I haven't tried it with any OS other than Linux just yet.

I know i just opened this issue, but I'm happy for it to be closed; I'm not sure what's reasonable to change in `cadquery` to "fix" this issue.
But this issue serves as a searchable paper-trail for anybody else that comes across this problem.

dcowden, 2017-12-20:
Sadly, this isnt the first time we've had to do yuky workarounds due to freecad.

Here is [another example](https://github.com/dcowden/cadquery/blob/master/cadquery/freecad_impl/__init__.py#L27), in which I found it necessary to gobble stderr and stdout because Freecad prints "FreeCAD version xxx" to stdout on startup and a bunch of stuff when exporting STEP-- but I needed to use stdout to output an object in the cadquery docker container.

I reviewed the above and I dont think it would gobble keyboard interrupts.

fragmuffin, 2017-12-21:
@dcowden 
I wrote something similar to try to stop FreeCAD from cluttering stdout, but it didn't work; I used a different method.
That's brilliant, I'm going to use that for the `cqparts` passthrough's for the exporters ;)

closed 2018-02-23 by jmwright